nordicsemi/IOS-DFU-Library #151: Latest Zip (1.0.0) pod doesn't compile

casey-pax, 2017-10-05:
I can't get the Zip Pod to compile with the latest iOSDFULibrary (4.0.2).

```/Volumes/Development/pax-app-ios/PaxApp/Pods/Zip/Zip/Zip.swift:174:118: Cannot convert value of type '[FileAttributeKey : Date]' to expected argument type '[String : Any]?'```

[image]

steffenmauch, 2017-10-09:
This is a problem of the Zip Pod; solution is given here https://github.com/marmelroy/Zip/pull/106/commits/ba620b25bff55a44770c00acb59ac8730e579221

mostafaberg, 2018-02-21:
This issue has been resolved a long time ago in Zip, and is not really related to our library, so will close :), if you have any more issues let us know!

closed 2018-02-21 by mostafaberg

cloudface, 2018-06-06:
I dont understand. When I install the cocoa pod for the DFU iOS library, it pulls the zip 1.0 pod with it which doesn't compile. How can I get the correct version of Zip that compiles when trying to install the DFU iOS library with cocoa pods?