nordicsemi/IOS-DFU-Library #152: DfuTarg after firmware update

fxtech, 2017-10-23:
Hello! I've gotten the latest iOS Pods library to work and I can upload firmware to my device. However, after I did this my device name had changed from a string/serial number to "DfuTarg". Everything seems to functioning however. I might have interrupted the firmware update mid-way through the first test, and I believe I had erroneously had an existing BT connection through the normal iOS central manager (I've removed that connection code on my end since then). What did I do wrong? Is there any way to get the device name back to the original name? Or is there a way to reset the name via the DFU library somehow?

Note - this is not my hardware design nor did I write the firmware. I'm just implementing the firmware update functionality in an app.

philips77, 2017-10-23:
Hi, looks like your iPhone has cached the device name. If you turn Bluetooth off and on again and scan/connect the name should be refreshed and taken from the adv packet. During the DFU process the device disconnects and advertises as DfuTarg.

fxtech, 2017-10-23:
Okay, good news. The original device name does show up now. Originally, you are right it was cached, and I had tried toggling BT mode and even restarting the phone, but that didn't seem to help. But now it is showing up using a BT scanner under the proper name. Thanks for the tip!

closed 2017-10-23 by fxtech