nordicsemi/IOS-DFU-Library #153: Error: SD version check failed

vikeri, 2017-10-24:
I'm updating APP+SD+BL and it works on Android and via an HDK but it fails on iOS. Could it be related to this PR: https://github.com/NordicSemiconductor/pc-nrfutil/pull/64 ?

The error I get is:
```
Error: SD version check failed
```

To be clear, my old app allowed only SD 4.0.2 but my new package has SD 4.0.4 and allows both SD 4.0.4 and SD 4.02.

vikeri, 2017-10-24:
NVM, I solved it by adding the old SD-id to `--sd-id` in the `pkg generate` command.

closed 2017-10-24 by vikeri

philips77, 2017-10-24:
Hi, in Secure DFU it's not possible to update the same bootloader twice. Each next must have higher bootloader version. If you flash the old sd+bl+app, and update it in iOS it should work just like it did on Android. Error SD version check is sent from the DFU bootloader.