nordicsemi/IOS-DFU-Library #154: Unable to retry BL+SD+APP update if aborted the first time during APP upload

vikeri, 2017-10-24:
When updating BL+SD+APP at the same time the process consists of two steps
1. BL+SD
2. APP

If the process is aborted during 1. it will recover fine and re-flash it successfully but if it fails during 2. it the process can't be restarted because then it will complain about FW version mismatch. I suppose this is due to the fact that a BL can't be overwritten by one with the same version this is fine when only flashing a BL but problematic when flashing all three since then the APP update is blocked. A possible solution is that if the process would detects that the BL is the same version it directly tries to flash the APP given that it's a bundle.

This issue is also prevalent in Android and pc-ble-driver-js

philips77, 2017-10-24:
Yes, it's true. If you get SD version check error you may however send only app. On Android you may use setScope(..) method in the DfuServiceInitiator with Scope Application (then only app will be sent even if you have sd+bl in the zip). On iOS you may create Firmware again, this time seeing the DfuFirmwareType to .application and try again. Those libs don't do this automatically as it's higher level logic, not always necessary.

vikeri, 2017-10-24:
Ah ok, wasn't aware of the `setScope` method. What do you mean by creating the firmware again? I'm creating it with `nrfutil pkg generate`.

philips77, 2017-10-24:
No, I'm just taking about the [DfuFirmware](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/Firmware/DFUFirmware.swift) object in iOS. There is a constructor https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/b2c2f4f7c2f26a8b13b44e5d3675e83ddff665f1/iOSDFULibrary/Classes/Implementation/Firmware/DFUFirmware.swift#L106 which takes type as an argument, so you may usev the same zip file for update-all and update-app-only.

vikeri, 2017-10-24:
Nice! I'm using the React Native wrapper that does not support that yet. Worth adding.

philips77, 2017-11-24:
Let's close this issue here.

closed 2017-11-24 by philips77