nordicsemi/IOS-DFU-Library #155: Unable to perform DFU on a new peripheral

anlaital, 2017-10-26:
We have an older peripheral that is using the legacy DFU service and everything works fine with it using the latest version of the library.

Our new peripheral, however, is using the secure DFU service. DFU fails as the library does not seem to find the required control point characteristics. Any guess as to why this might happen?

It's worth noting that for some reason the DFU service requires using the `enableUnsafeExperimentalButtonlessServiceInSecureDfu` flag to `true`. I don't fully understand the meaning of this flag, so this could also indicate an issue with the bootloader.

DFU succeeds when using NRF connect on the peripheral.

philips77, 2017-10-26:
Hi, what SDK are you using in the new peripheral? The `enable..Dfu` is required only for SDK 12's buttonless DFU service. If you are using SDK 13+ this flag is ignored.
Regarding finding the control point characteristic, do you use the standard UUIDs or custom ones? Is the buttonless update working for you, just the dfu fails, or the opposite? 
Also, make sure you have the Service Changed characteristic in your application.
Could you post the logs you get from the library here?

anlaital, 2017-10-30:
Sorry to keep you waiting!

We are using SDK 12.2. We use standard UUIDs.

Here are the logs from a failed attempt:

```
Connecting to ***...
centralManager.connect(peripheral, options: nil)
DFU state changed to Connecting
[Callback] Central Manager did connect peripheral
Connected to ***
Discovering services...
peripheral.discoverServices(nil)
Services discovered
Starting Secure DFU...
Connected to ***
Services discovered
Secure DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: FE59)
DFU characteristics discovered
DFU Control Point characteristics not found
Disconnecting...
centralManager.cancelPeripheralConnection(peripheral)
[Callback] Central Manager did disconnect peripheral
Disconnected
DFU error DFUError with message DFU Control Point characteristic not found
```

```
(lldb) po peripheral.services!
▿ 2 elements
  - 0 : <CBService: 0x1c6c79700, isPrimary = YES, UUID = 8E400001-F315-4F60-9FB8-838830DAEA50>
  - 1 : <CBService: 0x1c6e71f80, isPrimary = YES, UUID = 98ED0001-A541-11E4-B6A0-0002A5D5C51B>

(lldb) po dfuService
<CBService: 0x1c6c79700, isPrimary = YES, UUID = 8E400001-F315-4F60-9FB8-838830DAEA50>
```

DFU succeeds when using `pc-nrfutil` to flash the device.

philips77, 2017-10-30:
Yes, so `enableUnsafeExperimentalButtonlessServiceInSecureDfu` should be set to true in your case, but this log seems to be from the connection to the bootloader, not app with buttonless jump feature. I'll come back to you on Wednesday and test if this still works as it should. Verify that it's not related to ios caching services. Try using a new device, that you didn't connect before. Check if it has Service Changed char in both app and bootloader modes.

anlaital, 2017-10-31:
We use our own API to set the peripheral to DFU mode, so we are also using the `forceDfu` flag set to `true`.

Here are the captures from both the iOS failure case and nrfutil success case in case they are of any help.

[Archive.zip](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/1429686/Archive.zip)

I don't think we are using the service changed characteristics at all. I will ask further info from our firmware guys about this.

anlaital, 2017-11-01:
We managed to fix this issue. The root cause was incorrectly using Service Changed indication in the bootloader of the new peripheral. Now everything is working perfectly, so thank you for your great support! 👍

closed 2017-11-01 by anlaital