nordicsemi/IOS-DFU-Library #156: EXC_BAD_ACCESS in [IntelHex2BinConverter convert]

bisikli, 2017-10-27:
Hello, 

I am having the following crash during a dfu. Could you please have a look? 

Thanks. 

[image]

philips77, 2017-10-28:
Are you sure your hex file is correct? Seems like its size is 0 or is not a hex file. Put a breakpoint in that line and check the calculated sizes and why the loop didn't break before crashing. Could you send us the file if it looks ok?
Also, we recommend using the distribution packets (Zip files) instead of hex+dat file.

bisikli, 2017-11-14:
Closing this thread for now as i am not sure about the hex file and will try to test with a zip version. 

Thank you!

closed 2017-11-14 by bisikli

mostafaberg, 2017-11-14:
Ok keep us updated, If you still have issues just let us know