nordicsemi/IOS-DFU-Library #157: Question: How do you discern which device to connect to during Buttonless DFU?

vikeri, 2017-10-27:
I have one device running the app and then a couple of devices in DFU mode. When the device is put in DFU mode by the Buttonless DFU functionality how do you determine which device to re-connect to? I know the device in DFU has a MAC address that is +1 of the normal MAC address but my understanding is that iOS does not expose the MAC-address.

The reason I'm asking is just to double check that there is no risk for me to do DFU on the wrong device when I have multiple devices available.

philips77, 2017-10-27:
Hi, so in all SDKs before 14 there is a chance you connect to a wrong one. DfuServiceInitiator exposes the [peripheralSelector](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L71) which could be set to override the default behavior which is selecting the first device advertising with DFU UUID (legacy or secure, depending on what buttonless service was found). You may for example use the RSSI to select the closes, etc. 

In SDK 14 the "set bootloader name" feature was added, where during jumping to the DFU mode the lib sends a random name before jump command and the bootloader will advertise with this name instead of DfuTarg. So if the device supported it the lib will then scan for this name (see [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/DFUPeripheralSelector.swift#L28)).

philips77, 2017-11-22:
Link to the specs: http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v14.0.0/service_dfu.html

closed 2017-11-22 by philips77