nordicsemi/IOS-DFU-Library #158: Stuck on "Jump to bootloader"

tompi, 2017-11-08:
Hi. Im assuming I am doing something wrong, but cant quite figure it out... Hope you can point me in the right direction!

I have used sharpie to bind your swift lib for use in xamarin. When I try to do DFU from ios code, the log looks like this(Im writing all of the errors and callbacks from Progress and Logger delegates):
It just stops after last line... I am keeping a reference to the objects...

```
Connected to 57
Discovering services...
peripheral.discoverServices(nil)
Services discovered
Starting Legacy DFU...
Connected to 57
Services discovered
Legacy DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
DFU characteristics discovered
Reading DFU Version number...
peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
Version number read: 0.1
Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Starting
Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
DFU Control Point notifications enabled
Application with buttonless update found
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
EnablingDfuMode
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
```

mostafaberg, 2017-11-08:
Hi @tompi , sorry for the inconvenience you are experiencing, to rule out that the issue is on your side, have you tried using one of our example apps to perform the DFU process and see that it actually works?

If so, you need to verify that this is not an issues with the iOS cache, to do so, quit the app, and toggle bluetooth on/off a couple of times from the settings app and retry, sometimes restarting the phone helps to force clear the cache.

is it only one peripheral or do you have more than one nearby ?

philips77, 2017-11-08:
After sending 0x0104 your device should disconnect and switch to bootloader mode, version 0.1 says it's in the app mode with buttonless service supported. It doesn't do that. The lib is waiting for it to disconnect. Ensure it does it using some other app, what Mostafa suggested, or that your xamarin wrapper really works and the date are sent (although enabling notifications worked, so i assume this works as well).

philips77, 2017-11-08:
It may be also, that you are actually on the bootloader mode and your device incorrectly returns version 0.1 instead of 0.5+.

tompi, 2017-11-08:
Thank you for following up. Lightning fast! :)

I tried a few other devices, same thing happening.
The device is definitely not put in DFU mode, since I can connect fine to it with other apps.

I will try to compile your lib with some more logging to find out exactly where things are stopping. 

BTW: I only needed to mod one thing to get the binding right for xamarin: needed to change all the "objc" annotations to include class names to avoid clang from generating dynamic names on compile time. Would be nice if you did this! So instead of "@objc public class DFUFirmware", do "@objc(DFUFirmware) public class DFUFirmware".

This is the lib we are using today, but it is starting to feel very obsolete...
https://bitbucket.org/arendiag/arendi_nordicdfu/

tompi, 2017-11-08:
OK, I talked to our firmware guy, and seems we need to send an authenticate message before device allows us to set it in dfu mode. I tried to call your lib on a device which was connected AND authenticated, but then it stops on Connecting... Does it support starting DFU on an already connected device, or do I need to put it in dfu mode BEFORE I use your lib?

```
Connected to 57
Services discovered
Starting Legacy DFU...
Connected to 57
Services discovered
Legacy DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
Connecting
```

philips77, 2017-11-08:
Yes, you may call it on already connected CBPeripheral in app mode. But it must be obtained and connected using the same CBCentralManager. You connect, send auth token and start dfu. The lib will just start from enabling notifications and continue forward.

tompi, 2017-11-08:
Hmm, the source code comments for "start()" says otherwise:

```
The peripheral should be disconnected prior to calling start() method.
```

I´ll give it another try, and make sure I use the same central when connecting and doing dfu.
P.S. Got this working for android without any problems, the android constructor I used when implementing dfu just took a string(ble mac address for device to upgrade).

philips77, 2017-11-08:
Ups.. bug in the docs! Check out the code here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift#L96

mostafaberg, 2017-11-08:
@tompi that makes sense, however i don't think you really need to manually disconnect or do anything, our library handles that normally.
@philips77 is it true that we need to be disconnected? I don't think so right?

tompi, 2017-11-08:
Sorry to keep bothering you about this, but Im still having trouble. I verified that the central used to discover device is the same as the one sent in to the initiator.

Added some logging, and tried setting the device in DFU mode manually, here is the log, hope you could suggest something I could try!

```
Central: <CBCentralManager: 0x1c426a380>
DFU initiated:  : LittleAnne
08.11.2017 13.58.31: DfuStateDidChangeTo: Connecting
08.11.2017 13.58.31: Logger: Level: Info. Message: Peripheral state: CBPeripheralState != connected => Initiate connect
08.11.2017 13.58.31: Logger: Level: Verbose. Message: Connecting to DfuTarg...
08.11.2017 13.58.31: Logger: Level: Debug. Message: centralManager.connect(peripheral, options: nil)
08.11.2017 13.58.31: Logger: Level: Debug. Message: [Callback] Central Manager did connect peripheral
08.11.2017 13.58.31: Logger: Level: Info. Message: Connected to DfuTarg
08.11.2017 13.58.32: Logger: Level: Verbose. Message: Discovering services...
08.11.2017 13.58.32: Logger: Level: Debug. Message: peripheral.discoverServices(nil)
08.11.2017 13.58.32: Logger: Level: Info. Message: Services discovered
08.11.2017 13.58.32: Logger: Level: Info. Message: A DFU service was found, congratulations!
08.11.2017 13.58.32: Logger: Level: Verbose. Message: Starting Legacy DFU...
08.11.2017 13.58.32: Logger: Level: Info. Message: Destroying peripheral
08.11.2017 13.58.32: Logger: Level: Info. Message: Connected to DfuTarg
08.11.2017 13.58.32: Logger: Level: Info. Message: Services discovered
08.11.2017 13.58.32: Logger: Level: Verbose. Message: Legacy DFU Service found
08.11.2017 13.58.32: Logger: Level: Verbose. Message: Discovering characteristics in DFU Service...
08.11.2017 13.58.32: Logger: Level: Debug. Message: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
08.11.2017 13.58.32: Logger: Level: Info. Message: DFU characteristics discovered
08.11.2017 13.58.32: Logger: Level: Verbose. Message: Reading DFU Version number...
08.11.2017 13.58.32: Logger: Level: Debug. Message: peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
08.11.2017 13.58.32: Logger: Level: Info. Message: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0600
08.11.2017 13.58.32: Logger: Level: Application. Message: Version number read: 0.6
08.11.2017 13.58.32: Logger: Level: Verbose. Message: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
08.11.2017 13.58.32: Logger: Level: Debug. Message: peripheral.writeValue(0x06, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
08.11.2017 13.58.33: Logger: Level: Info. Message: cleanup
08.11.2017 13.58.33: Logger: Level: Debug. Message: [Callback] Central Manager did disconnect peripheral
08.11.2017 13.58.33: Logger: Level: Info. Message: Disconnected by the remote device
08.11.2017 13.58.33: Logger: Level: Info. Message: disconnect
08.11.2017 13.58.33: Logger: Level: Info. Message: peripheralDidDisconnect
08.11.2017 13.58.33: Logger: Level: Info. Message: Destroying peripheral
08.11.2017 13.58.33: Logger: Level: Info. Message: cleanup
08.11.2017 13.58.33: ServiceDelegate: The init packet is required by the target device
```

Are the "destroy" for the peripheral as expected? (I added logging of these in your lib)

My C# code looks like this

```
                            Log.Info("DFU initiated: " + device.ManufacturerData.DeviceName + " : " + device.ManufacturerData.DeviceType);

                            // Load firmware bundle
                            var fwPath = Path.Combine(NSBundle.MainBundle.BundlePath, "Firmware/QCPRPlugin_Application_1_0_116_251.hex");
                            var fw = new DFUFirmware(new NSUrl(fwPath, false), null, DFUFirmwareType.Application);

                            // Initiate upgrade
                            initiator = new ServiceInitiator(Adapter.Current.Central, cbPeripheral);

                            initiator.EnableUnsafeExperimentalButtonlessServiceInSecureDfu = true;
                            initiator.ProgressDelegate = new ProgressDelegate();
                            initiator.Delegate = new ServiceDelegate();
                            initiator.Logger = new Logger();
                            controller = initiator.WithFirmware(fw).Start();
```

mostafaberg, 2017-11-08:
You are missing the init packet according to this line:
`08.11.2017 13.58.33: ServiceDelegate: The init packet is required by the target device`
to create one use nRFUtil, you can find a similar question on our devzone [here](https://devzone.nordicsemi.com/question/59317/dfu-what-is-init-packet-file-dat-and-how-to-get-it/)

or you may submit a new question there if you can't get around it :)

tompi, 2017-11-08:
OK, I thought the docs said the zip-option was only for all 3 bootloader+softdevice+app. Silly of me...

Tried with zip file and device in dfu mode and everything works perfectly. Thnx for all your help!

```
Part: 1 of 1 (92 %)
Part: 1 of 1 (93 %)
Part: 1 of 1 (94 %)
Part: 1 of 1 (95 %)
Part: 1 of 1 (96 %)
Part: 1 of 1 (97 %)
Part: 1 of 1 (98 %)
Part: 1 of 1 (99 %)
Part: 1 of 1 (100 %)
08.11.2017 14.30.36: Logger: Level: Info. Message: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100301
08.11.2017 14.30.36: Logger: Level: Application. Message: Response (Op Code = 3, Status = 1) received
08.11.2017 14.30.36: Logger: Level: Application. Message: Upload completed in 45.81 seconds
08.11.2017 14.30.36: Logger: Level: Verbose. Message: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
08.11.2017 14.30.36: Logger: Level: Debug. Message: peripheral.writeValue(0x04, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
08.11.2017 14.30.36: DfuStateDidChangeTo: Validating
08.11.2017 14.30.36: Logger: Level: Info. Message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
08.11.2017 14.30.36: Logger: Level: Application. Message: Validate Firmware (Op Code = 4) request sent
08.11.2017 14.30.36: Logger: Level: Info. Message: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100401
08.11.2017 14.30.36: Logger: Level: Application. Message: Response (Op Code = 4, Status = 1) received
08.11.2017 14.30.36: Logger: Level: Verbose. Message: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
08.11.2017 14.30.36: Logger: Level: Debug. Message: peripheral.writeValue(0x05, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
08.11.2017 14.30.36: DfuStateDidChangeTo: Disconnecting
08.11.2017 14.30.36: Logger: Level: Info. Message: Data written to 00001531-1212-EFDE-1523-785FEABCD123
08.11.2017 14.30.36: Logger: Level: Application. Message: Activate and Reset (Op Code = 5) request sent
Thread finished: <Thread Pool> #4
Thread started: <Thread Pool> #12
08.11.2017 14.30.37: Logger: Level: Info. Message: cleanup
08.11.2017 14.30.37: Logger: Level: Debug. Message: [Callback] Central Manager did disconnect peripheral
08.11.2017 14.30.37: Logger: Level: Info. Message: Disconnected by the remote device
08.11.2017 14.30.37: Logger: Level: Info. Message: disconnect
08.11.2017 14.30.37: Logger: Level: Info. Message: This part of firmware has been successfully
08.11.2017 14.30.37: Logger: Level: Info. Message: Destroying peripheral
08.11.2017 14.30.37: Logger: Level: Info. Message: cleanup
08.11.2017 14.30.37: DfuStateDidChangeTo: Completed
```

I can live with manually putting it into dfu mode(will try again with auth+ios-dfu but its not that important...).

Again, thank you for all your help!

mostafaberg, 2017-11-08:
Awesome! glad to hear things worked after all, yes the Init packet is required for all cases not just a merged zip

if you need more support with the other issue of having to reconnect just open up a new ticket and we will gladly take a look !, happy hacking !

closed 2017-11-08 by mostafaberg