nordicsemi/IOS-DFU-Library #161: @"NSInvalidArgumentException" - reason: @"-[AboutNavitaViewController logWith:message:]: unrecognized selector sent to instance 0x103673550"    0x00000001c0643720

VikramSinghG, 2017-11-23:
Stuck on this issue . Perfectly worked on android but with IOS not working getting same issue again

VikramSinghG, 2017-11-23:
@"NSInvalidArgumentException" - reason: @"-[AboutNavitaViewController logWith:message:]: unrecognized selector sent to instance 0x103673550"    0x00000001c0643720

                      
    func start() {
        DispatchQueue.main.async(execute: {
             self.delegate?.dfuStateDidChange(to: .connecting)
        })
        peripheral.start()
    }
    
Getting Error On # Line self.delegate?.dfuStateDidChange(to: .connecting)

mostafaberg, 2017-11-23:
The delegate is `AboutNavitaViewController` might not be conforming tot he delegate's protocol, can you check your class that it actually has the delegate methods and conforms to them ?

VikramSinghG, 2017-11-23:
Yes it has the delegate's protocol below is the process log

Printing description of message:
Connected to QS-00
Printing description of message:
Services discovered
Printing description of message:
Starting Legacy DFU...
Printing description of message:
Connected to QS-00
Printing description of message:
Services discovered
Printing description of message:
Legacy DFU Service found
Printing description of message:
Discovering characteristics in DFU Service...
Printing description of message:
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
Printing description of message:
DFU characteristics discovered
Printing description of message:
Reading DFU Version number...
Printing description of message:
peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
Printing description of message:
Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
Printing description of message:
Version number read: 0.1
Printing description of message:
Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
Printing description of message:
peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Printing description of message:
Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
Printing description of message:
DFU Control Point notifications enabled
Printing description of message:
Application with buttonless update found
Printing description of message:
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
Printing description of message:
peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Printing description of message:
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Printing description of message:
Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent

after that band restart but nothing happened

mostafaberg, 2017-11-23:
This is a different issue now, the first one is resolved ? if so close this one so people with similar issues know how to easily solve it in the future, and open a new issue with your new bug and we'll help you further

closed 2017-11-23 by VikramSinghG