nordicsemi/IOS-DFU-Library #162: dfuError:didOccurWithMessage: method dosen`t invoked in ios 9.3 and ios 11.1

PansGitHub, 2017-11-23:
When uploading package, i  closed the bluetooth but the method dfuError:didOccurWithMessage: dosen`t invoked, i test it many times in iOS 9.3 and iOS 11.1 .   

But in iOS  10.3.3, this method can be invoked correctly, how do I solve this problem?

 other info :
 iOSDFULibrary version   3.2.1
 xCode version   8

philips77, 2017-11-23:
I tested on iOSDFULibrary 4.02 (current one) and it worked on iOS 11.1.2. I disabled the Bluetooth on the Control Panel and I got the above error with 203 error: Bluetooth adapter powered off.

PansGitHub, 2017-12-01:
Thanks for your reply , I downloaded your newest example project, and the problem still appear
As you can see, when I disabled the Bluetooth during uploading, the result was different
[image iOS 11](https://user-images.githubusercontent.com/25220218/33467250-1a036afc-d68e-11e7-9515-eb7d77d79012.png)
[image  iOS 10](https://user-images.githubusercontent.com/25220218/33467245-10e30766-d68e-11e7-9d87-82bb65146077.jpeg)

in iOS 11(or9) func dfuError(_ error: DFUError, didOccurWithMessage message: String) dosen`t invoked

I notice that your callback chain is centralManager->executor->viewController

And I notice the func centralManager(_ central: CBCentralManager, didDisconnectPeripheral peripheral: CBPeripheral, error: Error?)  only invoked in iOS 10. so executor cannot receive disconnect event in iOS 11

Look forward to your reply

philips77, 2018-02-21:
Turning off Bluetooth on iOS 11 from the Control Center doesn't turn off the Bluetooth, just disables new connections. But connected devices are still connected.
You may check it by disabling it and then navigating to Settings -> Bluetooth. You'll see it enabled and there will be a button to allow new connections.
So you doesn't get this callback as the device wasn't in fact disconnected.

philips77, 2018-02-21:
Btw, sorry for the super late response!

closed 2018-06-19 by mostafaberg