nordicsemi/IOS-DFU-Library #164: DFU Process start but band restart nothing happened

VikramSinghG, 2017-11-23:
Printing description of message:
Connected to QS-00
Printing description of message:
Services discovered
Printing description of message:
Starting Legacy DFU...
Printing description of message:
Connected to QS-00
Printing description of message:
Services discovered
Printing description of message:
Legacy DFU Service found
Printing description of message:
Discovering characteristics in DFU Service...
Printing description of message:
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
Printing description of message:
DFU characteristics discovered
Printing description of message:
Reading DFU Version number...
Printing description of message:
peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
Printing description of message:
Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
Printing description of message:
Version number read: 0.1
Printing description of message:
Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
Printing description of message:
peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Printing description of message:
Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
Printing description of message:
DFU Control Point notifications enabled
Printing description of message:
Application with buttonless update found
Printing description of message:
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
Printing description of message:
peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Printing description of message:
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Printing description of message:
Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent

after that band restart but nothing happened

philips77, 2017-11-23:
Your buttonless service in the app should restart your device in bootloader mode at that point. The service is awaiting disconnection to proceed.
If your device is already in DFU bootloader mode it should return version other than 0.1 (0x0100).

VikramSinghG, 2017-11-23:
Device is in boot loader mode it return 0.1

philips77, 2017-11-23:
Version 0.1 means it's in app mode with buttonless service.
Bootloader should have versions 0.5, 0.6, 0.7, 0.8 (depending on which SDK you are using).

VikramSinghG, 2017-11-23:
How to change it to Bootloader mode from app mode

VikramSinghG, 2017-11-23:
Process stuck on Enabling DFU Bootloader...

philips77, 2017-11-24:
It's not necessary. Your device, after the lib sends 0x0104 should restart in Bootloader mode. It doesn't and the lib expects disconnection. The error is on your device side. Please, check devzone for existing questions or ask your question there as this is not related to the DFU library and our support team is more qualified to answer. You may post a link to the question here for others.

VikramSinghG, 2017-11-24:
Device is updated successfully with nrf toolbox ios app but not with my code i have added iOSDFULibrary and pod is having updated version but where the process get lost in between while it start very fast in starting dfu and than nothing happen it lost connection from same band which is successfully updated from the nrf toolbox app and last band restart .

philips77, 2017-11-24:
Did you pass to the DfuServiceInitiator the same instance of CBCentralManager that you used to obtain the peripheral?

VikramSinghG, 2017-11-24:
Below is the code 

 self.centralManager = [[CBCentralManager alloc] initWithDelegate:self queue:nil];

      DFUFirmware *selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile:[NSURL fileURLWithPath:[[NSBundle mainBundle] pathForResource:@"A12_dfu_app_v11b" ofType:@"zip"]]];
                                                
                                                DFUServiceInitiator *initiator = [[DFUServiceInitiator alloc] initWithCentralManager:_centralManager target:discoveredPeripheral];
                                                [initiator withFirmware:selectedFirmware];
                                                // Optional:
                                                initiator.forceDfu = NO; // default NO
                                                initiator.packetReceiptNotificationParameter = 12; // default is 12
                                                initiator.logger = self; // - to get log info
                                                initiator.delegate = self; // - to be informed about current state and errors
                                                initiator.progressDelegate = self; // - to show progress bar
                                                // initiator.peripheralSelector = ... // the default selector is used
                                                
                                                DFUServiceController *controller = [initiator start];
                                                

Method I am using :

-(void)centralManagerDidUpdateState:(CBCentralManager *)central {
    
    [self  centralManager:(central) didPeripheralSelected:(discoveredPeripheral)];
    NSLog(@"centralManagerDidUpdateState BLE");

}

mostafaberg, 2017-11-24:
Try holding a reference to your controller, it might be getting cleaned.
so basically this:
`DFUServiceController *controller = [initiator start];`
needs to be like this:
`self.controller = [initiator start];`
and set the controller as a property in your class like you do with `centralManager`  class, let us know if this works

VikramSinghG, 2017-11-24:
I set the property like this

@property (nonatomic, strong) DFUServiceController *controller;

and than

    DFUFirmware *selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile:[NSURL fileURLWithPath:[[NSBundle mainBundle] pathForResource:@"A12_dfu_app_v11b" ofType:@"zip"]]];
                                                
                                                DFUServiceInitiator *initiator = [[DFUServiceInitiator alloc] initWithCentralManager:_centralManager target:discoveredPeripheral];
                                                [initiator withFirmware:selectedFirmware];
                                                // Optional:
                                                initiator.forceDfu = NO; // default NO
                                                initiator.packetReceiptNotificationParameter = 12; // default is 12
                                                initiator.logger = self; // - to get log info
                                                initiator.delegate = self; // - to be informed about current state and errors
                                                initiator.progressDelegate = self; // - to show progress bar
                                                // initiator.peripheralSelector = ... // the default selector is used
                                                
                                                self.controller = [initiator start];

But same thing happening again band restart .

mostafaberg, 2017-11-24:
does the band restart again in app mode ?

VikramSinghG, 2017-11-24:
Below is the log i am getting after changes

Printing description of message:
Connected to QS-00
Printing description of message:
Services discovered
Printing description of message:
Starting Legacy DFU...
Printing description of message:
Connected to QS-00
Printing description of message:
Services discovered
Printing description of message:
Legacy DFU Service found
Printing description of message:
Discovering characteristics in DFU Service...
Printing description of message:
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
Printing description of message:
DFU characteristics discovered
Printing description of message:
Reading DFU Version number...
Printing description of message:
peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
Printing description of message:
Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
Printing description of message:
Version number read: 0.1
Printing description of message:
Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
Printing description of message:
peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Printing description of message:
Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
Printing description of message:
DFU Control Point notifications enabled
Printing description of message:
Application with buttonless update found
Printing description of message:
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
Printing description of message:
peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Printing description of message:
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Printing description of message:
Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent

mostafaberg, 2017-11-24:
Yes but does when it restarts does it start the application or does it start in bootloader ?
also, can you confirm you are using the exact same firmware in your app and in nRF Toolbox? can you try one more time now in nRF Toolbox and actually confirm that it works ?

VikramSinghG, 2017-11-24:
It restart the device(band) and disconnect to the app and nothing happen while again Device is successfully updated with the same firmware with nRF Toolbox and device is nRF51822 even device is tested with android version with nRF Toolbox and with my android code working fine but not with IOS objective c

mostafaberg, 2017-11-24:
What happens if you also try to hold a reference to the target peripheral as a property? Long shot bug might be something there, if not maybe we can take a deeper look at your code to see what's going on?

philips77, 2017-11-24:
Hi, another idea. During the upload process the lib sets as the CBCentralManager's delegate, so it could get the information when the device disconnects. It looks like you set the delegate to something else while DFU is in progress. If the device restarts in DFU, but the lib is still waiting for the event, it means that the delegate didn't work. Are you sure you aren't overwriting it until you get completed/error callbacks?

VikramSinghG, 2017-11-24:
Tried So many solutions but nothing worked as project is large i have created small project with the same code except peripheral connection to main viewController . I uploaded the code  pls look into this

[DFU-TEST.zip](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/1501807/DFU-TEST.zip)

VikramSinghG, 2017-11-27:
@philips77  as you said during the process the lib is sets as the CBCentral Manager's delegate and hence below is the detail of error while it get disconnected 

Main Thread Checker: UI API called on a background thread: -[UILabel setText:]
PID: 346, TID: 63073, Thread name: (none), Queue name: HYBLEManager, QoS: 0
Backtrace:
4   CB1_Demo                            0x0000000100e9c1b0 -[ViewController logWith:message:] + 180
5   iOSDFULibrary                       0x00000001013017cc _T013iOSDFULibrary12LoggerHelperC1iySSF + 248
6   iOSDFULibrary                       0x00000001012fe760 _T013iOSDFULibrary16LegacyDFUServiceC10peripheralySo12CBPeripheralC_So9CBServiceC29didDiscoverCharacteristicsFors5Error_pSg5errortF + 980
7   iOSDFULibrary                       0x00000001012ff378 _T013iOSDFULibrary16LegacyDFUServiceC10peripheralySo12CBPeripheralC_So9CBServiceC29didDiscoverCharacteristicsFors5Error_pSg5errortFTo + 168
8   CoreBluetooth                       0x00000001881c6e74 <redacted> + 248
9   CoreBluetooth                       0x00000001881c6f2c <redacted> + 132
10  CoreBluetooth                       0x00000001881c27e8 <redacted> + 364
11  CoreBluetooth                       0x00000001881bd12c <redacted> + 208
12  CoreBluetooth                       0x00000001881cb868 <redacted> + 60
13  libdispatch.dylib                   0x0000000101f3949c _dispatch_call_block_and_release + 24
14  libdispatch.dylib                   0x0000000101f3945c _dispatch_client_callout + 16
15  libdispatch.dylib                   0x0000000101f48110 _dispatch_queue_serial_drain + 692
16  libdispatch.dylib                   0x0000000101f3c9a4 _dispatch_queue_invoke + 332
17  libdispatch.dylib                   0x0000000101f47f30 _dispatch_queue_serial_drain + 212
18  libdispatch.dylib                   0x0000000101f3c9a4 _dispatch_queue_invoke + 332
19  libdispatch.dylib                   0x0000000101f49104 _dispatch_root_queue_drain_deferred_wlh + 424
20  libdispatch.dylib                   0x0000000101f50100 _dispatch_workloop_worker_thread + 652
21  libsystem_pthread.dylib             0x00000001829aefd0 _pthread_wqthread + 932
22  libsystem_pthread.dylib             0x00000001829aec20 start_wqthread + 4
2017-11-27 15:23:59.746193+0530 CB1_Demo[346:63073] [reports] Main Thread Checker: UI API called on a background thread: -[UILabel setText:]
PID: 346, TID: 63073, Thread name: (none), Queue name: HYBLEManager, QoS: 0
Backtrace:
4   CB1_Demo                            0x0000000100e9c1b0 -[ViewController logWith:message:] + 180
5   iOSDFULibrary                       0x00000001013017cc _T013iOSDFULibrary12LoggerHelperC1iySSF + 248
6   iOSDFULibrary                       0x00000001012fe760 _T013iOSDFULibrary16LegacyDFUServiceC10peripheralySo12CBPeripheralC_So9CBServiceC29didDiscoverCharacteristicsFors5Error_pSg5errortF + 980
7   iOSDFULibrary                       0x00000001012ff378 _T013iOSDFULibrary16LegacyDFUServiceC10peripheralySo12CBPeripheralC_So9CBServiceC29didDiscoverCharacteristicsFors5Error_pSg5errortFTo + 168
8   CoreBluetooth                       0x00000001881c6e74 <redacted> + 248
9   CoreBluetooth                       0x00000001881c6f2c <redacted> + 132
10  CoreBluetooth                       0x00000001881c27e8 <redacted> + 364
11  CoreBluetooth                       0x00000001881bd12c <redacted> + 208
12  CoreBluetooth                       0x00000001881cb868 <redacted> + 60
13  libdispatch.dylib                   0x0000000101f3949c _dispatch_call_block_and_release + 24
14  libdispatch.dylib                   0x0000000101f3945c _dispatch_client_callout + 16
15  libdispatch.dylib                   0x0000000101f48110 _dispatch_queue_serial_drain + 692
16  libdispatch.dylib                   0x0000000101f3c9a4 _dispatch_queue_invoke + 332
17  libdispatch.dylib                   0x0000000101f47f30 _dispatch_queue_serial_drain + 212
18  libdispatch.dylib                   0x0000000101f3c9a4 _dispatch_queue_invoke + 332
19  libdispatch.dylib                   0x0000000101f49104 _dispatch_root_queue_drain_deferred_wlh + 424
20  libdispatch.dylib                   0x0000000101f50100 _dispatch_workloop_worker_thread + 652
21  libsystem_pthread.dylib             0x00000001829aefd0 _pthread_wqthread + 932
22  libsystem_pthread.dylib             0x00000001829aec20 start_wqthread + 4
2017-11-27 15:23:59.886955+0530 CB1_Demo[346:63073] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:23:59.887298+0530 CB1_Demo[346:63073] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:23:59.937398+0530 CB1_Demo[346:63045] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:23:59.938701+0530 CB1_Demo[346:63045] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:23:59.939190+0530 CB1_Demo[346:63012] 1 <--not implemented!
2017-11-27 15:23:59.941341+0530 CB1_Demo[346:63045] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:23:59.941776+0530 CB1_Demo[346:63045] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:24:00.056158+0530 CB1_Demo[346:63045] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:24:00.056494+0530 CB1_Demo[346:63045] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:24:00.056809+0530 CB1_Demo[346:63045] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:24:00.057101+0530 CB1_Demo[346:63012] 2 <--not implemented!
2017-11-27 15:24:00.058370+0530 CB1_Demo[346:63045] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:24:00.059369+0530 CB1_Demo[346:63045] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:24:00.116545+0530 CB1_Demo[346:63073] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:24:00.117216+0530 CB1_Demo[346:63073] \M-x\^A|\^A\^A <--not implemented!
2017-11-27 15:24:00.303215+0530 CB1_Demo[346:63045] -----外设断开连接------Error Domain=CBErrorDomain Code=7 "The specified device has disconnected from us." UserInfo={NSLocalizedDescription=The specified device has disconnected from us.}

VikramSinghG, 2017-11-29:
@mostafaberg  as you asked for code i have shared my code pls look while you get time 
[DFU_Test_nRFLib.zip](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/1513590/DFU_Test_nRFLib.zip)

philips77, 2017-11-29:
Hi, @mostafaberg is on short leave but should be at work tomorrow or on Friday.
Regarding the version returned by the device. 0.1 = app mode with buttonless suport. THe service sends 0x0104 and expects it to reboot in bootloader mode. Looks like yours does reboot, but the library is not informed about it, as if the delegate was set to something else. You say the devices got restarted, but the lib still waits for disconnect. 
0.8 means bootloader form SDK 10(?) which can handle DFU.

VikramSinghG, 2017-11-29:
Right nRFToolbox Return 0.8(SDK 11) but here it returns version 0.1
Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
Delegate is set to 
@interface AboutNavitaViewController : UIViewController<CBCentralManagerDelegate, CBPeripheralDelegate, DFUServiceDelegate,DFUProgressDelegate, LoggerDelegate>

With Property   
@property (nonatomic, strong) CBCentralManager *centralManager;

Than Calling the Delegate 
self.centralManager = [[CBCentralManager alloc] initWithDelegate:self queue:nil];

-(void)centralManagerDidUpdateState:(CBCentralManager *)central {
    
    [self  centralManager:(central) didPeripheralSelected:(_discoveredPeripheral)];
    NSLog(@"centralManagerDidUpdateState BLE");

}

-(void)centralManager:(CBCentralManager *)central didConnectPeripheral:(CBPeripheral *)peripheral{

    NSLog(@"didConnectPeripheral BLE"); [_discoveredPeripheral discoverServices:nil];
    _discoveredPeripheral = peripheral;
    _centralManager = central;
}

#pragma mark Device Selection Delegate

-(void)centralManager:(CBCentralManager *)manager didPeripheralSelected:(CBPeripheral *)peripheral
{
    _discoveredPeripheral = peripheral;
    _centralManager = manager;
    
    peripheral.delegate = self;
    [self.discoveredPeripheral discoverServices:@[[CBUUID UUIDWithString:@"A65eBB2D-3D30-4981-9DB2-1996D88118A0"]]];

}

#define IS_SRVC_CHANGED_CHARACT_PRESENT 1  is Define in ViewController.m

Pod Lib Version is 4.0.3 

Peripheral is in already Connected State .

philips77, 2017-11-29:
Yes. Returns 0.1, the lib sends 0x0104 which should restart it in Bootloader mode (and the lib should get disconnect event, after which it would reconnect to the Bootloader and proceed).

VikramSinghG, 2017-11-29:
Right but thats what i am not able to figure it out it Lib successfully write  0x0104

peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Printing description of message:
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Printing description of message:
Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent

after that nothing happen no event call process stooped

philips77, 2017-11-29:
Exactly. Now the lib wait until it gets CBCentralManagerDelegate call peripheralDidDisconnected which would resume the process. So the question is why it didn't get this event.
1. The device didn't disconnect
2. The delegate has been changed to a different object before the update finished (with completed or error)
3. I can't see any other reason now

mostafaberg, 2017-12-01:
@VikramSinghG I took a look at your project but we can't do so much with it since the library that handles the communication with the smart band is an archive and we can't break through the code, so far seems like when we connect to any peripheral it sill scan services then disconnect..

here is the logs i get so far when trying to connect to a peripheral of ours.

```
2017-12-01 15:38:47.019857+0100 CB1_Demo[2262:983861] 连接外围设备成功!
2017-12-01 15:38:47.472057+0100 CB1_Demo[2262:983861] service     EF680100-9B35-4933-9B10-52FFA9740042
2017-12-01 15:38:47.472951+0100 CB1_Demo[2262:983861] service     EF680200-9B35-4933-9B10-52FFA9740042
2017-12-01 15:38:47.473678+0100 CB1_Demo[2262:983861] service     EF680400-9B35-4933-9B10-52FFA9740042
2017-12-01 15:38:47.474315+0100 CB1_Demo[2262:983861] service     EF680300-9B35-4933-9B10-52FFA9740042
2017-12-01 15:38:47.475417+0100 CB1_Demo[2262:983861] service     EF680500-9B35-4933-9B10-52FFA9740042
2017-12-01 15:38:47.476158+0100 CB1_Demo[2262:983861] service     Battery
2017-12-01 15:38:47.476636+0100 CB1_Demo[2262:983861] service     FE59
2017-12-01 15:38:47.477868+0100 CB1_Demo[2262:983861] -----外设断开连接------(null)
```

Do you know what is causing the last disconnect ?
`2017-12-01 15:38:47.477868+0100 CB1_Demo[2262:983861] -----外设断开连接------(null)`

if you have an example firmware that simulates your arm band that will run on one of our NRF51 or nRF52 DKs that will help me a lot .

Thanks

VikramSinghG, 2017-12-04:
@mostafaberg In my case band is disconnecting while process starts and nRFLib writes to band
 peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Printing description of message:
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Printing description of message:
Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent

than nothing happen i have uploaded the firmware below which is working fine with my android code and nRFToolbox app(android and ios) please look into this while you get time .

[a12_dfu_app_v11b.zip](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/1525742/a12_dfu_app_v11b.zip)

mostafaberg, 2017-12-05:
@VikramSinghG I tried flashing your Firmware on one of our NRF52 DKs, flash process completes, but the DK no longer works after that.. might be crashing on startup ? do you know which DK this firmware can run on without issues ?

VikramSinghG, 2017-12-05:
Yes but Firmware is compatible with nRF51822 as tested on nRFToolbox App

mostafaberg, 2017-12-05:
i receive an error (Data size exceeds limit) when trying to flash it OTA with an nRF51822, strange, do you have just the HEX file(s) for the app/softdevice perhaps ? that will help me check it out faster :) sorry for all the delays

VikramSinghG, 2017-12-05:
@mostafaberg  Thanks for your support . I do have zip with bin and dat file .and nRF Lib provide Method **initWithUrlToBinOrHexFile:** to use bin file. while bin and dat file can be taken by extracting zip. and Dk(110 version) is supported .

mostafaberg, 2017-12-05:
This won't work either in my case, I would need the actual hex files unfortunately, do you have the possibility of sending me those instead ?

mostafaberg, 2017-12-05:
Ok seems like issue is i had an unsupported Softdevice, using SDK8.0 and the flashing goes well, however, the peripheral has LED 1,2 and 3 powered on and is not advertising, any tips ?

zhoushenglin, 2018-06-03:
Hi ,Is this problem solved? How to solve?

closed 2018-10-09 by mostafaberg