nordicsemi/IOS-DFU-Library #167: Bytes lost when retrying firmware update

Michael-Bischof, 2017-11-24:
I am currently struggling with error code 308 (bytesLost). The following steps lead to said error:

1. Start APP firmware update
2. Abort while updating (using DFUServiceController.abort())
3. Wait until device is advertising normally again
4. Retry installing same file 

After step 4. I get the following console output:
`Data written to 8EC90001-F315-4F60-XXXX-838830DXXXXX`
`Notification received from 8EC90001-F315-4F60-XXXX-838830DXXXXX, value (0x): 600101`
`Data object 1/22 created`
`Application: Uploading firmware...`
`Sending firmware to DFU Packet characteristic...`
`dfuProgressDidChange(for:outOf:to:currentSpeedBytesPerSecond:avgSpeedBytesPerSecond:) > [1, 1, 0, 0.0, 0.0]`
`Disconnecting...`
`dfuError(_:didOccurWithMessage:) > 480 bytes were sent while 240 bytes were reported as received`

After killing my app and starting it again, the update works just fine.
Using the nRF Connect App also works.

mostafaberg, 2017-11-24:
Sorry to hear that you're experiencing an issue, to be clear, when you say after killing my app and starting again, do you mean the peripheral's app or the actual mobile app ?

Michael-Bischof, 2017-11-24:
I have to kill the app on my smartphone that is using the iOSDFULibrary. The device I am trying to connect to is not restarted. 

After my mobile app failed to update I can simply open the app switcher, kill the app, start it again and the update works.

mostafaberg, 2017-11-24:
I see, this is a strange issue, and from your explanation i understood that you do not need to kill the app when you do the same procedure in nRF Connect and nRF Toolbox ?

Michael-Bischof, 2017-11-24:
What I mean is, that aborting the update and retrying it works perfectly fine when using the nRF Connect app instead. That is why I thought the issue is probably not within the firmware itself, but a bug in the framework.

If the app fails to update, I can simply switch over to the nRF Connect app and do the update there without problems.

Thanks for your quick response by the way!

mostafaberg, 2017-11-24:
You're welcome :) it's our duty 👍 
One more question, does it happen every single time with those exact sent/reported numbers ? or is it intermittent ? Trying to rule out where the issue stems from, what's a bit confusing is that this library is the exact same  one used in the nRF Connect app, so the errors should be exactly the same

is it possible to also see a code snippet where you are initialising the DFU Lib and starting the DFU process ?

Michael-Bischof, 2017-11-24:
Oh, ok I just tried again to check if the numbers vary (they do) and got the nRF Connect app to fail too.

I just tried again twice in a row in my app and got
`dfuError(_:didOccurWithMessage:) > 5056 bytes were sent while 4336 bytes were reported as received`
both times

Without killing my app, I then switched to the nRF Connect app and tried to install the same firmware. Got stuck at "Connecting" with Update: 0%. The Device I am using indicates that it is connected and the ios settings suggest the same.

mostafaberg, 2017-11-24:
That's a good thing that they're consistent, if you don't disconnect the app the peripheral will remain connected and the other apps won't be able to connect to it.
can you try (while in this state) to kill the currently connected app and then reconnect using nRF Connect to see if it reproduces there ?

I'm assuming you are using the latest version of the DFU Library ?

Michael-Bischof, 2017-11-24:
Yes, I am using 4.0.3 in my mobile app. 
The nRF Connect app had the connection to the device. After killing the nRF app (and keeping my app alive), i started up the nRF app again and the update was successfull. After that, I switched back to my app (which was alive the whole time) and got the error again. 

It seems like the error will keep on occurring until I restart the app.

mostafaberg, 2017-11-24:
will bring in @philips77 to dig into this with me, any thoughts what could be wrong here ? should we simply restart the process from the previous stored page or start from scratch if this happens ?

philips77, 2017-11-24:
Hi @Michael-Bischof,
Could you post the whole log you get when the update fails, not only the error?
I'm interested in when the numbers start to split.
In theory you should be able to resume update from exactly where you aborted.
Thanks!

Michael-Bischof, 2017-11-24:
```
 13:18:11.800713+0100 [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:97] resetToDFUMode(for:completionHandler:) > resetToDFU
 13:18:11.801097+0100 [BluetoothScanner+UART+Protobuf.swift:57] buildSerializedProtobuf(for:parameter:) > command: resetdevice; parameter: bootloader
 13:18:11.801767+0100 [BluetoothScanner+UART+Protobuf.swift:167] buildSerializedProtobuf(for:parameter:) > protobufMessage; msg: Optional(XXXXBLETools.FMP_Base.OneOf_Msg.deviceResetRequest(XXXXBLETools.FMP_DeviceReset:
type: RESET_TO_BOOTLOADER
))
 13:18:11.802960+0100 [BluetoothScanner+UART+Protobuf.swift:183] buildSerializedProtobuf(for:parameter:) > payload (raw): 01A206020801409F
 13:18:11.803818+0100 [BluetoothScanner+UART+Protobuf.swift:184] buildSerializedProtobuf(for:parameter:) > payload (cobs): 0901A206020801409F00
 13:18:11.804698+0100 [BluetoothScanner+UART+Protobuf.swift:41] writeMessageToUART(_:for:) > write message to UART
 13:18:11.806753+0100 [BluetoothScanner.swift:494] processCapturedDevices() > isConnectable by state: false (true, true, true); appState: foreground
 13:18:11.873516+0100 [BluetoothScanner.swift:494] processCapturedDevices() > isConnectable by state: false (true, true, true); appState: foreground
 13:18:11.892521+0100 [FirmwareUpdateViewController.swift:312] setupBindings() > file:///var/mobile/Containers/Data/Application/XXXX-D691-4975-B9F6-C02FD6027849/Documents/DFU-APP-10042488-XXXXXXXX-SD158-BL8-APP11-P5413-J20771-XXXX_DK.zip
 13:18:11.933120+0100 [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:184] uploadFirmware(fileURL:to:) > identifier: XXXX-5A3A-B1F9-0267-98CCCC767E71; state: disconnected;
 13:18:11.935748+0100 [BluetoothScanner.swift:217] updateCapturedDevices(with:advertisementData:rssi:) > known device in DFU mode; connectable: false (1101)
 13:18:12.547113+0100 [BluetoothScanner+CBCentralManagerDelegate.swift:89] centralManager(_:didDisconnectPeripheral:error:) > uuid: XXXX-628D-28A9-8370-E36D20EB260B; The connection has timed out unexpectedly.
 13:18:12.979591+0100 [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:297] logWith(_:message:) > Connecting to DFU XXXX I...
 13:18:12.979993+0100 [11728:3039687] [Debug] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:295] logWith(_:message:) > centralManager.connect(peripheral, options: nil)
 13:18:12.981214+0100 [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:248] dfuStateDidChange(to:) > Connecting
 13:18:12.982134+0100 [FirmwareUpdateViewController.swift:329] setupBindings() > pauseRunningFirmwareUpdate
 13:18:13.285753+0100 [11728:3039687] [Debug] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:295] logWith(_:message:) > [Callback] Central Manager did connect peripheral
 13:18:13.286185+0100 [11728:3039687] [Info] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:299] logWith(_:message:) > Connected to DFU XXXX I
 13:18:13.286492+0100 [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:297] logWith(_:message:) > Discovering services...
 13:18:13.286785+0100 [11728:3039687] [Debug] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:295] logWith(_:message:) > peripheral.discoverServices(nil)
 13:18:13.585210+0100 [11728:3039687] [Info] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:299] logWith(_:message:) > Services discovered
 13:18:13.585631+0100 [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:297] logWith(_:message:) > Starting Secure DFU...
 13:18:13.586509+0100 [11728:3039687] [Info] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:299] logWith(_:message:) > Connected to DFU XXXX I
 13:18:13.586834+0100 [11728:3039687] [Info] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:299] logWith(_:message:) > Services discovered
 13:18:13.587188+0100 [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:297] logWith(_:message:) > Secure DFU Service found
 13:18:13.588012+0100 [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:297] logWith(_:message:) > Discovering characteristics in DFU Service...
 13:18:13.588397+0100 [11728:3039687] [Debug] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:295] logWith(_:message:) > peripheral.discoverCharacteristics(nil, for: FE59)
 13:18:13.765691+0100 [11728:3039687] [Info] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:299] logWith(_:message:) > DFU characteristics discovered
 13:18:13.766240+0100 [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:297] logWith(_:message:) > Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
 13:18:13.766582+0100 [11728:3039687] [Debug] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:295] logWith(_:message:) > peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
 13:18:13.767993+0100 [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:248] dfuStateDidChange(to:) > Starting
 13:18:13.768898+0100 [FirmwareUpdateViewController.swift:329] setupBindings() > pauseRunningFirmwareUpdate
 13:18:13.769767+0100 [FirmwareUpdateViewController.swift:332] setupBindings() > resumeRunningFirmwareUpdate
 13:18:13.944406+0100 [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:297] logWith(_:message:) > Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
 13:18:13.944787+0100 [11728:3039687] [Info] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:301] logWith(_:message:) > Application: Secure DFU Control Point notifications enabled
[... cut ...]
 13:18:14.489032+0100 [11728:3039687] [Debug] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:295] logWith(_:message:) > [Callback] Central Manager did disconnect peripheral
 13:18:14.489374+0100 [11728:3039687] [Info] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:299] logWith(_:message:) > Disconnected
 13:18:14.490156+0100 [11728:3039687] [Error] [main] [BluetoothScanner+DeviceActions+FirmwareUpdate.swift:265] dfuError(_:didOccurWithMessage:) > 8912 bytes were sent while 8432 bytes were reported as received
```

philips77, 2017-11-24:
It looks like something is wrong with PRNs. You send 3 times 12*20 bytes (8192 + 3*12*20 = 8912) and it looks like the 2 PRNs you received in between were OK, but the last one contained received bytes value 8432 (=8192 + 1*12*20). Could you also print number of bytes received before this https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/SecureDFU/Characteristics/SecureDFUControlPoint.swift#L469 line?

Michael-Bischof, 2017-11-26:
I figured it out! There is no problem with your Framework, it was my mistake. 

A bug in my app called dfuController?.pause() and dfuController?.resume() multiple times in a row while the update was being installed. This only happened if I canceled the update before. 

The log even said pauseRunningFirmwareUpdate and resumeRunningFirmwareUpdate a few times but I didn't give it any attention. 

I have no idea how and why I got the nRF Connect to crash too. Please forgive me for wasting your time. Thank you so much for the fast and kind support. You guys are awesome

philips77, 2017-11-26:
Great to hear that. But still there is a bug if pausing and resuming in random moments causes this issue. I guess that's because pause works only on during uploading while resume may... start uploading before it should? I'll have a look at it.

closed 2018-10-09 by mostafaberg