nordicsemi/IOS-DFU-Library #168: The problem of upgrading immediately after the connection is successful.

developerXiong, 2017-12-05:
I'm sorry to trouble you again!
I met a problem like this title, The method of calling the DFU upgrade immediately after I connect to the Bluetooth device will give me the wrong 'can only accept commands while in the connected state', But when I pause for 2 seconds after connecting the Bluetooth device, the method of calling the DFU upgrade will be successful.

developerXiong, 2017-12-05:
In fact, 1s is enough.

mostafaberg, 2017-12-05:
Good morning, seems like your DFU code is starting to execute before the connection is actually established, you might doing things on a different thread and seeing a race condition. without seeing actual code I would say that this is the most possible issue

developerXiong, 2017-12-06:
`- (void)manager:(QmjkDeviceManager *)manager connectPeripheralIsSuccess:(BOOL)isSuccess {
    if (isSuccess) {
        dispatch_after(dispatch_time(DISPATCH_TIME_NOW, (int64_t)(1 * NSEC_PER_SEC)), dispatch_get_main_queue(), ^{
            [self OTA];
        });
    } else {
        
    }
}`

I execute the DFU code when receive the callback that connect device successful.

mostafaberg, 2017-12-06:
Can you breakpoint and see when and what is calling the `connectPeripheralIsSuccess` method ? this might be getting fired too early or before the peripheral is actually connected.

closed 2018-10-09 by mostafaberg