nordicsemi/IOS-DFU-Library #170: newAddressExpected causing issues

khoogheem, 2018-01-08:
Right now newAddressExpected is set to internal which causes issues with older SDKs where the Name changes when going into DFU.  

We have had to modify the Lib to set it to true right before jumping to bootloader.

Would be nice if when setting up the DFUServiceInitiator we could let it know that Names will change.

philips77, 2018-01-08:
Whether the name will it will not change depends on the result of 0x02-len-name command. If status 01 is received, this will be true, otherwise false. No need to touch it. You may only forcefully disable changing the name, but that's not recommended.

newAddressExpected also depends on what dfu buttonless service is used and its value is set automatically.

Why do you need to touch any of them? Do you have a custom DFU? Do you have some path I didn't think during implementation? Could you give more details?

khoogheem, 2018-01-08:
I should have stated this is a product using SDK 6ish.. So old legacy DFU.  The Peripheral has a name "Sensor" (lets day) and the dfu is DfuTarg..  Without setting newAddressExpected = true the dfu fails.

khoogheem, 2018-01-08:
and if you look at 
```
    /// This flag must be set to true if the device will advertise with a new device address after it resets.
    /// The service will scan and use specified peripheral selector in order to connect to the new peripheral.
    internal var newAddressExpected  : Bool = false
```

it states that it must be set to true.. but the code never sets it to true.. which makes me think that at some point the intent was to make this public

philips77, 2018-01-08:
No. It should rather set this value to true automatically based on the information it got from the buttonless service. I'll have a look at it tomorrow. Thank you for reporting the issue.

philips77, 2018-01-12:
Hello again, sorry for the delay. I guess this issue is related to https://github.com/NordicSemiconductor/Android-DFU-Library/pull/45 and I will fix it accordingly. If there is no DFU Version characteristic it will set `newAddressExpected = true`. Hopefully it will work for you.

philips77, 2018-01-12:
Hi, I just made a PR with necessary change. Could you pull it to your project and verify it works before we will merger it? In my opinion it should solve your problem.
On Android it has been fixed by this commit: https://github.com/NordicSemiconductor/Android-DFU-Library/commit/1c18dd8c2e024376faa6fda014c74cd85fef7aa0

khoogheem, 2018-01-12:
> Hello again, sorry for the delay. I guess this issue is related to NordicSemiconductor/Android-DFU-Library#45 <https://github.com/NordicSemiconductor/Android-DFU-Library/pull/45> and I will fix it accordingly. If there is no DFU Version characteristic it will set newAddressExpected = true. Hopefully it will work for you.

Thanks we will give it a shot today!