nordicsemi/IOS-DFU-Library #171: DFU update fails without error message

thilosuelberg, 2018-01-11:
Hi,

I am running SDK 12.3.0 on a 51 series chip with s130. The boot loader is configured and installed as outlined here: https://devzone.nordicsemi.com/blogs/1085/getting-started-with-nordics-secure-dfu-bootloader/ . If I flash my application along with the correct CRC checksum, everything works as expected. The boot loader starts when a pushbutton is being pressed during power-on, otherwise my main program is being started. So far so good.

When I try to update the main program using DFU, the NRF toolbox app does not show any progress or error message. The console log looks like this:

Verbose: Connecting to DfuTarg...
Debug: centralManager.connect(peripheral, options: nil)
Debug: [Callback] Central Manager did connect peripheral
Info: Connected to DfuTarg
Verbose: Discovering services...
Debug: peripheral.discoverServices(nil)
Info: Services discovered
Verbose: Starting Secure DFU...
Info: Connected to DfuTarg
Info: Services discovered
Verbose: Secure DFU Service found
Verbose: Discovering characteristics in DFU Service...
Debug: peripheral.discoverCharacteristics(nil, for: FE59)
Info: DFU characteristics discovered
Verbose: Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
Verbose: Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
Application: Secure DFU Control Point notifications enabled
Verbose: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Info: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Info: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60060100010000870000005a6c65e3
Application: Command object info (Max size = 256, Offset = 135, CRC = E3656C5A) received
Application: Received CRC match Init packet
Verbose: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Info: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Info: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60040a
Error: Error 10: Operation failed
Verbose: Disconnecting...
Debug: centralManager.cancelPeripheralConnection(peripheral)
Debug: [Callback] Central Manager did disconnect peripheral
Info: Disconnected

The .zip file has been built using:

	nrfutil pkg generate --hw-version 51 --application-version 1 --application $(OUTPUT_DIRECTORY)/nrf51822_xxac.hex --sd-req 0x98 --key-file $(SDK_ROOT)/examples/dfu/bootloader_secure/private.key $(OUTPUT_DIRECTORY)/vend_dex_fw.zip

Do you have an idea what is going wrong here?
Thanks for your help in advance,
Best Regards,
   Thilo

philips77, 2018-01-11:
It fails on validating the init packet. That means one of the following reasons:
- invalid private key: check if the private key in Bootloader matches your private key, out generate the C file with it again using nrfutil
- wrong --sd-req value: check the ir of your current SD
- app version to small: if you already did dfu once with application version 2+ it has to be higher or equal to the previous one

philips77, 2018-01-11:
On the nrfutil GitHub page you'll find list of valid SD IDs. 0x98 is s132_nrf52_4.0.2, which is not yours. If you are using: s130_nrf51_2.0.1 try 0x87 instead.

philips77, 2018-01-11:
And the link to page I mentioned: https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/README.md

thilosuelberg, 2018-01-11:
Hi Aleksander,

Thank you very very much, after changing the ID for the SoftDevice it works as expected. Stupid me :-)
The support you and your colleagues provide is awesome. I really appreciate that.

Best Regards,
   Thilo

closed 2018-01-11 by thilosuelberg

philips77, 2018-01-11:
Really glad I could help! Good luck with the project!