nordicsemi/IOS-DFU-Library #177: DFU fails after 2-10% written

jan-ekholm, 2018-01-30:
I'm using the newest CocoaPod version with Swift 4 on iOS11. I have an issue where the updating seems to always fail after 2-10%. No update has completed yet and I've tried a few. Two different firmwares are used, which may account for the variations in progress before bailing out.

My code is basically:

let dfuInitiator = DFUServiceInitiator(centralManager: self.bluetoothManager.central, target: dfuPeripheral)
dfuInitiator.delegate = self
dfuInitiator.progressDelegate = self
dfuInitiator.logger = self
dfuInitiator.packetReceiptNotificationParameter = 1 // tested all from 1..12
dfuInitiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = false
dfuInitiator.forceDfu = false
self.dfuController = dfuInitiator.with(firmware: firmware).start()

Before that I've put the device in DFU mode and waited for it to advertise itself as a "device in DFU mode", ie the "XXXX" name as can be seen in the log. That's the "dfuPeripheral". 

This log below is basically just printing out everything that comes to the delegate methods. Any quick hint as to what may be going wrong, maybe there's some clear red light that I'm just not seeing (I don't really know much about low level BLE).

logWith: Connecting to XXXX...
logWith: centralManager.connect(peripheral, options: nil)
logWith: [Callback] Central Manager did connect peripheral
logWith: Connected to XXXX
logWith: Discovering services...
logWith: peripheral.discoverServices(nil)
logWith: Services discovered
logWith: Starting Secure DFU...
logWith: Connected to XXXX
logWith: Services discovered
logWith: Secure DFU Service found
logWith: Discovering characteristics in DFU Service...
logWith: peripheral.discoverCharacteristics(nil, for: FE59)
logWith: DFU characteristics discovered
dfuStateDidChange: new state: Starting
logWith: Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
logWith: peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
logWith: Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
logWith: Secure DFU Control Point notifications enabled
logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
logWith: peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
logWith: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
logWith: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60060100010000870000009a3a7634
logWith: Command object info (Max size = 256, Offset = 135, CRC = 34763A9A) received
logWith: Received CRC match Init packet
logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
logWith: peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
logWith: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
logWith: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600401
logWith: Command object executed
logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
logWith: peripheral.writeValue(0x0602, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
logWith: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
logWith: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 6006010010000000100000835d065f
logWith: Data object info (Max size = 4096, Offset = 4096, CRC = 5F065D83) received
dfuStateDidChange: new state: Uploading
logWith: 4096 bytes of data sent before, CRC match
logWith: Resuming uploading firmware...
logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
logWith: peripheral.writeValue(0x020100, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
logWith: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
logWith: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
logWith: Packet Receipt Notif enabled (Op Code = 2, Value = 1)
logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
logWith: peripheral.writeValue(0x010200100000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
logWith: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
logWith: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
logWith: Data object 2/27 created
logWith: Uploading firmware...
logWith: Sending firmware to DFU Packet characteristic...
dfuProgressDidChange: part 1 of 1, progress: 0 %, speed: 0.0 bytes/s, average speed: 0.0 bytes/s
dfuProgressDidChange: part 1 of 1, progress: 3 %, speed: 37101.3180008846 bytes/s, average speed: 37101.3180008846 bytes/s
dfuProgressDidChange: part 1 of 1, progress: 4 %, speed: 301.688725075274 bytes/s, average speed: 323.050285594573 bytes/s
logWith: [Callback] Central Manager did disconnect peripheral
logWith: Disconnected by the remote device
dfuError: DFU error: DFUError, message: Device disconnected unexpectedly

jan-ekholm, 2018-01-30:
Noteworthy here is that I can update the devices using the "nrf Connect" iOS application. I just have to set the device into DFU mode and then switch to "nrf Connect", select the device and the update finishes ok. So the devices can be updated and the firmwares are ok.

philips77, 2018-01-30:
I see one possible issue: after
`logWith: 4096 bytes of data sent before, CRC match` there should be Execute command, just to be sure the first object was executed. I don't know why there isn't.
Does it fail also when you do a fresh update, not a resume? Could you paste logs?

jan-ekholm, 2018-01-30:
I'm not sure what you mean with a fresh update vs a resume. All the logging comes after
```dfuInitiator.with(firmware: firmware).start()```
and our code doesn't do anything after that (except log whatever comes to the delegate methods).

philips77, 2018-01-30:
When you do an update the lib initially requests the CRC of an init packet that might have been saved as a result of not finished DFU. If offset > 0 and the CRC matches the CRC of the init data that are about to be sent, it just executes it (just to be sure as the previous dfu might have stopped after sending, but before executing the init packet. Double exe does nothing.
If offset == 0 it sends init packet, request CRC for verification and executes.
Then it does the same with the data object. Asks for offset, max size and CRC, and depending on the result starts sending from scratch, continues the last object or executes it, if a whole one has been sent again and offset == n * max size.
In your case, the logs say, that we have the last case. You did try dfu before, init packet was sent, crc is the same, 4096 bytes of the first data object were sent and now they should be executed before creating next data object. But it looks like the lib just sets PRNs to 1 and creates new object (result of creating this object without exe is undefined, as if the previous object wasn't executed, it will be erased and dfu Target will wait for it again (let's say crc wasn't exact). If it was, Target will wait for next object.

Nevertheless, it should not disconnect as it does in your case. The disconnection may be a different issue.

jan-ekholm, 2018-01-31:
Thank you for the explanation. We have not yet found out why the device disconnects, but our device firmware guy will do some more testing today. The firmware (or only the DFU part, I'm not exactly sure) is based on the Nordic Semiconductor sample code.

But based on your resume/fresh comment, then the first DFU trial should have finished ok, as it was done with no previous DFU data? I do of course not have the logs anymore from that first run.

philips77, 2018-01-31:
Well... That's what I wanted to check :)
Does it now also fail when you user nRF Connect? If not, try updating with nRF Connect and then again with your app, then it will be a fresh update as the last one ended successfully.

jan-ekholm, 2018-01-31:
So I've done some testing today, and my previous update came to 11%. Then I tried it again, mostly to get the device into DFU mode and I see that it did resume:

```
2018-01-31 12:05:30.7840 debug FirmwareUpdateManager.dfuStateDidChange: new state: Connecting
2018-01-31 12:05:30.7860 debug FirmwareUpdateManager.logWith: Connecting to XXXX...
2018-01-31 12:05:30.7870 debug FirmwareUpdateManager.logWith: centralManager.connect(peripheral, options: nil)
2018-01-31 12:05:31.0000 debug FirmwareUpdateManager.logWith: [Callback] Central Manager did connect peripheral
2018-01-31 12:05:31.0000 info  FirmwareUpdateManager.logWith: Connected to XXXX
2018-01-31 12:05:31.0010 debug FirmwareUpdateManager.logWith: Discovering services...
2018-01-31 12:05:31.0010 debug FirmwareUpdateManager.logWith: peripheral.discoverServices(nil)
2018-01-31 12:05:31.1200 info  FirmwareUpdateManager.logWith: Services discovered
2018-01-31 12:05:31.1230 debug FirmwareUpdateManager.logWith: Starting Secure DFU...
2018-01-31 12:05:31.1270 info  FirmwareUpdateManager.logWith: Connected to XXXX
2018-01-31 12:05:31.1280 info  FirmwareUpdateManager.logWith: Services discovered
2018-01-31 12:05:31.1280 debug FirmwareUpdateManager.logWith: Secure DFU Service found
2018-01-31 12:05:31.1300 debug FirmwareUpdateManager.logWith: Discovering characteristics in DFU Service...
2018-01-31 12:05:31.1310 debug FirmwareUpdateManager.logWith: peripheral.discoverCharacteristics(nil, for: FE59)
2018-01-31 12:05:31.1360 info  FirmwareUpdateManager.logWith: DFU characteristics discovered
2018-01-31 12:05:31.1410 debug FirmwareUpdateManager.dfuStateDidChange: new state: Starting
2018-01-31 12:05:31.1410 debug FirmwareUpdateManager.logWith: Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:05:31.1420 debug FirmwareUpdateManager.logWith: peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
2018-01-31 12:05:31.1790 debug FirmwareUpdateManager.logWith: Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-01-31 12:05:31.1790 info  FirmwareUpdateManager.logWith: Secure DFU Control Point notifications enabled
2018-01-31 12:05:31.1810 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:05:31.1830 debug FirmwareUpdateManager.logWith: peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-01-31 12:05:31.2380 info  FirmwareUpdateManager.logWith: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-01-31 12:05:31.2400 info  FirmwareUpdateManager.logWith: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60060100010000870000009a3a7634
2018-01-31 12:05:31.2430 info  FirmwareUpdateManager.logWith: Command object info (Max size = 256, Offset = 135, CRC = 34763A9A) received
2018-01-31 12:05:31.2450 info  FirmwareUpdateManager.logWith: Received CRC match Init packet
2018-01-31 12:05:31.2450 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:05:31.2470 debug FirmwareUpdateManager.logWith: peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-01-31 12:05:31.2980 info  FirmwareUpdateManager.logWith: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-01-31 12:05:31.2990 info  FirmwareUpdateManager.logWith: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600401
2018-01-31 12:05:31.3000 info  FirmwareUpdateManager.logWith: Command object executed
2018-01-31 12:05:31.3020 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:05:31.3030 debug FirmwareUpdateManager.logWith: peripheral.writeValue(0x0602, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-01-31 12:05:31.3590 info  FirmwareUpdateManager.logWith: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-01-31 12:05:31.3620 info  FirmwareUpdateManager.logWith: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601001000000030000006182697
2018-01-31 12:05:31.3620 info  FirmwareUpdateManager.logWith: Data object info (Max size = 4096, Offset = 12288, CRC = 97261806) received
2018-01-31 12:05:31.3630 debug FirmwareUpdateManager.dfuStateDidChange: new state: Uploading
2018-01-31 12:05:31.4070 info  FirmwareUpdateManager.logWith: 12288 bytes of data sent before, CRC match
2018-01-31 12:05:31.4070 info  FirmwareUpdateManager.logWith: Resuming uploading firmware...
2018-01-31 12:05:31.4080 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:05:31.4100 debug FirmwareUpdateManager.logWith: peripheral.writeValue(0x020c00, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-01-31 12:05:31.4490 info  FirmwareUpdateManager.logWith: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-01-31 12:05:31.4500 info  FirmwareUpdateManager.logWith: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
2018-01-31 12:05:31.4510 info  FirmwareUpdateManager.logWith: Packet Receipt Notif enabled (Op Code = 2, Value = 12)
2018-01-31 12:05:31.4520 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
[... cut ...]
2018-01-31 12:05:34.1090 error FirmwareUpdateManager.dfuError: DFU error: DFUError, message: Device disconnected unexpectedly
```

Here we see that it did start from 11% and advanced a bit before giving up as usual. Now, immediately after this failure, with the device still in DFU mode, I switched to "nrf Toolbox" where I had preselected the exact same firmware, selected the device and uploaded the firmware without issues. I saw that the progress jumped straight from 0% to something like 15%, which would mean that the update was resumed. I will now do another update in our app.

jan-ekholm, 2018-01-31:
Trying to update to another firmware now with the device being "fresh", I get the below. All tests with the same device and same iPhone 7, iOS11.

``` 
2018-01-31 12:16:04.2590 debug FirmwareUpdateManager.dfuStateDidChange: new state: Connecting
2018-01-31 12:16:04.2610 debug FirmwareUpdateManager.logWith: Connecting to XXX...
2018-01-31 12:16:04.2620 debug FirmwareUpdateManager.logWith: centralManager.connect(peripheral, options: nil)
2018-01-31 12:16:04.5270 debug FirmwareUpdateManager.logWith: [Callback] Central Manager did connect peripheral
2018-01-31 12:16:04.5270 info  FirmwareUpdateManager.logWith: Connected to XXXX
2018-01-31 12:16:04.5280 debug FirmwareUpdateManager.logWith: Discovering services...
2018-01-31 12:16:04.5280 debug FirmwareUpdateManager.logWith: peripheral.discoverServices(nil)
2018-01-31 12:16:04.6480 info  FirmwareUpdateManager.logWith: Services discovered
2018-01-31 12:16:04.6510 debug FirmwareUpdateManager.logWith: Starting Secure DFU...
2018-01-31 12:16:04.6540 info  FirmwareUpdateManager.logWith: Connected to XXXX
2018-01-31 12:16:04.6550 info  FirmwareUpdateManager.logWith: Services discovered
2018-01-31 12:16:04.6560 debug FirmwareUpdateManager.logWith: Secure DFU Service found
2018-01-31 12:16:04.6580 debug FirmwareUpdateManager.logWith: Discovering characteristics in DFU Service...
2018-01-31 12:16:04.6580 debug FirmwareUpdateManager.logWith: peripheral.discoverCharacteristics(nil, for: FE59)
2018-01-31 12:16:04.6640 info  FirmwareUpdateManager.logWith: DFU characteristics discovered
2018-01-31 12:16:04.6690 debug FirmwareUpdateManager.dfuStateDidChange: new state: Starting
2018-01-31 12:16:04.6700 debug FirmwareUpdateManager.logWith: Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:16:04.6710 debug FirmwareUpdateManager.logWith: peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
2018-01-31 12:16:04.7370 debug FirmwareUpdateManager.logWith: Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-01-31 12:16:04.7370 info  FirmwareUpdateManager.logWith: Secure DFU Control Point notifications enabled
2018-01-31 12:16:04.7390 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:16:04.7400 debug FirmwareUpdateManager.logWith: peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-01-31 12:16:04.7970 info  FirmwareUpdateManager.logWith: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-01-31 12:16:04.8280 info  FirmwareUpdateManager.logWith: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601000100000000000000000000
2018-01-31 12:16:04.8300 info  FirmwareUpdateManager.logWith: Command object info (Max size = 256, Offset = 0, CRC = 00000000) received
2018-01-31 12:16:04.8310 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:16:04.8330 debug FirmwareUpdateManager.logWith: peripheral.writeValue(0x010187000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-01-31 12:16:04.8860 info  FirmwareUpdateManager.logWith: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-01-31 12:16:04.8870 info  FirmwareUpdateManager.logWith: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
2018-01-31 12:16:04.8880 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:16:04.8900 debug FirmwareUpdateManager.logWith: peripheral.writeValue(0x020000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-01-31 12:16:04.9480 info  FirmwareUpdateManager.logWith: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-01-31 12:16:04.9490 info  FirmwareUpdateManager.logWith: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
2018-01-31 12:16:04.9490 info  FirmwareUpdateManager.logWith: Packet Receipt Notif disabled (Op Code = 2, Value = 0)
2018-01-31 12:16:04.9500 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:16:04.9520 debug FirmwareUpdateManager.logWith: peripheral.writeValue(0x1284010a3e0801123a080110341a029101200028, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
2018-01-31 12:16:04.9530 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:16:04.9550 debug FirmwareUpdateManager.logWith: peripheral.writeValue(0x00300038cce806422408031220df6994a045c7b8, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
2018-01-31 12:16:04.9580 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:16:04.9590 debug FirmwareUpdateManager.logWith: peripheral.writeValue(0xd5a73719ef0c387c693b3bc8bc447a0911661546, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
2018-01-31 12:16:04.9610 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
2018-01-31 12:16:04.9620 debug FirmwareUpdateManager.logWith: peripheral.writeValue(0xbbbea6cd57480010001a401d59f38d5de9972cb4, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
2018-01-31 12:16:04.9640 debug FirmwareUpdateManager.logWith: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
[... cut ...]
2018-01-31 12:16:07.6050 debug FirmwareUpdateManager.logWith: [Callback] Central Manager did disconnect peripheral
2018-01-31 12:16:07.6050 info  FirmwareUpdateManager.logWith: Disconnected by the remote device
2018-01-31 12:16:07.6060 error FirmwareUpdateManager.dfuError: DFU error: DFUError, message: Device disconnected unexpectedly
``` 

The difference here from my tests yesterday are:

``` 
            dfuInitiator.packetReceiptNotificationParameter = 12
            dfuInitiator.forceDfu = true
```

philips77, 2018-01-31:
And you say it works well with nRF Toolbox or nRF Connect... Are you sure you don't have another connection to this device in your app, or you aren't interrupting it in any way? For example by releasing CBCentralManager, disconnecting CBPeripheral in your app, or dismissing a view? All of those shouldn't matter, but.. looks like it may be a stupid mistake.

jan-ekholm, 2018-02-01:
It probably is a silly mistake, I have no doubt about that. I'm keeping the `CBCentralManager` alive, it's just temporarily borrowed to the `DFUServiceInitiator`. As the initiator takes over the delegate my old code doesn't react to anything while the updating is in progress. I also do not yet have the disconnect logic in place, so nothing disconnects it. As it always seems to happen a few seconds after the process starts I can't see anything that I have that would run after a few seconds.

But we are experimenting with some improvements to the bootloader code, assuming that perhaps it is the one that does something wrong.

mostafaberg, 2018-02-01:
Hi @jan-ekholm !

I noticed in your log that there's two lines that read

> 2018-01-31 12:16:06.7880 debug BackendConnection.process: checking saved uploadable motion data files
> 2018-01-31 12:16:06.7910 debug BackendConnection.upload: no uploadable files found

It seems like the disconnect happens after that, could there be anything related there ?

Also, can you try running it without borrowing the `CBCentralManager` ? maybe something happens there too.

You should only take back the delegates after you get a completion success or an error, but I'm not sure about the behaviour if you take the delegates in case the peripheral restarts when uploading multiple part DFU packages.

also can you confirm your `iOSDFULibrary` version from the `Podfile.lock` file ?

jan-ekholm, 2018-02-01:
Hm, that should not be related. It's just a background task that runs every 10s or so and checks a directory for generated files. It does not touch sensors in any way. If you look at my two other logs, there are no similar lines there. In this case it just happened to run which the update was in progress.

With the "borrowing" part I means that the `CBCentralManager` is normally in use by my own code which just *lends* it (English is hard) to the `DFUServiceInitiator`. I take back the delegate only after the process has completed or errored out.  I see no logging that there would be a completion of some first part, I log all state changes.

```
    @objc func dfuStateDidChange(to state: DFUState) {
        Log.debug("new state: \(state.description())")
        switch state {
        case .aborted, .completed:
            // restore the delegate when the state indicates completion of some form
            self.bluetoothManager.central.delegate = self.bluetoothManager
            self.dfuInitiator = nil
            self.dfuController = nil

        default:
            break
        }
    }
``` 

Podfile.lock:

```
PODS:
  - iOSDFULibrary (4.1.1):
    - Zip (~> 1.0)
  - Kronos (1.0.1)
  - Zip (1.1.0)

DEPENDENCIES:
  - iOSDFULibrary (~> 4.1.1)
  - Kronos (~> 1.0.1)
  - Zip (~> 1.1.0)
```

jan-ekholm, 2018-02-01:
Oh, and I do the same delegate restoring in `dfuError()`.

jan-ekholm, 2018-02-01:
The bootloader checks found nothing weird. But even if there was something silly in the bootloader code, it doesn't explain why "nrf Connect" manages to do the update 100% of the time on the same phone to the same device. If the bootloader was fubar it would also mess up "nrf Connect"'s attempts.

PR #191 (2018-04-20): Fixed update failures because of device disconnects
The problem this PR solves are "random" disconnect events during a firmware update. This happens because the lib is not checking if it is actually the to-be-updated peripheral that got disconnected. So if a completely unrelated device (or the same device with a different mac address) disconnects during the update, the update is aborted even though the target peripheral is still connected.

It's strange that this hasn't had a huge impact before (we may saw some cases without recognizing them), but I guess this may be related to changed timings [... cut]
closed 2018-04-20 by philips77