nordicsemi/IOS-DFU-Library #178: What to do if "service changed" characteristic is not included in Bootloader?

pvaibhav, 2018-01-31:
Hi all,

I am running into caching issue on iOS 11, where after DFU the phone caches the services and never reloads the actual application's services/chars. From some research I found that the 'service changed' characteristic must be included. Our application includes it, but the Bootloader does not.

We have used the bootloader source code included in the original/official SDK 7.1.0. After checking its main.c I found that the service changed char is turned OFF in it!: `#define IS_SRVC_CHANGED_CHARACT_PRESENT 0`

No we've got several thousand units of our product out in the field, which need a firmware update. But we can't do it (anymore) on iOS! The only solution is to ask the user to power their phone completely off and back on (turning Bluetooth off/on doesn't help).

I thought of updating the bootloader itself, since the app includes the DFU service. But found after trying that the DFU service in the app is only good for restarting the product into DFU mode. :-/  But once a device is booted into DFU, it gets "infected" and iOS won't properly discover updated services anymore.

What could be a solution?

Thanks.

philips77, 2018-01-31:
1. Is your device changing device address (+1) when in Bootloader mode? 
2. Is the dfu service on 3rd position in both Bootloader and app mode, so that handle numbers are the same?
3. Do you have an option to enable dfu mode with a button (i.e. keep button pressed and reset)? If yes, then it will increment device address.
4. Is Bootloader update available? I don't remember how was it in 7.1... guess yes.
5. Do you use bonding? Dfu with bonding wasn't really supported until SDK 8, afair.

If you have SC in the app mode, you can still jump to Bootloader mode "buttonlessly" and update the device with an app where you increment address by 2 (Bootloader may use +1 when switched with a button). Do you use random addresses or public, when it's not possible?

Ps. It may still be updated using Android.

philips77, 2018-01-31:
To be clear, as I understand, the issue is after DFU, when the device reboots into new app mode, but the iOS have cached Bootloader's services. Or when the first update fail and device reboots to old app mode and Bootloader's services are cached.
Only restarting phone can be then used to clear the cache.

As spec says, SC should always be enabled when you plan to change services. I really have no idea why was it disabled by default. Now not only we have SC there, but to be sure we increment address of no bonding.

pvaibhav, 2018-02-06:
Hello, thanks for your response and sorry for replying late.

> Is your device changing device address (+1) when in Bootloader mode?

I'm not sure, iOS exposes random Mac address to the app. I can check with another device if this info is important. I'm guessing the Mac address stays the same since otherwise the phone would not reuse the cache.

> Is the dfu service on 3rd position in both Bootloader and app mode, so that handle numbers are the same?

Nope I don't think so. The bootloader is compiled from nRF SDK 7.1 directly while the app is based on a customised fork of mbed.

> Do you have an option to enable dfu mode with a button (i.e. keep button pressed and reset)? If yes, then it will increment device address.

Oh I see. This is what we have (button-less DFU) since our product doesn't have any physical buttons. In this case I think the answer to your first question above would be yes.

> Is Bootloader update available? I don't remember how was it in 7.1... guess yes.

Unfortunately no, since we are using single-bank DFU. Our firmware won't fit otherwise (flight stabilisation algorithm... lots of math).

> Do you use bonding? Dfu with bonding wasn't really supported until SDK 8, afair.

Nope, no bonding.

> If you have SC in the app mode, you can still jump to Bootloader mode "buttonlessly" and update the device with an app where you increment address by 2 (Bootloader may use +1 when switched with a button). Do you use random addresses or public, when it's not possible?

Yes, the update succeeds. The problem is, once the device has booted into DFU *without* the SC, iOS will never run service rediscovery upon reboot. So the device is stuck in "dfu mode" according to the phone, though not in reality.

> To be clear, as I understand, the issue is after DFU, when the device reboots into new app mode, but the iOS have cached Bootloader's services. Or when the first update fail and device reboots to old app mode and Bootloader's services are cached.

The first case - after DFU when the device reboots into updated app, iOS still returns only bootloader's cached services.

> Only restarting phone can be then used to clear the cache.

I found that turning Bluetooth off/on in the Settings app (as opposed to control center) also works.

> As spec says, SC should always be enabled when you plan to change services. I really have no idea why was it disabled by default. Now not only we have SC there, but to be sure we increment address of no bonding.

Exactly what I wonder -- we used the nordic-supplied code not expecting that some time in the future apple will start caching aggressively, and that the SC would be the reason (we were not even aware of it in 2014). 😐

mostafaberg, 2018-06-19:
@pvaibhav are you still experiencing issues ? or should we go ahead and close this as resolved ? :)

pvaibhav, 2018-06-19:
@mostafaberg  This is unfortunately an 'unsolvable' issue. The only way is to instruct our users to turn Bluetooth off and on (from Settings app, not even Control Center), or turn the phone off/on.

And this needs to be done for every DFU since we cannot update the bootloader.

So I guess this issue can be closed, but probably not marked as resolved.

mostafaberg, 2018-06-19:
Sorry to hear that, yes it is indeed a tricky one to "fix", the only way to get it fixed without the workaround is to actually increment the mac address when in Bootloader. if that's not a possibility, then toggling bluetooth for at least 10 seconds solves the issue.

please note that the iOS given identifiers are not MACs, they are UUIDs that are "unique" per mac address within a single app, if a single bit changes in the mac address, you get a while new UUID. if you reinstall the app, you'll get a new UUID, but it'll stay the same for that given mac address during the lifetime of your app. this is just a clarification :)

closed 2018-06-19 by mostafaberg