nordicsemi/IOS-DFU-Library #179: Unable to load a bootloader/softdevice and application zip using iOS central

bitbytedog, 2018-02-02:
I have been unable to upgrade an nRF51 based peripheral with a zip file containing all three parts.

The SDK is nRF5-SDK v10.0.0 with S110 SoftDevice v8.0.0.

The DFU is legacy mode (no signing of the distribution package).

What I observe is the the first part (bootloader and softdevice) are transferred and flashed correctly. The target device enters DfuTarg mode, the iOS central tries to connect prior to sending part two and the connection hangs. The reason is, the MAC address of the peripheral has been incremented, so the CBPeripheral held by the DFUServiceInitiator is no longer valid. I would have expected the iOSDFULibrary to have scanned for the "new" peripheral. Checking the source code I found the following in *iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift*:
~~~~
 45     /// This flag must be set to true if the device will advertise with a new device address after it resets.
 44     /// The service will scan and use specified peripheral selector in order to connect to the new peripheral.
 43     internal var newAddressExpected  : Bool = false
~~~~

The variable newAddressExpected is always false, which stops the library from scanning when it jumps to the boot loader. I found #170 and thought I had found the same issue, but the fix for #170 will not work for SDK V10.0.0, because the DFU bootloader has a version characteristic (with value 0.8).

I am confused, because in the commit 2fde2307d20ebde3a8ba561f8297f260c45d29a6 associated with #172 it states the solution is limited to SDK v6.1, but the MAC address is also being incremented in SDK V10.0.0 (my mac is F1:83:19:9B:67:DE, but DfuTarg is advertising with the MAC F1:83:19:9B:67:DF). I confirmed the code does increment the MAC address in the file *sdk/components/libraries/bootloader_dfu/dfu_transport_ble.c*.

~~~~
1065         ble_gap_addr_t addr;                                                                                                                                                                                                                                                                                                                                        
   1 
   2         err_code = sd_ble_gap_address_get(&addr);
   3         APP_ERROR_CHECK(err_code);
   4 
   5         // Increase the BLE address by one when advertising openly.
   6         addr.addr[0] += 1;
   7 
   8         err_code = sd_ble_gap_address_set(BLE_GAP_ADDR_CYCLE_MODE_NONE, &addr);
   9         APP_ERROR_CHECK(err_code);
~~~~

This is the log output of the DFU process (which the transfer logs removed)

~~~~
 46 2018-02-01T23:33:46.525-0800|    bSafeBleCentral[192252]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:487: -[AgsCBCentral logWith:message:]: Initialize DFU Parameters completed
 47 2018-02-01T23:33:46.525-0800|               MAIN[191883]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:638: -[AgsCBCentral dfuStateDidChangeTo:]: uploading firmware to device
 48 2018-02-01T23:33:46.526-0800|    bSafeBleCentral[192252]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:481: -[AgsCBCentral logWith:message:]: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
 49 2018-02-01T23:33:46.531-0800|    bSafeBleCentral[192252]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:478: -[AgsCBCentral logWith:message:]: peripheral.writeValue(0x080c00, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
 50 2018-02-01T23:33:46.582-0800|    bSafeBleCentral[192680]|BLUETOOTH| INFO# AcoBleCentralImpl.mm:484: -[AgsCBCentral logWith:message:]: Data written to 00001531-1212-EFDE-1523-785FEABCD123
 51 2018-02-01T23:33:46.583-0800|    bSafeBleCentral[192680]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:487: -[AgsCBCentral logWith:message:]: Packet Receipt Notif Req (Op Code = 8, Value = 12) request sent
 52 2018-02-01T23:33:46.583-0800|    bSafeBleCentral[192680]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:481: -[AgsCBCentral logWith:message:]: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
 53 2018-02-01T23:33:46.584-0800|    bSafeBleCentral[192680]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:478: -[AgsCBCentral logWith:message:]: peripheral.writeValue(0x03, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
 54 2018-02-01T23:33:46.643-0800|    bSafeBleCentral[192252]|BLUETOOTH| INFO# AcoBleCentralImpl.mm:484: -[AgsCBCentral logWith:message:]: Data written to 00001531-1212-EFDE-1523-785FEABCD123
 55 2018-02-01T23:33:46.643-0800|    bSafeBleCentral[192252]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:487: -[AgsCBCentral logWith:message:]: Uploading firmware...
 56 2018-02-01T23:33:46.644-0800|    bSafeBleCentral[192252]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:481: -[AgsCBCentral logWith:message:]: Sending firmware to DFU Packet characteristic...
 57 2018-02-01T23:33:46.644-0800|               MAIN[191883]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:663: -[AgsCBCentral dfuProgressDidChangeFor:outOf:to:currentSpeedBytesPerSecond:avgSpeedBytesPerSecond:]: Sending part 1 of 2 : 0% done at 0B/s [average: 0B/s]
 58 .
 59 .
 60 .                                                                                                                                                                                                                                                                                                                                                                    
 61 2018-02-01T23:34:25.854-0800|               MAIN[191883]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:663: -[AgsCBCentral dfuProgressDidChangeFor:outOf:to:currentSpeedBytesPerSecond:avgSpeedBytesPerSecond:]: Sending part 1 of 2 : 100% done at 2827.57B/s [average: 2623.86B/s]
 62 2018-02-01T23:34:25.942-0800|    bSafeBleCentral[192684]|BLUETOOTH| INFO# AcoBleCentralImpl.mm:484: -[AgsCBCentral logWith:message:]: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100301
 63 2018-02-01T23:34:25.943-0800|    bSafeBleCentral[192684]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:487: -[AgsCBCentral logWith:message:]: Response (Op Code = 3, Status = 1) received
 64 2018-02-01T23:34:25.944-0800|    bSafeBleCentral[192684]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:487: -[AgsCBCentral logWith:message:]: Upload completed in 39.30 seconds
 65 2018-02-01T23:34:25.945-0800|               MAIN[191883]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:641: -[AgsCBCentral dfuStateDidChangeTo:]: validating firmware image
 66 2018-02-01T23:34:25.945-0800|    bSafeBleCentral[192684]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:481: -[AgsCBCentral logWith:message:]: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
 67 2018-02-01T23:34:25.945-0800|    bSafeBleCentral[192684]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:478: -[AgsCBCentral logWith:message:]: peripheral.writeValue(0x04, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
 68 2018-02-01T23:34:26.001-0800|             Thread[193023]|BLUETOOTH| INFO# AcoBleCentralImpl.mm:484: -[AgsCBCentral logWith:message:]: Data written to 00001531-1212-EFDE-1523-785FEABCD123
 69 2018-02-01T23:34:26.002-0800|             Thread[193023]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:487: -[AgsCBCentral logWith:message:]: Validate Firmware (Op Code = 4) request sent
 70 2018-02-01T23:34:26.122-0800|    bSafeBleCentral[192684]|BLUETOOTH| INFO# AcoBleCentralImpl.mm:484: -[AgsCBCentral logWith:message:]: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100401
 71 2018-02-01T23:34:26.122-0800|    bSafeBleCentral[192684]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:487: -[AgsCBCentral logWith:message:]: Response (Op Code = 4, Status = 1) received
 72 2018-02-01T23:34:26.123-0800|               MAIN[191883]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:644: -[AgsCBCentral dfuStateDidChangeTo:]: disconnecting from device
 73 2018-02-01T23:34:26.124-0800|    bSafeBleCentral[192684]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:481: -[AgsCBCentral logWith:message:]: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
 74 2018-02-01T23:34:26.124-0800|    bSafeBleCentral[192684]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:478: -[AgsCBCentral logWith:message:]: peripheral.writeValue(0x05, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
 75 2018-02-01T23:34:26.895-0800|             Thread[193023]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:478: -[AgsCBCentral logWith:message:]: [Callback] Central Manager did disconnect peripheral
[... cut ...]
 77 2018-02-01T23:34:26.895-0800|               MAIN[191883]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:629: -[AgsCBCentral dfuStateDidChangeTo:]: connecting to device
 78 2018-02-01T23:34:26.896-0800|             Thread[193023]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:481: -[AgsCBCentral logWith:message:]: Connecting to DfuTarg...
 79 2018-02-01T23:34:26.896-0800|             Thread[193023]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:478: -[AgsCBCentral logWith:message:]: centralManager.connect(peripheral, options: nil)
~~~~

bitbytedog, 2018-02-03:
I traced the problem to the jump to *BaseCommonDFUPeripheral.peripheralDidDisconnect()*.

After the first part is uploaded and verified, the **activating** flag is set to true. In the  *peripheralDidDisconnect()* method that calls a *connect()*, which fails because the peripheral is no longer valid (due to MAC change).

~~~~
        } else if activating {
            activating = false
            // This part of firmware has been successfully sent
            if (delegate?.peripheralDidDisconnectAfterFirmwarePartSent() ?? false) {
                connect()
            }
~~~~

I changed it to:

~~~~
        } else if activating {
            activating = false
            // This part of firmware has been successfully sent
            if (delegate?.peripheralDidDisconnectAfterFirmwarePartSent() ?? false) {
                if newAddressExpected {
                    newAddressExpected = false
                    // Scan for a new device and connect to it
                    switchToNewPeripheralAndConnect()
                } else {
                    connect()
                }
            }
~~~~

In *LegacyDFUPeripheral.activateAndReset()* I set the *newAddressExpected* flag to true.

I will attach the patch file.

Here is the log from the end of the transfer of part 1 of 2 to the successful completion of DFU.
~~~~
2018-02-02T17:08:28.115-0800|               MAIN[408822]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:658: -[AgsCBCentral dfuProgressDidChangeFor:outOf:to:currentSpeedBytesPerSecond:avgSpeedBytesPerSecond:]: Sending part 1 of 2 : 99% done at 2815.87B/s [average: 2591.82B/s]
2018-02-02T17:08:28.486-0800|               MAIN[408822]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:658: -[AgsCBCentral dfuProgressDidChangeFor:outOf:to:currentSpeedBytesPerSecond:avgSpeedBytesPerSecond:]: Sending part 1 of 2 : 100% done at 2750.46B/s [average: 2593.3B/s]
2018-02-02T17:08:28.569-0800|    bSafeBleCentral[408973]|BLUETOOTH| INFO# AcoBleCentralImpl.mm:479: -[AgsCBCentral logWith:message:]: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100301
2018-02-02T17:08:28.570-0800|    bSafeBleCentral[408973]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:482: -[AgsCBCentral logWith:message:]: Response (Op Code = 3, Status = 1) received
2018-02-02T17:08:28.580-0800|    bSafeBleCentral[408973]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:482: -[AgsCBCentral logWith:message:]: Upload completed in 39.77 seconds
2018-02-02T17:08:28.580-0800|    bSafeBleCentral[408973]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:476: -[AgsCBCentral logWith:message:]: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2018-02-02T17:08:28.580-0800|               MAIN[408822]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:636: -[AgsCBCentral dfuStateDidChangeTo:]: validating firmware image
2018-02-02T17:08:28.581-0800|    bSafeBleCentral[408973]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:473: -[AgsCBCentral logWith:message:]: peripheral.writeValue(0x04, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
2018-02-02T17:08:28.622-0800|             Thread[409142]|BLUETOOTH| INFO# AcoBleCentralImpl.mm:479: -[AgsCBCentral logWith:message:]: Data written to 00001531-1212-EFDE-1523-785FEABCD123
2018-02-02T17:08:28.622-0800|             Thread[409142]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:482: -[AgsCBCentral logWith:message:]: Validate Firmware (Op Code = 4) request sent
2018-02-02T17:08:28.743-0800|             Thread[409279]|BLUETOOTH| INFO# AcoBleCentralImpl.mm:479: -[AgsCBCentral logWith:message:]: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100401
2018-02-02T17:08:28.744-0800|             Thread[409279]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:482: -[AgsCBCentral logWith:message:]: Response (Op Code = 4, Status = 1) received
2018-02-02T17:08:28.744-0800|               MAIN[408822]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:639: -[AgsCBCentral dfuStateDidChangeTo:]: disconnecting from device
2018-02-02T17:08:28.745-0800|             Thread[409279]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:476: -[AgsCBCentral logWith:message:]: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2018-02-02T17:08:28.745-0800|             Thread[409279]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:473: -[AgsCBCentral logWith:message:]: peripheral.writeValue(0x05, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
2018-02-02T17:08:29.505-0800|             Thread[409279]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:473: -[AgsCBCentral logWith:message:]: [Callback] Central Manager did disconnect peripheral
2018-02-02T17:08:29.506-0800|             Thread[409279]|BLUETOOTH| INFO# AcoBleCentralImpl.mm:479: -[AgsCBCentral logWith:message:]: Disconnected by the remote device
2018-02-02T17:08:29.506-0800|             Thread[409279]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:476: -[AgsCBCentral logWith:message:]: Scanning for the DFU Bootloader...
2018-02-02T17:08:29.506-0800|               MAIN[408822]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:624: -[AgsCBCentral dfuStateDidChangeTo:]: connecting to device
2018-02-02T17:08:29.533-0800|             Thread[409142]|BLUETOOTH| INFO# AcoBleCentralImpl.mm:479: -[AgsCBCentral logWith:message:]: DFU Bootloader found with name DfuTarg
2018-02-02T17:08:29.534-0800|             Thread[409142]|BLUETOOTH|TRACE# AcoBleCentralImpl.mm:476: -[AgsCBCentral logWith:message:]: Connecting to DfuTarg...
[... cut ...]
2018-02-02T17:09:16.264-0800|    bSafeBleCentral[408973]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:473: -[AgsCBCentral logWith:message:]: [Callback] Central Manager did disconnect peripheral
2018-02-02T17:09:16.264-0800|    bSafeBleCentral[408973]|BLUETOOTH| INFO# AcoBleCentralImpl.mm:479: -[AgsCBCentral logWith:message:]: Disconnected by the remote device
2018-02-02T17:09:16.265-0800|               MAIN[408822]|BLUETOOTH|DEBUG# AcoBleCentralImpl.mm:642: -[AgsCBCentral dfuStateDidChangeTo:]: DFU completed successfully
~~~~

bitbytedog, 2018-02-03:
Patch file
https://gist.github.com/bitbytedog/f9e923a98de08eed5d55d40704f43db0

philips77, 2018-02-05:
Hmm.. i think we modified the part of the code you mentioned in a very recent commit actually to keep the device, not to scan for a new one. I'll look at it tomorrow again. Sorry for inconvenience.

philips77, 2018-02-06:
Hi again, you may switch to use version 4.4.3 before we fix this issue. It is only the latest version that is affected.

philips77, 2018-02-06:
Thanks for the patch. I copied your proposal and added a bunch of documentation.

bitbytedog, 2018-02-11:
Thanks for the update and the great library. Cheers.

PR #180 (2018-02-06): #179 fixed
This PR should fix #179. Some testing should be done to ensure it works as before on Secure DFU and it is possible to update Bootloader and App in one ZIP using Legacy DFU. 
In my opinion Secure DFU was intact and in worst case it will scan for the same peripheral in Legacy DFU, so no harm was made.
closed 2018-02-21 by philips77