nordicsemi/IOS-DFU-Library #185: Redefinition of module 'minizip' Error

saravanankk, 2018-03-07:
I am trying to create a Library that uses IOS-Pods-DFU-Library as one of the dependency. I am using COACOAPODS for my dependency management. I am releasing my library as a binary framework and customers will integrate it with their app using COACOAPODS from localhost server with our custom podspec file that will download our binary framework from locahost server and resolve other dependencies.

Before Integrating iOSDFULibrary its used to works fine in the Customer applications. But after specifying  iOSDFULibrary I am getting the following error 

**consumer app test/CustomerTest/Pods/Zip/Zip/minizip/module.modulemap:1:8: error: redefinition of module 'minizip'
module minizip [system][extern_c] {
       ^**

Am i missing something here? Any help will be appreciated.  

Below are our sample podspec file and example how user includes our library.

Pod::Spec.new do |s|
  s.name         = "xxxxxx"
  s.version      = "1.0.0"
  s.summary      = "xxxxxx description ."
  s.author = { "" => "" }
  s.license      = { :type => 'Apache-2.0', :file => 'LICENSE' }
 
   s.ios.preserve_paths = 'xxxxxx.framework/*.bundle' 

   s.ios.deployment_target = "9.0"

    s.source = { 
      :http => 'http://localhost:8000/xxxxxx.zip'
    }  

  s.ios.vendored_frameworks = 'xxxxxx.framework'
  s.dependency 'SSZipArchive', ' 2.1'
  s.dependency 'XCGLogger', ' 6.0'
  s.dependency "iOSDFULibrary", '4.1.1'
  
end

Customers will include our library by specifying above podsepec file path as below in their pod file

pod 'xxxxxx', :podspec => './xxxxxx.podspec'

mostafaberg, 2018-03-08:
Hi @saravanankk !, sorry to hear about your issue, seems like you have a very specific setup and we are not fully aware of how the inner workings of this setup works, it would be best to post your question to one of cocoapods support channels instead as they may give you better feedback of your setup and what is causing the issue. we can only support issues with the library here and not cocoapods related issue, hope that helps

saravanankk, 2018-03-09:
Thanks for the reply. I will ask in cocoapods support channels.

simonseyer, 2018-04-23:
I probably solved this with my PR in the Minizip repo: https://github.com/dexman/Minizip/pull/2. However, the https://github.com/marmelroy/Zip repo would need to include it manually (because they just copied and pasted the code into their project for some reason). Unfortunately, it doesn't seem like it is properly maintained. My other PR on this repo is also open for three months.

mostafaberg, 2018-04-23:
@simonseyer Yep that's true, I'm not sure how to move forward with Zip Support, but we will eventually look into some replacement or implement our own native Swift version, but that needs time :(

mostafaberg, 2018-06-19:
Closing this for now as there's a workaround in the comment above, we have still not looked into replacing the `Zip` dependency

closed 2018-06-19 by mostafaberg