nordicsemi/IOS-DFU-Library #187: Error 202: Device disconnected Unexpectedly

zurichrich, 2018-03-09:
Here is the console log of DFU process. It appears to fail as the firmware starts uploading. Can you give me any pointers how to fix?! Thanks.

DfuTarg discovered
Connection successful to DfuTarg
DfuTarg connected
DFUProcess started
I: Connected to DfuTarg
V: Discovering services...
D: peripheral.discoverServices(nil)
dfuStateDidChange to Connecting
I: Services discovered
V: Starting Legacy DFU...
I: Connected to DfuTarg
I: Services discovered
V: Legacy DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I: DFU characteristics discovered
V: Reading DFU Version number...
D: peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
I: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0800
A: Version number read: 0.8
V: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
dfuStateDidChange to Starting
V: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A: DFU Control Point notifications enabled
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
V: Writing image sizes (0b, 0b, 24692b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x000000000000000074600000, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Start DFU (Op Code = 1, Upload Mode = 4) request sent
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
A: Response (Op Code = 1, Status = 1) received
A: Writing Initialize DFU Parameters...
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0200, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
V: Writing to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0xffffffffffffffff0100feff1ce1, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0201, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100201
A: Response (Op Code = 2, Status = 1) received
A: Initialize DFU Parameters completed
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x080000, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
dfuStateDidChange to Uploading
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Packet Receipt Notif Req (Op Code = 8, Value = 0) request sent
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x03, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Uploading firmware...
V: Sending firmware to DFU Packet characteristic...
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
Error 202: Device disconnected unexpectedly

zurichrich, 2018-03-11:
Hi @mostafaberg, the .zip file I am using works fine with the nRF Connect DFU tool, so I suspect either that I am not getting the .zip correctly from my app main bundle, or there is something wrong with my DFU code. Can you see anything in the above log to give me any pointers? 

Here is my .zip file set up code:
       let fileURL = Bundle.main.url(forResource: "ohc_test_1.3q", withExtension: ".zip")
       firmwareFile = DFUFirmware(urlToZipFile: fileURL!)
       dfuController = dfuInitiator.with(firmware: firmwareFile!).start()

simonseyer, 2018-04-23:
@zurichrich you may wanna give the new release a try: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/releases/tag/4.1.3

mcaap, 2018-10-13:
We still face this issue on a iPhone X (12.0.1) with the latest DFU library 4.2.0.
And it occurs consistently. Any way to further debug this issue or any solution we can try?

closed 2018-10-15 by mostafaberg

mcaap, 2018-10-15:
Sure. This happens on a User's device so obviously not very easy to capture logs. What kind of logs would you need to further debug this issue?

mostafaberg, 2018-10-15:
You don't have a crash reporting system that captures the logs? if you don't , there's no way to actually know that the problem is from the DFU Library, could be anywhere, just open up a new ticket, tell us as much details as possible, and we will try to help :)