nordicsemi/IOS-DFU-Library #189: Device disconnected after Jump to bootloader request

shift00, 2018-03-28:
Hi,
I have problem with update my device with this framework. The same solution on android version is working fine.

My code:
`       let dfuInitiator = DFUServiceInitiator(centralManager: centralManager, target: dfuPeripheral)
        dfuInitiator.forceDfu = true
        dfuInitiator.delegate = self
        dfuInitiator.progressDelegate = self
        dfuInitiator.logger = self
        dfuInitiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true
        dfuInitiator.packetReceiptNotificationParameter = 15
        
        dfuController = dfuInitiator.with(firmware: firmware).start()`

I saw solutions with changes in packetReceiptNotificationParameter, but my change to 0, 1, 2, 12, 15 values not working. Change of forceDfu field (true/false) does not give any changes.

`D: [Callback] Central Manager did connect peripheral
I: Connected to beaconDevice
V: Discovering services...
D: peripheral.discoverServices
I: Services discovered
V: Legacy DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I: DFU characteristics discovered
V: Reading DFU Version number...
D: peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
I: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
A: Version number read: 0.1
V: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
dfuStateDidChange(to:) - 💡 DFUState - starting
V: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A: DFU Control Point notifications enabled
W: Application with buttonless update found
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
dfuStateDidChange(to:) - 💡 DFUState - enablingDfuMode
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Connecting to beaconDevice...
D: centralManager.connect(peripheral, options: nil)`

After **Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent** device was disconnected and app trying reconnect it (?).

I have IOS-Pods-DFU-Library 4.1.2. Tested with two different devices.

Does anyone have an idea what I need to change to make it work?

philips77, 2018-03-28:
This is intended. Your device restarts into Bootloader mode and should start advertising. Dfu lib will search for DfuTarg name in the adv packet and reconnect to. If you have changed how the Bootloader advertises, use dfuPeripheralSelector and match it to your Bootloader.

ForceDfu is used only on rare situations. Please read the incode documentation for details.

shift00, 2018-03-28:
@philips77 do you have some example with this dfuPeripheralSelector? 

I saw android implementation and after `enablingDfuMode` state, MAC address of beacon is changed (last character is incremented) and DFU process is restarted.

philips77, 2018-03-28:
On iOS device address is not available for apps, so we could do what we have done on Android...
There is a peripheralSelector property in initiator, which you may set to a custom selector.

philips77, 2018-03-28:
The point is, that the default selector won't work only if your Bootloader has a custom adv name, or doesn't advertise with the secure dfu uuid. It will work with the default Bootloader impl.

mostafaberg, 2018-06-19:
@shift00 are you still having problems or is this now resolved ? will close within this week if there's no further help needed

shift00, 2018-06-19:
@mostafaberg problem solved 👍 I forgot to close the topic, sorry.

closed 2018-06-19 by shift00