nordicsemi/IOS-DFU-Library #190: Import via `Carthage` issue: "error: missing required module 'minizip'"

letko-dmitry, 2018-04-14:
We imported the library via `Carthage` and didn't added `Checkout` folder to our repo. And it led to the following error during build:
```
<unknown>:0: error: missing required module 'minizip'
```
It seems it happened because of incorrect configuration of `Zip` library which relies on declaration of `minizip` module.

There are a lot of issues with that like the following one:
https://github.com/marmelroy/Zip/issues/134

simonseyer, 2018-04-23:
I have a PR open in the Zip repo that should solve the issue: https://github.com/marmelroy/Zip/pull/119

mostafaberg, 2018-04-23:
@simonseyer Perfect!, I'll wait for the maintainer to give a response there, I can merge but have no access to push to trunk, so I have to wait for a bit until it's reviewed by him

mostafaberg, 2018-06-19:
This seems to have been merged, closing the issue as it's now resolved

closed 2018-06-19 by mostafaberg

MarcSteven, 2018-08-23:
@mostafaberg  Perhaps it's not a solution as most developers met the issue ,and now no perfect solutions to solve it absolutely