nordicsemi/IOS-DFU-Library #192: IUO overkill

DevAndArtist, 2018-04-21:
Hi there, I'm reading the source code and I'm shocked by the fact on how many IUO's are used. Why was this library designed so poorly in an unsafe an dangerous way? Is there any plans to rework that and make the internals safe? This library heavily relies on a lot of assumptions that values are non-optional and uses a lot of force casts. I hope you understand my criticism and we can have a constructive democratic conversation. To wrap up: Because of that knowledge I'm afraid of touching this library even though I must do it for our project.

For reference: 
- IUO means Implicit Unwrapped Optional

mostafaberg, 2018-04-21:
Hi @DevAndArtist , Thanks for your feedback !

The references you see to implicit unwrapping comes form the old days of the library being automatically converted to swift from it's original Obj-C implementation, Glad you can see the unsafeness, but for the time being this is how it will be until we have some time to look at bigger feature changes.

But as you say, that's the internals of the library and it's usually a very bad idea to modify any 3rd party source code to fit your needs, if you see that for any reason you need to modify our codebase to get a feature working, then you need to open up a feature request, or modify the code but risk being stuck with a custom fork that will break with every upgrade :)

Are you having any issues that require the library to actually need your changes ? we would love to hear those :)

On a side note, we always welcome PRs and forks :) so don't feel afraid to modify, break things and send us questions or PRs !

philips77, 2018-04-23:
Hi,
I did those implicit unwrapped optionals on purpose to make sure the lib will fail as fast as possible if such reference was nil. Where a reference is unwrapped it means that at the time of executing that line of code the value can't be nil and dfu wouldn't work otherwise. Actually, in the lib there should be no optionals that are not implicitly unwrapped if they can't be nil. Basically, only delegates set by user are truly optionals (maybe something else as well, don't remember). 
The lib wasn't really designed to be altered, rather used as is. Out of curiosity, what are the changes you need?

DevAndArtist, 2018-04-23:
I wasn't really talking about altering the lib, but rather about the fact that I'm afraid using it. You're telling me that the lib is designed to fail fast when references become nil, this is simply wrong and dangerous design. Non of the references that are non-optional will ever become nil, hence non-optional and they hold strong references (except `unowned`). I recall that the migrator in the early days of Swift migrated Objective-C code into a lot of IOU's but this is not only mentally wrong, but also completely unintuitive. I as a library user do not expect the library to crash or fail if it's not a compiler error or some wrong calculation that leads to an undefined behaviour or the common bugs as we know them. Therefore if the library fails unexpectedly because of a null pointer (which is basically what IOU's lead to) then the whole app that relies on it will crash and you cannot do anything about it.

mostafaberg, 2018-10-09:
Closing as this is not an issue but rather a feedback that we will not work on at the moment, but will keep in the backlog, thanks a lot for the feedback :)

closed 2018-10-09 by mostafaberg