nordicsemi/IOS-DFU-Library #193: Strong reference cycle

DevAndArtist, 2018-04-22:
I'm still in process of reading and understanding most of the internal code but I just noticed a potential issue. I say potential because it might already be resolved automatically at some point. If not then this is really an issue:

- https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L258-L260-
- https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/DFUSelector/DFUServiceSelector.swift#L41
- https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/DFUServiceController.swift#L27

philips77, 2018-04-22:
Hi, the strong reference is there intended, in order to be able to "start dfu and forget initiator and controller". It's broken when the dfu ends in either way. I hope.

lovexiaoxia, 2018-05-17:
Hi,how do I after upgrade success or failure, release it?

philips77, 2018-05-17:
It does it automatically on success or error. All deinit were called when I was testing it.

closed 2018-10-05 by mostafaberg