nordicsemi/IOS-DFU-Library #196: Showing "Error while creating ZIP stream: The operation couldn't be completed" while executing DFUFirmware(URL)

usamaaftab80, 2018-05-30:
I am running into the same problem. For some reason, as soon as it executes the DFUFirmware(URL) constructor, it gives me the error: "Error while creating ZIP stream: The operation couldn't be completed."

Here is my code:

```
let documentsUrl:URL =  (FileManager.default.urls(for: .documentDirectory, in: .userDomainMask).first as URL?)!
        let destinationFileUrl = documentsUrl.appendingPathComponent(Constant.firmwareLocalPath)
        
        let selectedFirmware = DFUFirmware(urlToZipFile: destinationFileUrl)
        
        let initiator = DFUServiceInitiator(centralManager: self.central, target: self.peripheral).with(firmware: selectedFirmware!)
        initiator.logger = self; // - to get log info
        initiator.delegate = self; // - to be informed about current state and errors
        initiator.progressDelegate = self; // - to show progress bar
        initiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true
        
        self.dfuServiceController = initiator.start()
```
and my firmware path looks like this:
`file:///var/mobile/Containers/Data/Application/64D07EC7-A64F-4D7B-9F89-7A5A700CAD5A/Documents/firmware.zip`

A quick reply will be appreciated. I am not sure why it is behaving like this? To be clear, nothing is nil. Central, Peripheral and path, everything is valid.

closed 2018-05-31 by usamaaftab80

hzw0407, 2019-04-19:
How did you solve your problem? I've come across this problem now.

philips77, 2019-04-23:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/296