nordicsemi/IOS-DFU-Library #197: Unable to add library without Cocoapods

cloudface, 2018-06-06:
Hello

I am unable to add the DFU iOS library using Cocoapods. I added the following to my pod file:
`pod 'iOSDFULibrary'`
and ran pod install. When I tried to run my app, I got the following error:

[image]

In my podfile.lock I noticed the following:
`- iOSDFULibrary (4.1.3): - Zip (~> 1.0)`

I saw that there was a new version of the Zip library (1.1.0). Maybe this issue has been fixed in that version of the library but I don't know how to get iOSDFULibrary to use the new version of Zip. How can I successfully add the iOSDFULibrary to my project using Cocoapods? Thanks guys.

philips77, 2018-06-06:
Hi, @mostafaberg, who's maintaining the project is on vacations, but I'll try to help you tomorrow.

cloudface, 2018-06-07:
Thanks @philips77 

In the meantime I did a bit of digging. My project is currently using Swift 3.2. I quickly created an empty Swift 4.0 app and installed the pod there and it worked. Do I have to upgrade my project to Swift 4.0 or is there a way to get it to work with Swift 3.2 projects? Is that even the root cause?

philips77, 2018-06-07:
Apple deprecates old Swift versions quite quickly. I'd recommend using the latest version. There weren't many changes from 3.2 to 4, afair.

mostafaberg, 2018-06-19:
@cloudface This issue is not related to our library, but the Zip dependency, from what I understand it seems to have been fixed, but we do recommend moving on to Swift 4.0 as soon as possible :)

closed 2018-06-19 by mostafaberg