nordicsemi/IOS-DFU-Library #198: alternativeAdvertisingNameEnabled and SDK 13

tompi, 2018-06-07:
You are probably aware of this, but it caused us some grief.

The alternativeAdvertisingNameEnabled parameter defaults to true, and causes the updater to try and rename the device during dfu(secure dfu) which is only supported by sdk 14.

Our devices are at sdk 13, so this failed, but NOT for android devices, so you may want to do this like android lib has done it(not sure if it defaults to false there, or if it is able to sniff out sdk version somehow...)

Sorry to post a non-problem here, but I think it might save other some headaches in the future :)

philips77, 2018-06-07:
On Android we don't use this feature, as there device address is available, so we base the device scan on address. On iOS we had to do this workaround to eliminate connecting to a wrong device. But you may force disable this feature by seeing this flash to false it your using sdk 13. Although, on sdk 13 it should try to set the name and fail, and resume without it, if you didn't do many changes in the bootloader. It you did, disable it.

tompi, 2018-06-07:
OK, we did some changes yes(to enable updating other stuff than the BLE processor).

We set it to false and it works fine. (was just some extra hassle since we use xamarin, and the flag was relatively new, so not exposed in bindings :)

Thank you for quick response, I just figured if you KNOW the remote SDK is 13, you can ignore this flag, but you probably have no way of telling.

closed 2018-06-07 by tompi

philips77, 2018-06-07:
Yes... There is no way for us to know that. We know only after the request to set the name is rejected.