nordicsemi/IOS-DFU-Library #199: dfuStateDidChange(to state: DFUState) does not go through state completed

blackcodefr, 2018-06-19:
When updating an nRF52 device I don't get the DFUState.completed when done. Here are the logs:

I only get a DFUState.disconnecting when the logs say it's completed.

`2018-06-19 08:51:38.036192+0200 1 : Connecting to MAS-03DE5EN...
2018-06-19 08:51:38.036377+0200 0 : centralManager.connect(peripheral, options: nil)
2018-06-19 08:51:38.038874+0200 0 : [Callback] Central Manager did connect peripheral
2018-06-19 08:51:38.038980+0200 5 : Connected to MAS-03DE5EN
2018-06-19 08:51:38.039118+0200 1 : Discovering services...
2018-06-19 08:51:38.039195+0200 0 : peripheral.discoverServices(nil)
2018-06-19 08:51:38.044541+0200 5 : Services discovered
2018-06-19 08:51:38.044820+0200 1 : Starting Secure DFU...
2018-06-19 08:51:38.045288+0200 5 : Connected to MAS-03DE5EN
2018-06-19 08:51:38.045427+0200 5 : Services discovered
2018-06-19 08:51:38.045510+0200 1 : Secure DFU Service found
2018-06-19 08:51:38.045835+0200 1 : Discovering characteristics in DFU Service...
2018-06-19 08:51:38.046207+0200 0 : peripheral.discoverCharacteristics(nil, for: FE59)
2018-06-19 08:51:38.082317+0200 5 : DFU characteristics discovered
2018-06-19 08:51:38.082705+0200 1 : Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
2018-06-19 08:51:38.082748+0200 0 : peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
2018-06-19 08:51:38.201566+0200 1 : Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
2018-06-19 08:51:38.201654+0200 10 : Buttonless DFU indications enabled
2018-06-19 08:51:38.201702+0200 15 : Application with buttonless update found
2018-06-19 08:51:38.201955+0200 1 : Trying setting bootloader name to Dfu35047
2018-06-19 08:51:38.202211+0200 1 : Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
2018-06-19 08:51:38.202856+0200 0 : peripheral.writeValue(0x02084466753335303437, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-06-19 08:51:38.293745+0200 5 : Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
2018-06-19 08:51:38.294158+0200 5 : Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200201
2018-06-19 08:51:38.294449+0200 10 : Response (Op Code = 2, Status = 1) received
2018-06-19 08:51:38.294574+0200 10 : Bootloader name changed successfully
2018-06-19 08:51:38.295057+0200 1 : Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
2018-06-19 08:51:38.295235+0200 0 : peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-06-19 08:51:38.383659+0200 5 : Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
2018-06-19 08:51:38.384487+0200 5 : Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101
2018-06-19 08:51:38.385133+0200 10 : Response (Op Code = 1, Status = 1) received
2018-06-19 08:51:39.137871+0200 DID DISCONNECT: MAS-03DE5EN
2018-06-19 08:51:39.139109+0200 0 : [Callback] Central Manager did disconnect peripheral
2018-06-19 08:51:39.139223+0200 5 : Disconnected by the remote device
2018-06-19 08:51:39.139436+0200 1 : Scanning for the DFU Bootloader...
2018-06-19 08:51:39.171778+0200 5 : DFU Bootloader found with name Dfu35047
2018-06-19 08:51:39.171856+0200 1 : Connecting to Dfu82986...
2018-06-19 08:51:39.171895+0200 0 : centralManager.connect(peripheral, options: nil)
2018-06-19 08:51:39.614831+0200 0 : [Callback] Central Manager did connect peripheral
2018-06-19 08:51:39.615136+0200 5 : Connected to Dfu82986
2018-06-19 08:51:39.615357+0200 1 : Discovering services...
2018-06-19 08:51:39.617952+0200 0 : peripheral.discoverServices([FE59])
2018-06-19 08:51:39.673761+0200 5 : Services discovered
2018-06-19 08:51:39.674089+0200 1 : Secure DFU Service found
2018-06-19 08:51:39.674946+0200 1 : Discovering characteristics in DFU Service...
2018-06-19 08:51:39.675210+0200 0 : peripheral.discoverCharacteristics(nil, for: FE59)
2018-06-19 08:51:39.680027+0200 5 : DFU characteristics discovered
2018-06-19 08:51:39.680312+0200 1 : MTU set to 247
2018-06-19 08:51:39.683279+0200 1 : Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-06-19 08:51:39.683433+0200 0 : peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
2018-06-19 08:51:39.702837+0200 1 : Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-06-19 08:51:39.703107+0200 10 : Secure DFU Control Point notifications enabled
2018-06-19 08:51:39.704181+0200 1 : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-06-19 08:51:39.704797+0200 0 : peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-06-19 08:51:39.732828+0200 5 : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-06-19 08:51:39.734104+0200 5 : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601000100000000000000000000
2018-06-19 08:51:39.736576+0200 10 : Command object info (Max size = 256, Offset = 0, CRC = 00000000) received
2018-06-19 08:51:39.736965+0200 1 : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-06-19 08:51:39.740703+0200 0 : peripheral.writeValue(0x01018b000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-06-19 08:51:39.796012+0200 5 : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-06-19 08:51:39.797552+0200 5 : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
2018-06-19 08:51:39.798099+0200 1 : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-06-19 08:51:39.798735+0200 0 : peripheral.writeValue(0x020000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-06-19 08:51:39.822580+0200 5 : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-06-19 08:51:39.823082+0200 5 : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
2018-06-19 08:51:39.823287+0200 10 : Packet Receipt Notif disabled (Op Code = 2, Value = 0)
[... 900 lines cut ...]
2018-06-19 08:52:11.097433+0200 5 : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60030100400300e0b62cbc
2018-06-19 08:52:11.097554+0200 10 : Checksum (Offset = 212992, CRC = BC2CB6E0) received
2018-06-19 08:52:11.166970+0200 1 : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-06-19 08:52:11.167061+0200 0 : peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-06-19 08:52:11.186336+0200 5 : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-06-19 08:52:11.187040+0200 5 : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600401
2018-06-19 08:52:11.187094+0200 10 : Data object executed
2018-06-19 08:52:11.187152+0200 1 : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-06-19 08:52:11.187219+0200 0 : peripheral.writeValue(0x0102a40e0000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-06-19 08:52:11.216387+0200 5 : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-06-19 08:52:11.217043+0200 5 : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
2018-06-19 08:52:11.217101+0200 5 : Data object 53/53 created
2018-06-19 08:52:11.217152+0200 10 : Uploading firmware...
2018-06-19 08:52:11.217289+0200 1 : Sending firmware to DFU Packet characteristic...
2018-06-19 08:52:11.607187+0200 1 : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-06-19 08:52:11.608098+0200 0 : peripheral.writeValue(0x03, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-06-19 08:52:11.698467+0200 5 : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-06-19 08:52:11.701213+0200 5 : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600301a44e030047ce7742
2018-06-19 08:52:11.701686+0200 10 : Checksum (Offset = 216740, CRC = 4277CE47) received
2018-06-19 08:52:11.801325+0200 1 : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-06-19 08:52:11.801424+0200 0 : peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-06-19 08:52:11.832090+0200 5 : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-06-19 08:52:12.132236+0200 5 : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600401
2018-06-19 08:52:12.132467+0200 10 : Data object executed
2018-06-19 08:52:12.134250+0200 10 : Upload completed in 32.45 seconds`

mostafaberg, 2018-06-19:
Hard to see from those logs as that doesn't show what your delegates are doing.
first, make sure you are using the latest version of the library, then make sure you are not overriding your delegates and that you are implementing the proper delegate protocol, for reference see [here](https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/blob/e44ea97721bc6d995bcd4099606ef04d48688653/nRF%20Toolbox/DFU/DFUViewController/NORDFUViewController.swift#L457-L467) and [here](https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/blob/e44ea97721bc6d995bcd4099606ef04d48688653/nRF%20Toolbox/DFU/DFUViewController/NORDFUViewController.swift#L290)

let me know if that does work out for you

blackcodefr, 2018-06-19:
Pod used: iOSDFULibrary (4.1.3)
I am indeed implementing the proper delegates.

Initiation:

```
- (void)initiateDFUControllerForController:(Controller *)controller {
    _isPerformingDFUUpdate = YES;
    [self startProgress];
    
    CBCentralManager *manager = [SLMBluetoothUpdateManager sharedInstance].centralManager;
    NSArray *peripherals = [manager retrievePeripheralsWithIdentifiers:@[[[NSUUID alloc] initWithUUIDString:self.moduleToUpdate.hardwareAddress]]];
    self.shouldRetry = YES;
    
DFUFirmware *selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile:[self getUpdateFileForDevice:self.moduleToUpdate]];
    
    self.dfuServiceInitiator = [[DFUServiceInitiator alloc] initWithCentralManager: [SLMBluetoothUpdateManager sharedInstance].centralManager target:[peripherals lastObject]];
    self.dfuServiceInitiator = [self.dfuServiceInitiator withFirmware:selectedFirmware];
    self.dfuServiceInitiator.forceDfu = NO; // default NO
    self.dfuServiceInitiator.logger = self; // - to get log info
    self.dfuServiceInitiator.delegate = self; // - to be informed about current state and errors
    self.dfuServiceInitiator.progressDelegate = self; // - to show progress bar
    self.dfuServiceController = [self.dfuServiceInitiator start];
}
```

I am implementing the related delegates to handle the different states. 

This is new for us with SDK 15.

mostafaberg, 2018-06-19:
IIRC, upload being completed is not the same as DFU process being completed.
Does the peripheral actually function properly after you flash that firmware?

bringing @philips77 into the loop in case he sees something I don't

blackcodefr, 2018-06-19:
Yes, the peripheral works fine after.

mostafaberg, 2018-06-19:
Interesting, it doesn't look like you're doing anything wrong here, code is correct so it might be a bug on our side or on SDK 15, let's see what @philips77 says.

philips77, 2018-06-19:
Hi @blackcodefr!

The last log output you get is sent from [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/SecureDFU/DFU/SecureDFUExecutor.swift#L183). As the comment above says, the peripheral should initiate restart, disconnect and start advertising again with the new firmware. The library after sending the last Execute command waits for the disconnect event. 
Here's how it works:
1. Just before sending the last Execute command you are here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/SecureDFU/DFU/SecureDFUExecutor.swift#L158
2. This sets the `firmwareSent` flag to true, as all the bytes were sent. Also, one line below it calls `sendExecuteCommand(andActivateIf: true)`.
3. You end up here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/SecureDFU/Peripheral/SecureDFUPeripheral.swift#L192
4. This sets the `activating` flag.
5. Then, after receiving a confirmation of the execution, you go to the first link I sent ([here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/SecureDFU/DFU/SecureDFUExecutor.swift#L183)), where it sets the state to `.disconnecting`.
6. Now the peripheral should swap images and restart.
7. after disconnection, you end up here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift#L242, then here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift#L468 and as the `activating` flag is set to true, you go to this place: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift#L494.
8. This is executed here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/53564566e87030d2d13e1cbab0565a2c221a5056/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUExecutor.swift#L142 and as the firmware was completed, you skip the if statement and set the state to `.completed` [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/53564566e87030d2d13e1cbab0565a2c221a5056/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUExecutor.swift#L152).

So, as I see it, there are 2 possible reasons;
1. Your device doesn't disconnect after receiving the whole firmware.
2. Before you got `.completed` state you set the CBCentralManager.delegate to some other object, and the library isn't notified about the disconnection.

As far as I know there weren't any changes in the SDK 15 that would prevent the bootloader from disconnecting after receiving the new firmware. Please, double check the suggestions I gave. If you find nothing weird, I can investigate further, but then it looks like there is a problem in your DFU bootloader and you could ask a question on DevZone as they are more qualified to solve such problems. The lib should behave correctly had the device disconnected at the end.

blackcodefr, 2018-06-19:
Completing the previous logs, after "Upload completed in 32.45 seconds" I get this:

```
2018-06-19 13:55:17.441781+0200 0 : [Callback] Central Manager did disconnect peripheral
2018-06-19 13:55:17.442091+0200 5 : Disconnected by the remote device
2018-06-19 13:55:17.442684+0200 1 : Scanning for the DFU Bootloader...
```

So it does say "Disconnected by the remote device" but I still don't get the DFUState.Completed event.

I have also tested, after the Disconnecting event, if I turn off Bluetooth, then I get the Completed event.

blackcodefr, 2018-06-19:
I did some debugging with breakpoints, it seems the "activating" flag is not set to true, and thus, the completed event is not sent.

blackcodefr, 2018-06-19:
Breakpoints removed, it now works. Did not change anything in the code nor the library.

This is not very reassuring for DFU system, using iPhone 7 Plus on iOS 11.4 (15F79)

Logs below:

```
2018-06-19 15:10:27.119530+0200 1 : Connecting to MAS-03DE5EN...
2018-06-19 15:10:27.119588+0200 0 : centralManager.connect(peripheral, options: nil)
2018-06-19 15:10:27.120294+0200 0 : [Callback] Central Manager did connect peripheral
2018-06-19 15:10:27.120554+0200 5 : Connected to MAS-03DE5EN
2018-06-19 15:10:27.120584+0200 1 : Discovering services...
2018-06-19 15:10:27.120608+0200 0 : peripheral.discoverServices(nil)
2018-06-19 15:10:27.122020+0200 5 : Services discovered
2018-06-19 15:10:27.123732+0200 1 : Starting Secure DFU...
2018-06-19 15:10:27.124433+0200 5 : Connected to MAS-03DE5EN
2018-06-19 15:10:27.124571+0200 5 : Services discovered
2018-06-19 15:10:27.124593+0200 1 : Secure DFU Service found
2018-06-19 15:10:27.124685+0200 1 : Discovering characteristics in DFU Service...
2018-06-19 15:10:27.124714+0200 0 : peripheral.discoverCharacteristics(nil, for: FE59)
2018-06-19 15:10:27.128964+0200 DFU STATE DID CHANGE TO: 0
2018-06-19 15:10:27.160990+0200 5 : DFU characteristics discovered
2018-06-19 15:10:27.161188+0200 DFU STATE DID CHANGE TO: 1
2018-06-19 15:10:27.161443+0200 1 : Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
2018-06-19 15:10:27.161485+0200 0 : peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
2018-06-19 15:10:27.280337+0200 1 : Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
2018-06-19 15:10:27.280431+0200 10 : Buttonless DFU indications enabled
2018-06-19 15:10:27.280489+0200 15 : Application with buttonless update found
2018-06-19 15:10:27.280605+0200 DFU STATE DID CHANGE TO: 2
2018-06-19 15:10:27.280791+0200 1 : Trying setting bootloader name to Dfu31682
2018-06-19 15:10:27.280991+0200 1 : Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
2018-06-19 15:10:27.284289+0200 0 : peripheral.writeValue(0x02084466753331363832, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-06-19 15:10:27.341070+0200 5 : Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
2018-06-19 15:10:27.341861+0200 5 : Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200201
2018-06-19 15:10:27.342149+0200 10 : Response (Op Code = 2, Status = 1) received
2018-06-19 15:10:27.342272+0200 10 : Bootloader name changed successfully
2018-06-19 15:10:27.342442+0200 1 : Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
2018-06-19 15:10:27.342625+0200 0 : peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-06-19 15:10:27.403493+0200 5 : Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
2018-06-19 15:10:27.406187+0200 5 : Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101
2018-06-19 15:10:27.406526+0200 10 : Response (Op Code = 1, Status = 1) received
2018-06-19 15:10:28.154329+0200 DID DISCONNECT: MAS-03DE5EN
2018-06-19 15:10:28.156743+0200 0 : [Callback] Central Manager did disconnect peripheral
2018-06-19 15:10:28.156823+0200 5 : Disconnected by the remote device
2018-06-19 15:10:28.156892+0200 1 : Scanning for the DFU Bootloader...
2018-06-19 15:10:28.194961+0200 5 : DFU Bootloader found with name Dfu31682
2018-06-19 15:10:28.195061+0200 1 : Connecting to Dfu75200...
2018-06-19 15:10:28.195113+0200 0 : centralManager.connect(peripheral, options: nil)
2018-06-19 15:10:28.700941+0200 0 : [Callback] Central Manager did connect peripheral
2018-06-19 15:10:28.701263+0200 5 : Connected to Dfu75200
2018-06-19 15:10:28.701491+0200 1 : Discovering services...
2018-06-19 15:10:28.703437+0200 0 : peripheral.discoverServices([FE59])
2018-06-19 15:10:28.761175+0200 5 : Services discovered
2018-06-19 15:10:28.761508+0200 1 : Secure DFU Service found
2018-06-19 15:10:28.762321+0200 1 : Discovering characteristics in DFU Service...
2018-06-19 15:10:28.762588+0200 0 : peripheral.discoverCharacteristics(nil, for: FE59)
2018-06-19 15:10:28.769586+0200 5 : DFU characteristics discovered
2018-06-19 15:10:28.769814+0200 1 : MTU set to 247
2018-06-19 15:10:28.770196+0200 DFU STATE DID CHANGE TO: 1
2018-06-19 15:10:28.771259+0200 1 : Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-06-19 15:10:28.771397+0200 0 : peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
2018-06-19 15:10:28.803261+0200 1 : Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-06-19 15:10:28.803432+0200 10 : Secure DFU Control Point notifications enabled
2018-06-19 15:10:28.804231+0200 1 : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-06-19 15:10:28.804589+0200 0 : peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-06-19 15:10:28.833475+0200 5 : Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-06-19 15:10:28.835819+0200 5 : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601000100000000000000000000
2018-06-19 15:10:28.838430+0200 10 : Command object info (Max size = 256, Offset = 0, CRC = 00000000) received
2018-06-19 15:10:28.839195+0200 1 : Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-06-19 15:10:28.839989+0200 0 : peripheral.writeValue(0x01018b000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
[... 118 lines cut ...]
[... cut ...]
2018-06-19 15:11:00.378548+0200 5 : Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600401
2018-06-19 15:11:00.378763+0200 10 : Data object executed
2018-06-19 15:11:00.381189+0200 10 : Upload completed in 31.61 seconds
2018-06-19 15:11:00.381337+0200 DFU STATE DID CHANGE TO: 5
2018-06-19 15:11:00.622633+0200 0 : [Callback] Central Manager did disconnect peripheral
2018-06-19 15:11:00.623038+0200 5 : Disconnected by the remote device
2018-06-19 15:11:00.623318+0200 DFU STATE DID CHANGE TO: 6
```

blackcodefr, 2018-06-20:
After more investigating, I figured out that I passed the wrong centralManager to the DFUServiceInitiator. Passing the correct one makes it work reliably (still don't understand how it could work from time to time or with breakpoints but never mind).

After making the change I still had issue getting back "control" of my CentralManager instance after the DFU completed. I had to create a new one to be able to start scanning and connecting and getting delegate results reliably. Do you do anything particular to the centralManager passed to you after dfu completion?

Let me know if you need more details about the later behaviour I'm describing.

Thanks for your help.

mostafaberg, 2018-10-05:
Glad to hear your issue is resolved, if you have any other new issues that are not related to this one , please open it up as a new issue and we will help you out as soon as possible :) will close this now as it's resolved! good luck 👍

closed 2018-10-05 by mostafaberg

pdecostervgls, 2020-12-07:
I stumbled on this issue during research, as my application does not have the DFUState go to completed when I build in release mode. In debug mode it works fine. Also, somtimes I dont get any progress notifications either. Should metion that I am using a binding library for iOS since I am using the Uno Platform/Xamarin.

philips77, 2021-01-19:
Hi, did you manage to fix the issue on your side?
If not, it would be better if you opened a new issue, as this one is quite old and relates to an older version of the library.

pdecostervgls, 2021-01-19:
@philips77 No, sadly not. I will open a new issue for it, since this issue is coming up to need to be fixed soon, priority wise.