nordicsemi/IOS-DFU-Library #200: Crash BaseDFUPeripheral.discoversServices (DFUPeripheral.swift:349)

Drusy, 2018-07-05:
I am having the following crash on a few devices:

```
# Date: 2018-07-05T05:58:00Z
# OS Version: 11.4.0 (15F79)
# Device: iPhone SE
# RAM Free: 2.4%
# Disk Free: 8%

#0. Crashed: com.apple.main-thread
0  iOSDFULibrary                  0x105c43cb4 BaseDFUPeripheral.discoverServices() -> () (DFUPeripheral.swift:349)
1  iOSDFULibrary                  0x105c47d24 specialized BaseDFUPeripheral.centralManager(CBCentralManager, didConnect : CBPeripheral) -> () (DFUPeripheral.swift:193)
2  iOSDFULibrary                  0x105c42a58 BaseDFUPeripheral.centralManager(CBCentralManager, didConnect : CBPeripheral) -> () (DFUPeripheral.swift)
3  iOSDFULibrary                  0x105c42ab4 @objc BaseDFUPeripheral.centralManager(CBCentralManager, didConnect : CBPeripheral) -> () (DFUPeripheral.swift)
4  CoreBluetooth                  0x187c5b9d0 -[CBCentralManager handlePeripheralConnectionCompleted:] + 400
5  CoreBluetooth                  0x187c5c4d4 -[CBCentralManager handleMsg:args:] + 608
6  CoreBluetooth                  0x187c6a570 __30-[CBXpcConnection _handleMsg:]_block_invoke + 60
7  libdispatch.dylib              0x181f0caa0 _dispatch_call_block_and_release + 24
8  libdispatch.dylib              0x181f0ca60 _dispatch_client_callout + 16
9  libdispatch.dylib              0x181f169b4 _dispatch_queue_serial_drain$VARIANT$mp + 608
10 libdispatch.dylib              0x181f172fc _dispatch_queue_invoke$VARIANT$mp + 336
11 libdispatch.dylib              0x181f19534 _dispatch_main_queue_callback_4CF$VARIANT$mp + 716
12 CoreFoundation                 0x1825c3070 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 12
13 CoreFoundation                 0x1825c0bc8 __CFRunLoopRun + 2272
14 CoreFoundation                 0x1824e0da8 CFRunLoopRunSpecific + 552
15 GraphicsServices               0x1844c5020 GSEventRunModal + 100
16 UIKit                          0x18c4fd758 UIApplicationMain + 236
17 MichelinDDI                    0x104c125c0 main (main.swift:13)
18 libdyld.dylib                  0x181f71fc0 start + 4

--

#0. Crashed: com.apple.main-thread
0  iOSDFULibrary                  0x105c43cb4 BaseDFUPeripheral.discoverServices() -> () (DFUPeripheral.swift:349)
1  iOSDFULibrary                  0x105c47d24 specialized BaseDFUPeripheral.centralManager(CBCentralManager, didConnect : CBPeripheral) -> () (DFUPeripheral.swift:193)
2  iOSDFULibrary                  0x105c42a58 BaseDFUPeripheral.centralManager(CBCentralManager, didConnect : CBPeripheral) -> () (DFUPeripheral.swift)
3  iOSDFULibrary                  0x105c42ab4 @objc BaseDFUPeripheral.centralManager(CBCentralManager, didConnect : CBPeripheral) -> () (DFUPeripheral.swift)
4  CoreBluetooth                  0x187c5b9d0 -[CBCentralManager handlePeripheralConnectionCompleted:] + 400
5  CoreBluetooth                  0x187c5c4d4 -[CBCentralManager handleMsg:args:] + 608
6  CoreBluetooth                  0x187c6a570 __30-[CBXpcConnection _handleMsg:]_block_invoke + 60
7  libdispatch.dylib              0x181f0caa0 _dispatch_call_block_and_release + 24
8  libdispatch.dylib              0x181f0ca60 _dispatch_client_callout + 16
9  libdispatch.dylib              0x181f169b4 _dispatch_queue_serial_drain$VARIANT$mp + 608
10 libdispatch.dylib              0x181f172fc _dispatch_queue_invoke$VARIANT$mp + 336
11 libdispatch.dylib              0x181f19534 _dispatch_main_queue_callback_4CF$VARIANT$mp + 716
12 CoreFoundation                 0x1825c3070 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 12
13 CoreFoundation                 0x1825c0bc8 __CFRunLoopRun + 2272
14 CoreFoundation                 0x1824e0da8 CFRunLoopRunSpecific + 552
15 GraphicsServices               0x1844c5020 GSEventRunModal + 100
16 UIKit                          0x18c4fd758 UIApplicationMain + 236
17 MichelinDDI                    0x104c125c0 main (main.swift:13)
18 libdyld.dylib                  0x181f71fc0 start + 4

#1. com.apple.uikit.eventfetch-thread
0  libsystem_kernel.dylib         0x18207fe08 mach_msg_trap + 8
1  libsystem_kernel.dylib         0x18207fc80 mach_msg + 72
2  CoreFoundation                 0x1825c2e40 __CFRunLoopServiceMachPort + 196
3  CoreFoundation                 0x1825c0908 __CFRunLoopRun + 1568
4  CoreFoundation                 0x1824e0da8 CFRunLoopRunSpecific + 552
5  Foundation                     0x182f55674 -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 304
6  Foundation                     0x182f554dc -[NSRunLoop(NSRunLoop) runUntilDate:] + 84
7  UIKit                          0x18c1e2768 -[UIEventFetcher threadMain] + 136
8  Foundation                     0x183065efc __NSThread__start__ + 1040
9  libsystem_pthread.dylib        0x182241220 _pthread_body + 272
10 libsystem_pthread.dylib        0x182241110 _pthread_body + 290
11 libsystem_pthread.dylib        0x18223fb10 thread_start + 4

#2. Thread
0  libsystem_kernel.dylib         0x1820a1d84 __workq_kernreturn + 8
1  libsystem_pthread.dylib        0x18223feb4 _pthread_wqthread + 928
[... 41 lines cut ...]
3  CoreFoundation                 0x1825c0908 __CFRunLoopRun + 1568
4  CoreFoundation                 0x1824e0da8 CFRunLoopRunSpecific + 552
5  CFNetwork                      0x182dee6e0 -[__CoreSchedulingSetRunnable runForever] + 220
6  Foundation                     0x183065efc __NSThread__start__ + 1040
7  libsystem_pthread.dylib        0x182241220 _pthread_body + 272
8  libsystem_pthread.dylib        0x182241110 _pthread_body + 290
9  libsystem_pthread.dylib        0x18223fb10 thread_start + 4

#8. Thread
0  libsystem_kernel.dylib         0x1820a1d84 __workq_kernreturn + 8
1  libsystem_pthread.dylib        0x18223feb4 _pthread_wqthread + 928
2  libsystem_pthread.dylib        0x18223fb08 start_wqthread + 4

#9. Thread
0  libsystem_kernel.dylib         0x1820a1d84 __workq_kernreturn + 8
1  libsystem_pthread.dylib        0x18223feb4 _pthread_wqthread + 928
2  libsystem_pthread.dylib        0x18223fb08 start_wqthread + 4

#10. Thread
0  libsystem_kernel.dylib         0x1820a1d84 __workq_kernreturn + 8
1  libsystem_pthread.dylib        0x18223feb4 _pthread_wqthread + 928
2  libsystem_pthread.dylib        0x18223fb08 start_wqthread + 4
```

I don't have any further information at the moment, will update if so.

philips77, 2018-10-05:
What version of the DFU library are you using?
Btw, `enableUnsafeExperimentalButtonlessServiceInSecureDfu` is used only if your app based on SDK 12.x. And it's *false* by default.

Drusy, 2018-10-05:
I am using the last tag in date `4.1.3`

I agree, `enableUnsafeExperimentalButtonlessServiceInSecureDfu` is useless here, this is legacy code and should be removed, thanks for the tip

philips77, 2018-10-05:
Some comments:
1. Sorry for super late response. I just realized the issue was made in July.
2. 349th line in DFUPeripheral contains a comment in 4.1.3 tag: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/c63e1d64c352d3c2127843334b94439bdc02f132/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift#L349-L363
3. As far as I understand, the `peripheral` was set to *nil*. I don't think that's possible, as this is nullified only in `switchToNewPeripheralAndConnect()`.
4. Then it may be an unrelated reason that the object became *nil*, despite being non-weak reference. I can imaging Bluetooth goes off (also shouldn't make it *nil*), or setting the peripheral to *nil* by low memory, or some other weird reason.

So I made a PR that may at least fix the crash.

philips77, 2019-01-29:
I assume the issue was fixed. Closing it now. Feel free to reopen or create another one if more help is needed.

closed 2019-01-29 by philips77

Drusy, 2019-01-29:
Sorry for the late reply, I am not able to reproduce the crash myself but crashlytics didn't report anything since then, thank you 👍