nordicsemi/IOS-DFU-Library #204: 'DFUFirmware' is unavailable: cannot find Swift declaration for this class

mmajewski, 2018-07-24:
Hello,

I added `iOSDFULibrary` to my project using Carthage. Everything looks ok, when I try to use it in application directly. When I add `iOSDFULibrary.framework` to dynamic library, I've got swift compiler error `'DFUFirmware' is unavailable: cannot find Swift declaration for this class`.
[image]

I prepared [demo application](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/2223563/DFUTest.zip) to show you this error - maybe there is some missing configuration in framework project? 
I will be grateful if you can help me :)

mostafaberg, 2018-07-24:
hi @mmajewski ! sorry to hear about that, let me take a quick look into your project and get back to you!

mostafaberg, 2018-07-24:
Your project looks a bit wrong to me, how did you set this up ?
if you follow the steps in the readme you'll have no issue, I just did that and it works flawlessly, I've attached my project and you may see the code [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/2223772/dfuApp.zip)

if this is not the setup you want let me know

mmajewski, 2018-07-24:
This is not the setup I want to achieve;)

In my project I have application and 7 dynamic frameworks. `iOSDFULibrary` should be added to dynamic library, not to application itself.
In my demo application I created workspace (`DFUTest.xcworkspace`) with application project (`DFUTest`) and dynamic framework (`DFUFramework`). 
Framework works good (it prints "it's working" in the xcode console), but when I trying to use some class from `iOSDFULibrary`, compilation error occurs.

mostafaberg, 2018-07-24:
Yeah I see your point, let me take a deeper look into this tomorrow and get back to you ASAP!

mostafaberg, 2018-07-25:
I've looked a bit at that now, and I'm still a bit confused on why the `DFUFramework` project is there? Isn't it possible to add our library directly to your project as a dynamic framework?

how are the other dynamic frameworks added to your project ?

mmajewski, 2018-07-25:
Unfortunately, I can't add `iOSDFULibrary` directly to application project. I'm creating dynamic framework, which will be shared between several apps, so I need to add firmware update feature to this framework.
Other dynamic frameworks (like, for example, [AWSCore](https://github.com/aws/aws-sdk-ios)) are added in the same way - with Carthage - and there are no problems with them.

philips77, 2019-01-29:
Hi, did you manage to solve the issue? It's been a while since the last post.

philips77, 2019-02-26:
Closing due to inactivity.

closed 2019-02-26 by philips77