nordicsemi/IOS-DFU-Library #205: Disconnects/Error 308

lukecorbin, 2018-07-25:
Hello,

I have seen others with this issue, but none of the solutions presented have seemed to fix this in our implementation.

Pods DFU Library - Version 4.1.3

Error Log - Here is an example of the disconnects we are seeing.  Sometimes we see a CRC error, sometimes not. Any ideas on how to modify things such that we do not see these disconnects and with Error 308?

2018-07-25 11:39:34.786564-0400 bdx[7126:2777353] V: Discovering characteristics in DFU Service...
2018-07-25 11:39:34.786761-0400 bdx[7126:2777353] D: peripheral.discoverCharacteristics(nil, for: FE59)
2018-07-25 11:39:34.787869-0400 bdx[7126:2777353] --> Changed state to: Connecting
2018-07-25 11:39:34.830618-0400 bdx[7126:2777427] I: DFU characteristics discovered
2018-07-25 11:39:34.831205-0400 bdx[7126:2777353] --> Changed state to: Starting
2018-07-25 11:39:34.831976-0400 bdx[7126:2777427] V: Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-07-25 11:39:34.832650-0400 bdx[7126:2777427] D: peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
2018-07-25 11:39:34.836652-0400 bdx[7126:2777427] I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60030100f000006182f7b9
2018-07-25 11:39:34.837042-0400 bdx[7126:2777427] A: Checksum (Offset = 61440, CRC = B9F78261) received
2018-07-25 11:39:34.889084-0400 bdx[7126:2777404] V: Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-07-25 11:39:34.889295-0400 bdx[7126:2777404] A: Secure DFU Control Point notifications enabled
2018-07-25 11:39:34.890337-0400 bdx[7126:2777404] V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-07-25 11:39:34.891257-0400 bdx[7126:2777404] D: peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-07-25 11:39:34.949036-0400 bdx[7126:2777427] I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-07-25 11:39:34.979875-0400 bdx[7126:2777404] I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 6006010001000089000000ecd8c3cc
2018-07-25 11:39:34.980375-0400 bdx[7126:2777404] A: Command object info (Max size = 256, Offset = 137, CRC = CCC3D8EC) received
2018-07-25 11:39:34.981684-0400 bdx[7126:2777404] A: Received CRC match Init packet
2018-07-25 11:39:34.982021-0400 bdx[7126:2777404] V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-07-25 11:39:34.982989-0400 bdx[7126:2777404] D: peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-07-25 11:39:35.039007-0400 bdx[7126:2777427] I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-07-25 11:39:35.039951-0400 bdx[7126:2777427] I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600401
2018-07-25 11:39:35.040206-0400 bdx[7126:2777427] A: Command object executed
2018-07-25 11:39:35.040545-0400 bdx[7126:2777427] V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-07-25 11:39:35.040933-0400 bdx[7126:2777427] D: peripheral.writeValue(0x0602, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-07-25 11:39:35.098440-0400 bdx[7126:2777404] I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-07-25 11:39:35.099343-0400 bdx[7126:2777404] I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 6006010010000000f000006182f7b9
2018-07-25 11:39:35.099544-0400 bdx[7126:2777404] A: Data object info (Max size = 4096, Offset = 61440, CRC = B9F78261) received
2018-07-25 11:39:35.099795-0400 bdx[7126:2777353] --> Changed state to: Uploading
2018-07-25 11:39:35.153654-0400 bdx[7126:2777404] I: 61440 bytes of data sent before, CRC match
2018-07-25 11:39:35.153763-0400 bdx[7126:2777404] I: Resuming uploading firmware...
2018-07-25 11:39:35.153919-0400 bdx[7126:2777404] V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-07-25 11:39:35.154049-0400 bdx[7126:2777404] D: peripheral.writeValue(0x020c00, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-07-25 11:39:35.188392-0400 bdx[7126:2777427] I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-07-25 11:39:35.189336-0400 bdx[7126:2777427] I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
2018-07-25 11:39:35.189490-0400 bdx[7126:2777427] A: Packet Receipt Notif enabled (Op Code = 2, Value = 12)
2018-07-25 11:39:35.189883-0400 bdx[7126:2777427] V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-07-25 11:39:35.190189-0400 bdx[7126:2777427] D: peripheral.writeValue(0x010200100000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-07-25 11:39:35.249075-0400 bdx[7126:2777404] I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2018-07-25 11:39:35.250446-0400 bdx[7126:2777404] I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
2018-07-25 11:39:35.250720-0400 bdx[7126:2777404] I: Data object 16/17 created
2018-07-25 11:39:35.251092-0400 bdx[7126:2777404] A: Uploading firmware...
2018-07-25 11:39:35.251357-0400 bdx[7126:2777404] V: Sending firmware to DFU Packet characteristic...
2018-07-25 11:39:35.252554-0400 bdx[7126:2777353] --> DFU Progress: 0.0
2018-07-25 11:39:35.254044-0400 bdx[7126:2777353] --> DFU Progress: 0.9
2018-07-25 11:39:35.264964-0400 bdx[7126:2777353] --> DFU Progress: 0.91
2018-07-25 11:39:35.429520-0400 bdx[7126:2777427] V: Disconnecting...
2018-07-25 11:39:35.429755-0400 bdx[7126:2777427] D: centralManager.cancelPeripheralConnection(peripheral)
2018-07-25 11:39:35.433144-0400 bdx[7126:2777401] D: [Callback] Central Manager did disconnect peripheral
2018-07-25 11:39:35.433342-0400 bdx[7126:2777401] I: Disconnected
2018-07-25 11:39:35.433703-0400 bdx[7126:2777401] -------------------- DISCONNECTED --------------------
2018-07-25 11:39:35.434505-0400 bdx[7126:2777353] DFU Error 308: 61680 bytes were sent while 57584 bytes were reported as received

philips77, 2018-09-13:
I see 2 issues here:
1. Why did you get a notification just after enabling notifications:
```
Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
2018-07-25 11:39:34.832650-0400 bdx[7126:2777427] D: peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
// Here:
2018-07-25 11:39:34.836652-0400 bdx[7126:2777427] I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60030100f000006182f7b9
2018-07-25 11:39:34.837042-0400 bdx[7126:2777427] A: Checksum (Offset = 61440, CRC = B9F78261) received
// End
2018-07-25 11:39:34.889084-0400 bdx[7126:2777404] V: Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
```
This should not happen, according to my knowledge. I guess, based on what I wrote in 2., this is a notification requested in a previous connection after 61440 byte was sent. 

2. The notification mentioned in 1., and the correct notification received a little later both report that 61440 bytes were sent before and the CRC matched. After receiving the correct notification (in Data Object Info), the library should Execute the image (as 61440 divides by object size = 4096) to make sure the data sent before were executed (double executing does not do any harm, but no Execute will erase the last page). It didn't. **This is a bug in the library, at least it looks like it. I'll look into it right now.**

You were unlucky, that the previous disconnection happened just after sending a whole page, but before it was executed. This of course could happen, but the library should handle this case.

philips77, 2018-09-18:
Hi, could you check if the latest changes fixed the issue on your side? Thanks!

lukecorbin, 2018-09-19:
I will try to get this implemented and give you some feedback. I am traveling so it might be a few days.

mostafaberg, 2018-10-09:
@lukecorbin Have you gotten time to look into it ? we have a couple updates that would be nice if you try out

philips77, 2019-01-29:
I hope you manage to solve the issue. I'm closing it due to inactivity. Feel free to reopen or create a new one if more help is needed.

PR #216 (2018-09-18): Bugfix/205
This PR fixed #205 issue, and perhaps helps with #213.
Also, is adds a more DFU automated tests, covering SDK 15 and 15.2 + some older SDKs due to a MBR workaround.
closed 2019-01-29 by philips77