nordicsemi/IOS-DFU-Library #209: Error 4: Operation Failed when EnablingDfuMode (Indication value 0x 200204)

jagan, 2018-08-21:
Hi

I am integrating your DFU library with our mobile app to do FOTA to our Nordic device. Firmware update using zip file is failing for 1 out of 5 devices with the following error, just after changing state to 'EnablingDfuMode':

> [0:] DFU Log: Info Connected to DEVICE_CA45043D
> [0:] DFU Log: Verbose Discovering services...
> [0:] DFU Log: Debug peripheral.discoverServices(nil)
> [0:] DFU Log: Info Services discovered
> [0:] DFU Log: Verbose Starting Secure DFU...
> [0:] DFU Log: Info Connected to DEVICE_CA45043D
> [0:] DFU Log: Info Services discovered
> [0:] DFU Log: Verbose Secure DFU Service found
> [0:] DFU Log: Verbose Discovering characteristics in DFU Service...
> [0:] DFU Log: Debug peripheral.discoverCharacteristics(nil, for: FE59)
> [0:] DFU StateDidChangeTo: Connecting
> [0:] DFU Log: Info DFU characteristics discovered
> [0:] DFU Log: Verbose Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
> [0:] DFU Log: Debug peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
> [0:] DFU StateDidChangeTo: Starting
> [0:] DFU Log: Verbose Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
> [0:] DFU Log: Application Buttonless DFU indications enabled
> [0:] DFU Log: Warning Application with buttonless update found
> [0:] DFU Log: Verbose Trying setting bootloader name to Dfu39100
> [0:] DFU Log: Verbose Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
> [0:] DFU Log: Debug peripheral.writeValue(0x02084466753339313030, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
> [0:] DFU StateDidChangeTo: EnablingDfuMode
> [0:] DFU Log: Info Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
> [0:] DFU Log: Info Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200204
> [0:] DFU Log: Error Error 4: Operation failed
> [0:] DFU Log: Verbose Disconnecting...
> [0:] DFU Log: Debug centralManager.cancelPeripheralConnection(peripheral)
> [0:] DFU Log: Debug [Callback] Central Manager did disconnect peripheral
> [0:] DFU Log: Info Disconnected
> [0:] DFU Error: RemoteButtonlessDFUOperationFailed | Operation failed

Once this error occurs in a device, subsequent retries of Firmware Update fail with the same error. Now if I take the failing device and try flashing from Android device using Nordic's Android library, it resumes the flashing from, say, 24% and then proceeds until 100% and then shows 'Verification Failed' error. Then in the next retry (from Android), it will flash from the beginning and succeed.

As mentioned in another issue, I tried setting the PacketReceiptNotificationParameter to 1:

```
            _initiator = new DFUServiceInitiator(centralManager: centralManager, target: peripheral);
            _initiator.WithFirmware(file: selectedFirmware);
            // initiator.ForceDfu = true/false; // default false
            _initiator.PacketReceiptNotificationParameter = 1; // default is 12
            //_initiator.AlternativeAdvertisingNameEnabled = true; // default is true
            //_initiator.EnableUnsafeExperimentalButtonlessServiceInSecureDfu = true;

            _initiator.Delegate = new MyDfuServiceStateDelegate(progressListener); // to be informed about current state and errors 
            _initiator.ProgressDelegate = new MyDfuProgressDelegate(progressListener); // to show progress bar
            _initiator.LoggerDelegate = new MyDfuLoggerDelegate(progressListener); // to show log messages
            _initiator.PeripheralSelectorDelegate = new MyDfuPeripheralSelectorDelegate(progressListener); // not using the default selector

            _controller = _initiator.Start();
```
It succeeded on one occasion, but today that solution is also not helping.

Can you please let me know if there is any fix/work-around I can try? For instance, is there any flag that I set so as to instruct the library to restart flashing from beginning instead of trying to resume from where it left off?

The enum value of the 'Operation Failed' error is:

`        RemoteButtonlessDFUOperationFailed = 34,
`

philips77, 2018-08-21:
Which SDK are you using?

philips77, 2018-08-21:
Try setting AlternativeAdvertisingNameEnabled to false. This uses a feature from SDK 14, but if you're using an older SDK it won't work either way. With this param set to false, it will not try to set bootloader name, but will send 0x01 to jump to the bootloader immediately.

There's no API to cancel resume and start from the beginning. It should not be required.

jagan, 2018-08-21:
Wow, thanks for the fast reply! :) We are using SDK 14.2. I will try setting AlternativeAdvertisingNameEnabled to false and let you know the result.

closed 2018-09-13 by philips77