nordicsemi/IOS-DFU-Library #212: Error 309: CRC error

SihemMED, 2018-09-10:
Hello, 

I integrated your firmware (iOSDFULibrary) in our project but I got error when I tried to make the DFU to our Nordic device.

Error : Error raised when the CRC reported by the remote device does not match. Service has done 3 tries to send the data.

Please could you help me to understand this problem and to solve it.

philips77, 2018-09-13:
Hi, could you post some logs from the library here?
Did you try updating using nRF Toolbox or nRF Connect on iPhone? Does it work?
What iOS version and phone are you testing on? I got some issues on iOS 11.4 related to CRC errors. Did you try on older iOS versions?
When does the error happen? At the beginning of the upload, in the middle or after the last object was sent? Logs will help figure that out.

SihemMED, 2018-09-13:
Hi, thank you for your response.
I set the packetReceiptNotificationParameter = 12 for DFUServiceInitiator and it works fine now. I think the problem is related to my code not to the firmware.

philips77, 2018-09-13:
Seems like you have just slow the upload down, which gives more time to the dfu bootloader to do flash operations.

mostafaberg, 2018-10-05:
@SihemMED is your problem resolved after slowing down the process ?

SihemMED, 2018-10-05:
@mostafaberg yes it solved.

closed 2018-10-05 by mostafaberg