nordicsemi/IOS-DFU-Library #213: Force unwrapping crash 😱

DevAndArtist, 2018-09-12:
Hi Nordic fellas, as I previously mentioned my concerns about the [overkilling misuse of force unwraps](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/192) in the framework I finally hit the first crash that I could catch and have no idea how to fix. This requires an emergency fix since our app is already production ready.

[image]

philips77, 2018-09-12:
I'll have a look at it first thing tomorrow morning.

philips77, 2018-09-13:
Hello @DevAndArtist,

As I see on the logs, the problem is not with `firmwareRanges` being unwrapped, but with the whole DFU process. 

On the screen you attached it can be seen, that the phone requested CRC (op code 0x03), and after the valid CRC was received it sent Execute command (0x04). However, the next notification that was received is again the CRC (why?!). In here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/9afd17199e0542b98634556f332128475cf505cf/iOSDFULibrary/Classes/Implementation/SecureDFU/Characteristics/SecureDFUControlPoint.swift#L490-L492 the same response callback is called (as it hasn't been cleared here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/9afd17199e0542b98634556f332128475cf505cf/iOSDFULibrary/Classes/Implementation/SecureDFU/Characteristics/SecureDFUControlPoint.swift#L359-L364 which sends the Execute command again.

What happens after that is even weirder. To make the `firmwareRanges` **nil**, the code must have reached https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/9afd17199e0542b98634556f332128475cf505cf/iOSDFULibrary/Classes/Implementation/SecureDFU/DFU/SecureDFUExecutor.swift#L292-L298
The only place from where it is called is here:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/9afd17199e0542b98634556f332128475cf505cf/iOSDFULibrary/Classes/Implementation/SecureDFU/DFU/SecureDFUExecutor.swift#L64-L75
and this is called when... characteristics become discovered. But in that case, the log would also contain "DFU characteristics discovered", which is not written, so I guess that an old callback has been called incorrectly due to this invalid notifications. But I can't see any way this callback could have been called.

Are you able to reproduce this issue? Do you have all the logs, even from before those visible on the right?

philips77, 2018-09-18:
Hi, did you try library with the latest changes?

DevAndArtist, 2018-09-20:
I could only reproduce (randomly) the issue one more time, but I couldn't gather any useful logs or stack trace. If it will come up again, I'll try to get more data for you.

We have really weird issues in our app during and after DFU updates. On my iPhone 6 Plus I almost never have any issues. On an iPhone 6S it seems to crash the bluetooth stack after DFU (the app does not crash though) - it does send the `complete` state.

A quick summary of our app:
- Our peripheral has two modes (application / bootloader)
- We treat each mode as an own peripheral (due to limitations on iOS)
- There are two central managers, one handles application peripherals the other the ones in the bootloader
- We're using in both cases `RxBluetoothKit` framework for more convenient reactive API
- Since the DFU lib takes full control of the delegates of the central and the peripheral we had to [inject two proxy types](https://github.com/DevAndArtist/IOS-Pods-DFU-Library/tree/proxy-injection/iOSDFULibrary/Classes/Implementation/Proxies) which sends the events to the previous delegate and ensure correct restoration when the DFU library unlinks itself. (I can create a PR if you want me to include these proxies into the official library!?)

I'll try to merge your updates into our branch and see if I somehow will be able to reproduce the crash again.

philips77, 2018-10-05:
Any success with this issue?

DevAndArtist, 2018-10-05:
I could not reproduce it yet, I will update the framework to 4.2 and test it next week. If you want you can close the ticket for now, it the issue comes back I'll reopen it.

closed 2018-10-05 by philips77