nordicsemi/IOS-DFU-Library #214: OC project can not use pod install iOSDFULibrary

FristResponder, 2018-09-13:
hi,when I update iOSDFULibrary in my oc project,xcode has a few error.The pod can not find the ‘iOSDFULibrary’ lib.i test many times use different project,can you help me? if I had a mistake.thank you
[image]

FristResponder, 2018-09-13:
hi，this is a supplement picture
[image]

mostafaberg, 2018-10-05:
still having the issue ? if we don' there back within next week we'll assume it's resolved and close the issue :)

closed 2018-10-09 by mostafaberg