nordicsemi/IOS-DFU-Library #222: cbcentralmanagerstate

MMooreGitHub, 2018-09-19:
Good Afternoon,

As of iOS 10, cbcentralmanagerstate is [deprecated](https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerstate).

Are there any plans to replace it?

Thanks

mostafaberg, 2018-10-05:
This is not a problem, you don't access `CBCentralManagerState` enum itself but it's values, the actual `CBCentralManagerState` property is now deprecated and iOS relies on the previously available `CBManagerState` property instead, this is now shared between `CBCentralManager` and `CBManager` objects, which helps unify the API.

mostafaberg, 2018-10-05:
Just to clarify, `CBCentralManagerState` and `CBManagerState` have the exact same values, so `.poweredOn`, `.poweredOff` can be used as is, unless you set the variable type to `CBCentralManagerState`, which is not really needed in the first place :)

So your code can be like:
```
let manager = CBCentralManager(....)

switch manager.state {
    case .poweredOn:
        print("Powered on")
    case .poweredOff:
       print("Powered off")
    default:
      print(manager.state.rawValue)
}

closed 2018-10-05 by mostafaberg

MMooreGitHub, 2018-10-08:
Ok, sounds good.
Thanks, @mostafaberg!