nordicsemi/IOS-DFU-Library #226: Random: DFU Control Point characteristic not found

DevAndArtist, 2018-10-04:
I do get from time to time a random error on first attempt of flashing our peripheral.

```
DFU State: Connecting

Message: Connecting to Peripheral Boot...
Message: centralManager.connect(peripheral, options: nil)
Message: [Callback] Central Manager did connect peripheral
Message: Connected to Peripheral Boot
Message: Discovering services...
Message: peripheral.discoverServices(nil)
Message: [Callback] Central Manager did connect peripheral
Message: Connected to Peripheral Boot
Message: Discovering services...
Message: peripheral.discoverServices(nil)
Message: Services discovered
Message: Starting Secure DFU...
Message: Connected to Peripheral Boot
Message: Services discovered
Message: Secure DFU Service found
Message: Discovering characteristics in DFU Service...
Message: peripheral.discoverCharacteristics(nil, for: FE59)
Message: DFU characteristics discovered
Message: DFU Control Point characteristics not found
Message: Disconnecting...
Message: centralManager.cancelPeripheralConnection(peripheral)
Message: [Callback] Central Manager did disconnect peripheral
Message: Disconnected

Error: DFUError Message: DFU Control Point characteristic not found
```

Then on the second try the flash operation starts as expected:

```
🔸 DFU State: Connecting

Message: Connected to Peripheral Boot
Message: Services discovered
Message: Starting Secure DFU...
Message: Connected to Peripheral Boot
Message: Services discovered
Message: Secure DFU Service found
Message: Discovering characteristics in DFU Service...
Message: peripheral.discoverCharacteristics(nil, for: FE59)
Message: DFU characteristics discovered
Message: MTU set to 183
Message: Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
Message: peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)

🔸 DFU State: Starting

Message: Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
Message: Secure DFU Control Point notifications enabled
Message: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
Message: peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Message: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Message: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601000100000000000000000000
Message: Command object info (Max size = 256, Offset = 0, CRC = 00000000) received
Message: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
Message: peripheral.writeValue(0x010188000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Message: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Message: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
Message: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
Message: peripheral.writeValue(0x020000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Message: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Message: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
Message: Packet Receipt Notif disabled (Op Code = 2, Value = 0)
Message: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
Message: peripheral.writeValue(## REMOVE FROM PUBLIC LOG ##, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
[... 261 lines cut ...]
🔷 Progress: 83 Current Speed: 167288.88101041436 Bytes/s Average Speed: 7105.12975869707 Bytes/s
🔷 Progress: 84 Current Speed: 59582.88375029595 Bytes/s Average Speed: 7185.995359029455 Bytes/s
🔷 Progress: 85 Current Speed: 62100.78101544346 Bytes/s Average Speed: 7267.065458754206 Bytes/s
Message: [Callback] Central Manager did disconnect peripheral
Message: Disconnected by the remote device
🔷 Progress: 86 Current Speed: 1144.7410397070694 Bytes/s Average Speed: 6864.94400461906 Bytes/s
🔷 Progress: 87 Current Speed: 84406.58728827772 Bytes/s Average Speed: 6943.55471667404 Bytes/s
🔷 Progress: 88 Current Speed: 49717.635205215585 Bytes/s Average Speed: 7017.04779810307 Bytes/s
🔷 Progress: 89 Current Speed: 65632.85403807703 Bytes/s Average Speed: 7093.238997509166 Bytes/s
🔷 Progress: 90 Current Speed: 72565.81314878893 Bytes/s Average Speed: 7170.121662238265 Bytes/s
🔷 Progress: 91 Current Speed: 67177.53436846554 Bytes/s Average Speed: 7246.142278041116 Bytes/s
🔷 Progress: 92 Current Speed: 64423.13507978497 Bytes/s Average Speed: 7283.883263799624 Bytes/s
🔷 Progress: 93 Current Speed: 1351.7773020038755 Bytes/s Average Speed: 6932.752620315001 Bytes/s
🔷 Progress: 94 Current Speed: 40277.67021886712 Bytes/s Average Speed: 6998.862808730261 Bytes/s
🔷 Progress: 95 Current Speed: 90135.47277936962 Bytes/s Average Speed: 7072.452175744495 Bytes/s
🔷 Progress: 96 Current Speed: 79347.82521873932 Bytes/s Average Speed: 7145.0522969372605 Bytes/s
🔷 Progress: 97 Current Speed: 39495.42099343465 Bytes/s Average Speed: 7210.205979472791 Bytes/s
🔷 Progress: 98 Current Speed: 1203.4103561927418 Bytes/s Average Speed: 6880.80391396526 Bytes/s
🔷 Progress: 99 Current Speed: 60301.49520766773 Bytes/s Average Speed: 6947.3240183758335 Bytes/s
🔷 Progress: 100 Current Speed: 37523.687683120486 Bytes/s Average Speed: 6981.27847409135 Bytes/s
🔸 DFU State: Disconnecting
🔸 DFU State: Completed
```

Any suggestions what could be wrong?

philips77, 2018-10-05:
Looks like a caching issue. iOS caches services of non-bonded devices if they didn't have Services Changed characteristic in Generic Attribute service last time they connected.

philips77, 2018-10-05:
The question remains: why it only failed to get the characteristic, but the service was found properly. Are you sure your device responds correctly to service discovery? Do you have a mode, where the DFU service contains some other characteristics? I'll improve logging for this case.

DevAndArtist, 2018-10-05:
Our device obviously switches between two modes: application / bootloader

In our iOS app we handle those modes differently. There is a wrapper object for the physical peripheral which can contain up to two CBPeripheral objects (one for application one for bootloader). In our app we also have two central managers so that the peripheral objects don't get mixed up. One important point is that THIS library takes full control of the delegates from the peripheral and central manager which breaks the communication between the app and other connected peripherals. That was the reason to offload the bootloader peripherals to a separate central manager and only give the control to this DFU lib over that central manager. That said the other peripherals in the application mode remain fully functional even during DFU update of a different peripheral.

philips77, 2018-10-05:
I wanted to fix this delegates issue for a long time. I think it's the time. Expect a PR soon to review.

DevAndArtist, 2018-10-05:
@philips77 I build a custom workaround for this in a fork: https://github.com/DevAndArtist/IOS-Pods-DFU-Library/tree/proxy-injection

Ignore some of the commits I had there, I was trying to find an issue which I had with proxies but it required to always pull and compile with carthage. That's why I had a few weird `pirint`'s here and there.

I think I only included the proxies in the part of the lib our app uses so I might not replaced all delegates. Proxies are build with one simple assumption in mind: The delegates are not re-assigned from outside this library during the time it's actively doing something. If there was an observer before, it will be saved and restored properly when this library no longer needs any control. That way the previous delegate also receives all events (this is required to know if the peripheral is connected / reconnection etc.).

philips77, 2018-10-05:
Have a look at #228. I think that's much simpler solution.

DevAndArtist, 2018-10-05:
I had a quick glance at the implementation, I'm not quite sure it solves the issue for our application because we still lose important information. For instance in our App everything is piped trough RxBluetoothKit for more convenient API. In our app we have a small algorithm witch checks if the paired peripheral is not connected but set as active. In that case it will always try to re-connect. Since the solution in your PR also need to connect I assume that the peripheral from my central manager will receive a disconnection event which will trigger that mechanism and then my central manager will try to get the control back over the peripheral.

The solution I had above is a little bit more complex, that's a fact, but it allows us to observe connection events without breaking any logic.

philips77, 2018-10-05:
No, you may be connected multiple times to the same device. When you cancel one connection, you will get peripheralDidDisconnect only in this CBPeripheral.delegate. But all other will still be connected. They may be used by one or more apps. Only after the last CBPeripheral cancels connection, or the Disconnect Request is sent from the peripheral side, the physical connection is cancelled.

philips77, 2018-10-05:
Your CBPeripheral's delegate will get `peripheralDidDisconnect` during the update, when the device switches to Bootloader mode, or restarts when part was sent.

philips77, 2018-10-05:
>  If you have two central managers where each is connected to the same peripheral (same UUID) if one of the central manager calls connect method then the other central manager would have an invalid peripheral object, no?

No. The other will have a valid instance, will still be able to send data, receive notifications, etc. You may cancel its connection, connect again, etc, without any impact on the other CBPeripheral. The OS, each CBPeripheral treats as a virtual connection. Interesting case is with notifications. You may call [setNotifyValue(:forCharacteristic)](https://developer.apple.com/documentation/corebluetooth/cbperipheral/1518949-setnotifyvalue) with *true* on both CBPeripherals and both will start getting notifications. If you call it only on one, only this one will be getting them. If you, after enabling them on both, disable on one, only this one will stop getting them. The other will not know whether the CCCD was actually written with 0x00-00 or not, as it doesn't know if it's the only CBPeripheral instance, or not.

You will get disconnect even in your CBPeripheral during the DFU process. Either during jumping to the bootloader, or after the reset. You would still get it now, with your approach. But you will get it only once, unless you start connecting again with your instance of CBPeripheral (which many be unreachable, as the device may be in DFU bootloader mode, with different address).

DevAndArtist, 2018-10-05:
Okay that sounds interesting, I still need some time to wrap my head around this, but maybe you're right and the new approach will fix a bunch of issues 👍 

The issue with different address is not present in our iOS application because we do search for the peripheral by serial numbers (after it was paired once) which is always unique. Even for the DFU lib we have a peripheral selector that overrides the default behavior.

```swift
final class PeripheralSelector : DFUPeripheralSelectorDelegate {
  let serialNumber: SerialNumber

  init(serialNumber: SerialNumber) {
    self.serialNumber = serialNumber
  }

  func select(
    _ peripheral: CBPeripheral,
    advertisementData: [String : AnyObject],
    RSSI: NSNumber,
    hint name: String? = nil
  ) -> Bool {
    let key = CBAdvertisementDataManufacturerDataKey
    // Extract the manufacturer data
    if let data = advertisementData[key] as? Data {
      // The first tow bytes are reserved so drop them
      let serialNumberData = Data(data.dropFirst(2))
      // Convert the last 4 bytes into the serial number string
      guard
        let serialNumber = convertToSerialNumber(from: serialNumberData)
      else { return false }
      // If the serial numbers matches then we return true or false otherwise
      return serialNumber == self.serialNumber
    }
    return false
  }

  func filterBy(hint dfuServiceUUID: CBUUID) -> [CBUUID]? {
    return [dfuServiceUUID]
  }
}
```

philips77, 2018-10-05:
Btw, release 4.2 is out!

philips77, 2018-10-05:
Ok, let's close the issue now. If you find any other issues with the new release, feel free to create a new one.

closed 2018-10-05 by philips77