nordicsemi/IOS-DFU-Library #230: CentralManager instance after the DFU completed can't be used to start scanning and receive events

blackcodefr, 2018-10-05:
CentralManager instance after the DFU completed can't be used to start scanning and receive events.

I have to create a new one to be able to start scanning and connecting and getting delegate results reliably. Do you do anything particular to the centralManager passed to you after dfu completion?

Let me know if you need more details about the later behaviour I'm describing.

Thanks for your help.

philips77, 2018-10-05:
Hi, you have to restore the `CBCentralManager.delegate`, as it was modified and nullified by the lib. We are soon releasing version 4.2 where the lib no longer needs the `CBCentralManager` instance. Should be published today.

mostafaberg, 2018-10-05:
that's because the library unfortunately at the time of writing takes over the centralManager and it's delegate causing this, the new upcoming version resolves this issue.

if you do not want to update the the new version straight away when it's out, just hold a reference to the central manager and it's delegate before you start the DFU initialiser, after it's complete, simply restore the centralManager's delegate as soon as you're completed, and everything should work with no issues

blackcodefr, 2018-10-05:
Yes, I do that but it doesn't seem to work.

```
CBCentralManager *manager = [BluetoothControllersManager sharedInstancewithDelegate:nil].centralManager;
manager.delegate = [BluetoothControllersManager sharedInstancewithDelegate:nil];
[[BluetoothControllersManager sharedInstancewithDelegate:nil] startScanningForDevices];
```

mostafaberg, 2018-10-05:
Couple of questions, 
1-Are you sure the delegate is properly set on the second line?
2-Can you show how you initialise the DFU process ?

I have a feeling the issue might be in the underlying code in `BluetoothControllersManager` and not in our library

blackcodefr, 2018-10-05:
oh.. Yes indeed...

blackcodefr, 2018-10-05:
```

@property (atomic, strong) CBCentralManager *centralManager;

static BluetoothControllersManager *sharedInstance;

+ (BluetoothControllersManager *)sharedInstancewithDelegate:(id)delegate {
    static dispatch_once_t onceToken;
    dispatch_once(&onceToken, ^{
        if (!centralQueue) {
            centralQueue = dispatch_queue_create("{{queue_name}}", DISPATCH_QUEUE_SERIAL);
        }
        sharedInstance = [[BluetoothControllersManager alloc] init];
        sharedInstance.centralManager = [[CBCentralManager alloc] initWithDelegate:sharedInstance queue:centralQueue];
        sharedInstance.delegate = delegate;
        sharedInstance.availablePeripherals = [[NSMutableArray alloc] init];
        [sharedInstance registerForApplicationStatusNotifications];
        sharedInstance.moduleCache = [[NSMutableArray alloc] init];
        sharedInstance.modulesToIgnore = [[NSMutableArray alloc] init];
        sharedInstance.nearbyModules = [[NSMutableArray alloc] init];
    });
    if (delegate) {
        sharedInstance.delegate = delegate;
    }
    return sharedInstance;
}
```

blackcodefr, 2018-10-05:
```
CBCentralManager *manager = [BluetoothControllersManager sharedInstancewithDelegate:nil].centralManager;
manager.delegate = [BluetoothControllersManager sharedInstancewithDelegate:nil];
[[BluetoothControllersManager sharedInstancewithDelegate:nil] startScanningForDevices];
```

So, the code fetches the centralManager stored in the singleton sharedInstance and sets the sharedInstance as the delegate for the centralManager.

That is what you expect us to do right?

blackcodefr, 2018-10-05:
Anyway, if a new version of the app hands back control of the centralManager automatically I will gladly update to remove this logic. Still, in its current state something doesn't seems right.

blackcodefr, 2018-10-05:
```
- (void)initiateDFUControllerForController:(Controller *)controller {
    _isPerformingDFUUpdate = YES;
    [self startProgress];
    
    [self stopScanning];
    [[BluetoothControllersManager sharedInstancewithDelegate:nil] stopScanning];
    CBCentralManager *manager = [SLMBluetoothUpdateManager sharedInstance].centralManager;
    if (self.moduleToUpdate.typeUsesNewDFU) {
        manager = [BluetoothControllersManager sharedInstancewithDelegate:nil].centralManager;
        [[BluetoothControllersManager sharedInstancewithDelegate:nil] stopScanning];
    }
    NSArray *peripherals = [manager retrievePeripheralsWithIdentifiers:@[[[NSUUID alloc] initWithUUIDString:self.moduleToUpdate.hardwareAddress]]];
    self.shouldRetry = YES;
    
    DFUFirmware *selectedFirmware = [[DFUFirmware alloc] initWithUrlToZipFile:[self getUpdateFileForDevice:self.moduleToUpdate]];
    
    self.dfuServiceInitiator = [[DFUServiceInitiator alloc] initWithCentralManager:manager target:[peripherals lastObject]];
    self.dfuServiceInitiator = [self.dfuServiceInitiator withFirmware:selectedFirmware];
    // Optional:
    self.dfuServiceInitiator.forceDfu = NO; // default NO
    // initiator.packetReceiptNotificationParameter = N; // default is 12
    self.dfuServiceInitiator.logger = self; // - to get log info
    self.dfuServiceInitiator.delegate = self; // - to be informed about current state and errors
    self.dfuServiceInitiator.progressDelegate = self; // - to show progress bar
    
    self.dfuServiceInitiator.peripheralSelector = self; // the default selector is used
    self.dfuServiceInitiator.alternativeAdvertisingNameEnabled = NO;
    
    self.dfuServiceController = [self.dfuServiceInitiator start];
}

```

mostafaberg, 2018-10-05:
hmm this implementation is a bit confusing, but let's go with it anyways.
After the first time you init, you're manually setting the delegate to the shared instance which is ok, however, at that moment, you don't have any reference to the delegate still, so when our library takes over and completes the process, you don't have that stored anywhere.

What I would say, after the dfu service is completed and returns success or failure, you have to execute following code you showed first.

```
manager.delegate = [BluetoothControllersManager sharedInstancewithDelegate:nil];
[[BluetoothControllersManager sharedInstancewithDelegate:nil] startScanningForDevices];
```

is that what you're currently doing ?

mostafaberg, 2018-10-05:
also, is `[SLMBluetoothUpdateManager sharedInstance]` and `[BluetoothControllersManager sharedInstancewithDelegate:nil]` the same class ?

mostafaberg, 2018-10-05:
also, what's going on here, this is very unclear:
```
    [self stopScanning];
    [[BluetoothControllersManager sharedInstancewithDelegate:nil] stopScanning];
    CBCentralManager *manager = [SLMBluetoothUpdateManager sharedInstance].centralManager;
    if (self.moduleToUpdate.typeUsesNewDFU) {
        manager = [BluetoothControllersManager sharedInstancewithDelegate:nil].centralManager;
        [[BluetoothControllersManager sharedInstancewithDelegate:nil] stopScanning];
    }
```

You're doing the following:
1- Stop scanning
2- Tell the singleton to stop scanning
3- override the manager to even another centralManager from another instance
4- Stop that central too from scanning
5- Check if it's using the new dfu (what is this? is it secure vs legacy DFU ? if that's the case, we handle that in the library already.)
6- if it's using the new DFU, override it with the first central again.
7- then you ask the actual singleton to stop scanning again.

I see this is a very confusing piece of code that I'm not sure what's going on, I'd say if you are having lots of logic that's overriding our libraries functionality, we might not be able to support that as we will have to see lots more code than that before we can debug.

can you clarify a bit more ?

blackcodefr, 2018-10-05:
Yes, it is what I'm currently doing.

No, SLMBluetoothUpdateManager and BluetoothControllersManager are different singletons.

mostafaberg, 2018-10-05:
If they're different singletons, this will not work, the central manager that discovered and connected to a peripheral will be the only one that can reconnect and discover it. so as I said, you have some extra logic that's causing this and it's not a bug in the library unfortunately.

I'm not sure if our new version will help you out either as the issue is in your logic that handles the centrals

blackcodefr, 2018-10-05:
The manager that discovers .typeUsesNewDFU devices is not the BluetoothUpdateManager. Thus the code:

```
CBCentralManager *manager = [SLMBluetoothUpdateManager sharedInstance].centralManager;
    if (self.moduleToUpdate.typeUsesNewDFU) {
        manager = [BluetoothControllersManager sharedInstancewithDelegate:nil].centralManager;
        [[BluetoothControllersManager sharedInstancewithDelegate:nil] stopScanning];
    }

```

Before I hand the manager to the DFULibrary

mostafaberg, 2018-10-09:
I'm sorry we can't help, but this still is your own code that's causing this and it's not an issue in the library itself, I would suggest opening a ticket on DevZone and we can try to support you further there, but a solution might not be easy to catch without some refactoring on your side to also remove those singletons.

closed 2018-10-09 by mostafaberg

blackcodefr, 2018-10-09:
If you're sure there is no issue with the library I'm fine with that. I already found a solution before I opened this ticket, just wanted to make sure you didn't miss something is the library. Will be updating to the latest version when it becomes available.

mostafaberg, 2018-10-09:
100% sure it's not in the library :), the new version `4.2.1` is out so you should give it a try whenever you have time, thanks a lot for the feedback :)