nordicsemi/IOS-DFU-Library #238: `sendInitPacket(fromOffset:)` caused a crash

DevAndArtist, 2018-10-11:
I just flashed our peripheral a bunch times and it randomly crashed in the `SecureDFUExecutor`.

I do supply firmware files as `Data` instances using the following init (both bin and dat files are provided):

```swift
public init?(binFile: Data, datFile: Data?, type: DFUFirmwareType)
```

Here are some informations from the Debug navigator.

```
#0	0x00000001031b6d3c in SecureDFUExecutor.sendInitPacket(fromOffset:) at /IOS-Pods-DFU-Library/iOSDFULibrary/Classes/Implementation/SecureDFU/DFU/SecureDFUExecutor.swift:353
#1	0x00000001031b9898 in SecureDFUExecutor.peripheralDidSetPRNValue() [inlined] at /IOS-Pods-DFU-Library/iOSDFULibrary/Classes/Implementation/SecureDFU/DFU/SecureDFUExecutor.swift:129
#2	0x00000001031b988c in closure #1 in SecureDFUPeripheral.setPRNValue(_:) [inlined] at /IOS-Pods-DFU-Library/iOSDFULibrary/Classes/Implementation/SecureDFU/Peripheral/SecureDFUPeripheral.swift:156
#3	0x00000001031b988c in SecureDFUService.setPacketReceiptNotificationValue(_:onSuccess:onError:) [inlined] at /IOS-Pods-DFU-Library/iOSDFULibrary/Classes/Implementation/SecureDFU/Services/SecureDFUService.swift:226
#4	0x00000001031b988c in SecureDFUPeripheral.setPRNValue(_:) at /IOS-Pods-DFU-Library/iOSDFULibrary/Classes/Implementation/SecureDFU/Peripheral/SecureDFUPeripheral.swift:155
#5	0x00000001031ba194 in SecureDFUExecutor.peripheralDidCreateCommandObject() [inlined] at /IOS-Pods-DFU-Library/iOSDFULibrary/Classes/Implementation/SecureDFU/DFU/SecureDFUExecutor.swift:124
#6	0x00000001031ba18c in closure #1 in SecureDFUPeripheral.createCommandObject(withLength:) [inlined] at /IOS-Pods-DFU-Library/iOSDFULibrary/Classes/Implementation/SecureDFU/Peripheral/SecureDFUPeripheral.swift:128
```

[image]

As far as I could understand the issue myself here is what technically happens:

```swift
let data = Data(bytes: [0x00, 0x01])
let initPacketLength = data.count
let offset = data.count

// Fatal error: Can't form Range with upperBound < lowerBound
data.subdata(in: offset ..< (initPacketLength - offset))
```

philips77, 2018-10-12:
Do you have a full log from this event? I have a question about the notifications you get. They seem to be in very weird order. At the beginning, partially hidden, you sent 0x**01**0188000000 = Create Command object with size equal to 133 bytes (partially cut). For that request you get a response 0x60**02**01, which is PRNs had been set correctly.
Then, you write 0x**02**0000 to disable PRNs and you get 0x60**01**01 = Command Object created.
Later you get the PRN confirmation again. Something is in very odd state right now. As if two DFUs were done at the same time?

philips77, 2018-10-12:
Hi, despite of the issues described above, I found an issue with repeating sending Init Packet. The offset wan't zeroed which might have, according to my tracking abilities, cause your issue. Could you try adding those 2 lines to your code and try to recreate the issue?

I think that happens when you upload an Init Packet of 136 bytes (0x88) and the disconnect before DFU is complete. I'm not sure if restarting the chip will not erase unsaved data, so perhaps a link loss can be better. Then, in the 23nd attempt, you try to upload a different firmware with same-size-but-different Init Packet.

DevAndArtist, 2018-10-14:
I'll try to re-create the issue somehow tomorrow (on Monday) and if I find a re-producible way I'll check if the PR resolved it. I also would provide the missing logs, which I missed to save for the filed issue here, sorry.

DevAndArtist, 2018-10-15:
@philips77 so far, no luck. I cannot re-produce the crash (I did not used the fix from above during testing). Is it okay for you if we consider this issue as fixed now, I can still re-open the ticket if it re-appears. I'll update the framework to 4.2.2 then and test.

closed 2018-10-15 by philips77