nordicsemi/IOS-DFU-Library #239: New `init(target:queue:)` caused a crash

DevAndArtist, 2018-10-11:
Framework version: 4.2.1

The new init on `DFUServiceInitiator` caused a crash because of a wrong assumption. I'm fixing some weird issues in our app and I could force the framework to crash in the following scenario. `DFUServiceInitiator` was initialized while **bluetooth was off** which probably caused the system to return an empty array of known peripherals. `first!` will then of course crash. 

[image]

DevAndArtist, 2018-10-14:
@philips77 wouldn't it make more sense to let the `init` as non-failing init and rather save the `identifier` und try to access it in the already failable `start` method?

Furthermore you could retry more conveniently if the API would look something like this:

```swift
internal var identifier: UUID

// Allows 
func start(with identifier: UUID? = nil) -> DFUServiceController? {
  // Override the identifier if needed
  self.identifier = identifier ?? self.identifier
  guard 
    let peripheral = self.centralManager
      .retrievePeripherals(withIdentifiers: [self.identifier])
      .first 
  else { return nil  }
  // Save the target
  self.target = peripheral
  ...
}
```

That way event the target in the `init` is optional and it would make the whole `DFUServiceInitiator` object reusable. It's definitely more convenient to fail on a method instead on the init, but that's my personal preference I guess.

philips77, 2018-10-15:
Hi, how about that: #241?

DevAndArtist, 2018-10-15:
I had a quick look at the implementation, it looks good to me. I'll migrate and test it soon, thank you for fixing all my issues. 🤝 (First I need to re-produce the other crash I filed.)

closed 2018-10-15 by philips77