nordicsemi/IOS-DFU-Library #243: Write char failed when trying setting bootloader name

nvelozsavino, 2018-10-22:
I'm developing an app with the react-native and react-native-nordic-dfu library which uses this IOS-Pods-DFU-Library, After configuring everything and building the project, I'm getting an error from this library. When the library is trying to set the bootloader name it fails and disconnect.

I'm getting the following logs:

2018-10-22 10:47:29.967539+0200 UnlimitedSkate[5169:1339653] logWith: 5 message: 'Connected to Unlimited_Mesc'
2018-10-22 10:47:29.967746+0200 UnlimitedSkate[5169:1339653] logWith: 5 message: 'Services discovered'
2018-10-22 10:47:29.967833+0200 UnlimitedSkate[5169:1339653] logWith: 1 message: 'Starting Secure DFU...'
2018-10-22 10:47:29.968028+0200 UnlimitedSkate[5169:1339653] logWith: 5 message: 'Connected to Unlimited_Mesc'
2018-10-22 10:47:29.968149+0200 UnlimitedSkate[5169:1339653] logWith: 5 message: 'Services discovered'
2018-10-22 10:47:29.968233+0200 UnlimitedSkate[5169:1339653] logWith: 1 message: 'Secure DFU Service found'
2018-10-22 10:47:29.968337+0200 UnlimitedSkate[5169:1339653] logWith: 1 message: 'Discovering characteristics in DFU Service...'
2018-10-22 10:47:29.968494+0200 UnlimitedSkate[5169:1339653] logWith: 0 message: 'peripheral.discoverCharacteristics(nil, for: FE59)'
2018-10-22 10:47:29.970989+0200 UnlimitedSkate[5169:1338172] logWith: 5 message: 'DFU characteristics discovered'
2018-10-22 10:47:29.971378+0200 UnlimitedSkate[5169:1338172] logWith: 1 message: 'Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...'
2018-10-22 10:47:29.971492+0200 UnlimitedSkate[5169:1338172] logWith: 0 message: 'peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)'
2018-10-22 10:47:29.986692+0200 UnlimitedSkate[5169:1338172] logWith: 1 message: 'Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50'
2018-10-22 10:47:29.986849+0200 UnlimitedSkate[5169:1338172] logWith: 10 message: 'Buttonless DFU indications enabled'
2018-10-22 10:47:29.986935+0200 UnlimitedSkate[5169:1338172] logWith: 15 message: 'Application with buttonless update found'
2018-10-22 10:47:29.987098+0200 UnlimitedSkate[5169:1338172] logWith: 1 message: 'Trying setting bootloader name to Dfu16751'
2018-10-22 10:47:29.987232+0200 UnlimitedSkate[5169:1338172] logWith: 1 message: 'Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...'
2018-10-22 10:47:29.987500+0200 UnlimitedSkate[5169:1338172] logWith: 0 message: 'peripheral.writeValue(0x02084466753136373531, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)'
2018-10-22 10:47:30.046209+0200 UnlimitedSkate[5169:1338172] logWith: 20 message: 'Writing to characteristic failed'
2018-10-22 10:47:30.046413+0200 UnlimitedSkate[5169:1338172] logWith: 20 message: 'Error 3: Writing is not permitted.'
2018-10-22 10:47:30.046649+0200 UnlimitedSkate[5169:1338172] logWith: 1 message: 'Disconnecting...'
2018-10-22 10:47:30.046730+0200 UnlimitedSkate[5169:1338172] logWith: 0 message: 'centralManager.cancelPeripheralConnection(peripheral)'
2018-10-22 10:47:30.048176+0200 UnlimitedSkate[5169:1338172] logWith: 0 message: '[Callback] Central Manager did disconnect peripheral'
2018-10-22 10:47:30.048321+0200 UnlimitedSkate[5169:1338172] logWith: 5 message: 'Disconnected'
2018-10-22 10:47:30.048554+0200 UnlimitedSkate[5169:1338172] onDFUError

It's important to note that the Android counterpart of this library works flawless. Any ideas?

nvelozsavino, 2018-10-22:
I tried to replicate the issue with the nrf connect app sending some commands manually
When I connect to the device, enable indication on Butonless DFU characteristic and send the command to enter bootloader, I got the following cases:

- If the nrf connect app says that the indication were enabled before I enabled them, I get the Error 3 Writing Not permited
- If the nrf connect app says that the indication were disabled before I enabled them, it works fine entering the bootloader
- If the nrf connect app says that the indication were enabled but before enabled them I disable them and enable them again, it works

philips77, 2018-10-22:
Is your device bonded? Why would the indications be enabled before starting DFU?
Which SDK is your project based on?

nvelozsavino, 2018-10-22:
Yes, the device is bonded, required by my application. the sdk I'm using is 15.0.0 and the device is a nRF52832, I'm trying to disable the indication before enabling it again in the library code but it doesn't work, I still get the Error 3 Write not permited.

As I mentioned before, this is only happening with this library, using the nrf-connect app (on Android) or using the Android DFU library embedded in our android app works perfect.

nvelozsavino, 2018-10-22:
I just tested the nrf-connect application on an iphone 6 with iOS 12.0.1, I connected the device, make the bond and try to start the dfu process. I got a disconnection as soon as it started, probably because the same issue. It seems to me that there's a compatibility problem with the iOS 12.0.1 and the library.

However, I found something interesting, if the device is already in bootloader mode, the nrfConnect application on iOS 12.0.1 is able to do the dfu upgrade, therefore, the problem seems to be when the application writes to the buttonless dfu characteristic to enter bootloader mode or even for changing the name.

In android works flawless

nvelozsavino, 2018-10-22:
I solve it, tomorrow I'll create a pull request with the code after some more tests

philips77, 2018-10-22:
I'll pass your comments to the sdk team.

nvelozsavino, 2018-10-23:
Reviewing the code of the nRF52, it seems that my function handling the BLE_GATTS_EVT_WRITE for one of my services is responding BLE_GATT_STATUS_ATTERR_WRITE_NOT_PERMITTED for any unknown charactereristic for my service (this includes the Butonless DFU characteristic).

As far as I understand, the BLE_GATTS_EVT_WRITE event coming from the Softdevice is passed to all registered handlers, it seems that my service is responding with BLE_GATT_STATUS_ATTERR_WRITE_NOT_PERMITTED but the handler for the Buttonless DFU is actually evaluating the request correctly and somehow the ALLOW response is being override by the response of my handler.

This is a mistake in my code that shouldn't respond anything if the request characteristic handler is outside of my service boundaries. However what's odd is the difference in behavior of the Android library and the iOS library. it seems that the Android library is ignoring the rejection of the write and continuing the dfu process, meanwhile the iOS library is stopping the process as soon as it gets the rejection of the write. 

The solution I implemented inside the library was split in 2 parts:

1. Disable the indication and enable them again in the Buttonless DFU. this aims to solve the issue that the indications status are cached by the iOS and to force the 0200 command to enable the indication.
2. Ignore the Write not permited error after writing the enter bootloader command and also after renaming the device command

I also implemented some changes in the upper layers of the application, to disconnect the device, then connect again, discover the services and start the dfu process without writing or reading any other characteristic that would require a secure link, therefore the dfu should start without bonding the device.

Since the baseline problem lies within my code, I won't be making the pull request unless you ask me to do it if you want to have both android and ios behaving the same

philips77, 2018-10-23:
Great that you found the bug and a workaround for it. I don't think we need any changes then, as it only applies to your device. Others should not be affected. Therefore I'm closing the issue.

closed 2018-10-23 by philips77