nordicsemi/IOS-DFU-Library #244: symbol(s) not found for architecture x86_64

Handby, 2018-10-25:
Xcode10.0 | Zip (1.1.0) | iOSDFULibrary (4.2.1)

I have set the target swift-version to 4.2，but encountered an error while compiling.

`iOSDFULibrary.DFUStreamHex in libiOSDFULibrary.a(DFUStreamHex.o)
      ...
  "nominal type descriptor for Swift.UInt8", referenced from:
      protocol conformance descriptor for Swift.UInt8 : iOSDFULibrary.DataConvertible in iOSDFULibrary in libiOSDFULibrary.a(Data.o)
  "(extension in Swift):Swift.SetAlgebra.isEmpty.getter : Swift.Bool", referenced from:
      protocol witness for Swift.SetAlgebra.isEmpty.getter : Swift.Bool in conformance __C.CBCharacteristicProperties : Swift.SetAlgebra in __C_Synthesized in libiOSDFULibrary.a(DFUControlPoint.o)
  "dispatch thunk of Swift.Hashable.hash(into: inout Swift.Hasher) -> ()", referenced from:
      resilient protocol witness table for iOSDFULibrary.DFUOpCode : Swift.Hashable in iOSDFULibrary in libiOSDFULibrary.a(DFUControlPoint.o)
      resilient protocol witness table for iOSDFULibrary.InitDfuParametersRequest : Swift.Hashable in iOSDFULibrary in libiOSDFULibrary.a(DFUControlPoint.o)
      resilient protocol witness table for iOSDFULibrary.DFUResultCode : Swift.Hashable in iOSDFULibrary in libiOSDFULibrary.a(DFUControlPoint.o)
      resilient protocol witness table for iOSDFULibrary.DFUFirmwareType : Swift.Hashable in iOSDFULibrary in libiOSDFULibrary.a(DFUFirmware.o)
      resilient protocol witness table for iOSDFULibrary.DFUUuidType : Swift.Hashable in iOSDFULibrary in libiOSDFULibrary.a(DFUUuidHelper.o)
      resilient protocol witness table for iOSDFULibrary.ButtonlessDFUOpCode : Swift.Hashable in iOSDFULibrary in libiOSDFULibrary.a(ButtonlessDFU.o)
      resilient protocol witness table for iOSDFULibrary.ButtonlessDFUResultCode : Swift.Hashable in iOSDFULibrary in libiOSDFULibrary.a(ButtonlessDFU.o)
      ...
  "type metadata for Swift.UInt8", referenced from:
      iOSDFULibrary.Request.data.getter : Foundation.Data in libiOSDFULibrary.a(DFUControlPoint.o)
      type metadata accessor for [Swift.UInt8] in libiOSDFULibrary.a(DFUControlPoint.o)
      iOSDFULibrary.Request.description.getter : Swift.String in libiOSDFULibrary.a(DFUControlPoint.o)
      iOSDFULibrary.Response.description.getter : Swift.String in libiOSDFULibrary.a(DFUControlPoint.o)
      iOSDFULibrary.DFUControlPoint.peripheral(_: __C.CBPeripheral, didUpdateValueFor: __C.CBCharacteristic, error: Swift.Error?) -> () in libiOSDFULibrary.a(DFUControlPoint.o)
      iOSDFULibrary.DFUFirmwareType.hash(into: inout Swift.Hasher) -> () in libiOSDFULibrary.a(DFUFirmware.o)
      iOSDFULibrary.DFUVersion.peripheral(_: __C.CBPeripheral, didUpdateValueFor: __C.CBCharacteristic, error: Swift.Error?) -> () in libiOSDFULibrary.a(DFUVersion.o)
      ...
  "(extension in Swift):Swift.Error._domain.getter : Swift.String", referenced from:
      protocol witness for Swift.Error._domain.getter : Swift.String in conformance iOSDFULibrary.DFUStreamZipError : Swift.Error in iOSDFULibrary in libiOSDFULibrary.a(DFUStreamZip.o)
      protocol witness for Swift.Error._domain.getter : Swift.String in conformance iOSDFULibrary.ZipError : Swift.Error in iOSDFULibrary in libiOSDFULibrary.a(ZipArchive.o)
      protocol witness for Swift.Error._domain.getter : Swift.String in conformance Zip.ZipError : Swift.Error in Zip in libZip.a(Zip-cef416b53dab88284dd62715be9c9e4d6b9851908b1a3f6eea4c1058861549cd.o)
  "dispatch thunk of Swift.SetAlgebra.symmetricDifference(A) -> A", referenced from:
      resilient protocol witness table for __C.CBCharacteristicProperties : Swift.SetAlgebra in __C_Synthesized in libiOSDFULibrary.a(DFUVersion.o)
  "(extension in Swift):Swift.Collection.first.getter : A.Element?", referenced from:
      iOSDFULibrary.DFUServiceInitiator.init(target: __C.CBPeripheral, queue: __C.OS_dispatch_queue?) -> iOSDFULibrary.DFUServiceInitiator in libiOSDFULibrary.a(DFUServiceInitiator.o)
      static Zip.Zip.unzipFile(_: Foundation.URL, destination: Foundation.URL, overwrite: Swift.Bool, password: Swift.String?, progress: (Swift.Double) -> ()?, fileOutputHandler: (Foundation.URL) -> ()?) throws -> () in libZip.a(Zip-cef416b53dab88284dd62715be9c9e4d6b9851908b1a3f6eea4c1058861549cd.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)`

mostafaberg, 2018-10-25:
how did you integrate the project ?

Handby, 2018-10-26:
hello mostafaberg,i integrate the project by cocopods

mostafaberg, 2018-10-26:
@Handby Can you give me more info? I need to know
- cocoapods binary version ? `pod --version`
- Xcode version

Handby, 2018-10-29:
@mostafaberg 
My cocoapods version is 1.5.3
My Xcode version is 10.0 (10A255).

I tried to reinstall and still failed.

mostafaberg, 2018-10-29:
Try with Cocoapods 1.6.0, this should fix your issue 👍

Handby, 2018-10-30:
thanks you @mostafaberg 👍