nordicsemi/IOS-DFU-Library #245: DFUError `operationFailed` is not reporting

DevAndArtist, 2018-10-29:
I experience a new issue where the `DFUError` is not getting reported through the delegate.

On our peripheral we have to flash a second chip after the STM firmware is finished uploading (therefore the  100 % you'll see in the logs '💙 100'). Sometimes it does fail, and I'll see logs of an error but the `DFUError` is not emitted which we would need to reach the custom `failed` state instead of `completed`.

```
💙 99
💛 Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
💛 peripheral.writeValue(0x03, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
💛 Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
💛 Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60030100300200f32802ef
💛 Checksum (Offset = 143360, CRC = EF0228F3) received
💛 Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
💛 peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
💛 Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
💛 Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600401
💛 Data object executed
💛 Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
💛 peripheral.writeValue(0x010210040000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
💛 Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
💛 Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
💛 Data object 36/36 created
💛 Uploading firmware...
💛 Sending firmware to DFU Packet characteristic...
💙 100
💛 Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
💛 peripheral.writeValue(0x03, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
💛 Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
💛 Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60030110340200f3362471
💛 Checksum (Offset = 144400, CRC = 712436F3) received
💛 Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
💛 peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
💛 Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
💛 Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60040a

💛 Error 10: Operation failed // HERE

💛 Disconnecting...
💛 centralManager.cancelPeripheralConnection(peripheral)
💛 [Callback] Central Manager did disconnect peripheral
💛 Disconnected
💚 Completed
```

If you have any suggestions what I could check or debug, I'll try my best to find to find the short circuit.

DevAndArtist, 2018-10-29:
Here is a small part of the delegate object you you better understand the logs I posted above:

```
extension _Box : DFUServiceDelegate {
  ///
  func dfuStateDidChange(to state: DFUState) {
    print("💚 \(state.description())")
    switch state {
    case .aborted:
      delegate?.event(.aborted)
    case .completed:
      delegate?.event(.completed)
    default:
      break
    }
  }

  ///
  func dfuError(_ error: DFUError, didOccurWithMessage message: String) {
    print("❤️ \(error) \(message)")
    delegate?.event(.failed)
  }
}

extension _Box : DFUProgressDelegate {
  ///
  func dfuProgressDidChange(
    for part: Int,
    outOf totalParts: Int,
    to progress: Int,
    currentSpeedBytesPerSecond: Double,
    avgSpeedBytesPerSecond: Double
  ) {
    self.progress = progress
    let event: Peripheral.Update.State = .updating(progress)
    print("💙", progress)
    delegate?.event(event)
  }
}

extension _Box : LoggerDelegate {
  ///
  func logWith(_ level: LogLevel, message: String) {
    print("💛", message)
  }
}
```

Initialization looks something like this:

```swift
let initiator = DFUServiceInitiator(queue: system(for: .bootloader).queue)
_initiator = initiator.with(firmware: _firmware)
let selector = PeripheralSelector(serialNumber: serialNumber)
_selector = selector
initiator.alternativeAdvertisingNameEnabled = false
initiator.peripheralSelector = selector
initiator.delegate = _box
initiator.progressDelegate = _box
#if DEBUG
  initiator.logger = _box
#endif
```

DevAndArtist, 2018-10-29:
Just so that you know, I get `DFUError` messages for disconnections. The operation failed error is just not reported in my case. (I cannot tell about other errors though.)

DevAndArtist, 2018-10-29:
I'll pull the branch and test it.

DevAndArtist, 2018-10-29:
```
❤️ DFUError Operation failed
```

I get the error I was missing now.

philips77, 2018-10-29:
We'll release 4.2.2 tomorrow, if no more bugs are found.

PR #246 (2018-11-01): Bugfix/245
This should fix #245. The reason the error wasn't reported to the app was the *activating* flag, that was set prior to sending the Execute command, but not reset when the Execute failed. Then, in [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift#L531) it was reporting state Completed without checking if delegate.error was set. With this flag cleared properly, it won't go to into this if, and will call `super.peripheralDidDisconnect()` which will check for the error.
closed 2018-10-29 by philips77