nordicsemi/IOS-DFU-Library #249: Modify the behavior on which queue callbacks are emitted and queue guarantees

DevAndArtist, 2018-11-15:
Hi there, I'm currently working on improving some threading issues in our application and I would like to reduce thread hopping as much as possible. According to the documentation the library is pushing all delegation events onto the main queue. This is not ideal in our case because we have to transform the events into custom events since we chain multiple updates manually and compute a custom update state which only then can be forwarded to the UI related code. Before that I would like the whole library to run on the queue specified during initialization.

There is a method called [`dispatchPrecondition`](https://developer.apple.com/documentation/dispatch/1780605-dispatchprecondition) which helps to prove a queue we're executing code on. It would be great if the whole library can be guaranteed to run on the same queue which we specify during the init. Furthermore to avoid much breakage you can add an additional flag which by default falls back to the old behavior and report callbacks on the main queue if not disabled (during init?).

I think of something along these lines:

```swift
@objc public class DFUServiceInitiator : NSObject {
    // The default is `true` to preserve old behavior. However if disabled, the
    // library guarantees that it will operate on the specified `queue`.
    // If `quque` is nil then everything will operate on the `main` queue and the
    // second paramenter is ignored.
    @objc public init(queue: DispatchQueue? = nil, reportsOnMainQueue: Bool = true) {
       ...
    }
}
```

philips77, 2018-11-15:
Hi, that's very interesting case. I'm busy with another task right now (and will be for couple of weeks), but can look into it when free.
Ideally, I would ask you for a PR that we can discuss and accept, as you are already working with it.

DevAndArtist, 2018-11-15:
@philips77 I'd happily provide a PR to fix this. However I would need a little guidance to implement it correctly. For that I need to know what you think it should behave like or what API changes you personally would make. Bikeshedding is enough, so I can get an idea of your preferences. ;)

philips77, 2018-11-15:
I think the easiest would be to add a property in the `DFUServiceInitiator` called `delegateQueue: DispatchQueue`, initially set to `DispatchQueue.main` in the constructor.
From places like [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/develop/iOSDFULibrary/Classes/Implementation/SecureDFU/DFU/SecureDFUExecutor.swift#L71), you could just call `initiator.delegateQueue.async(execute...`. That would allow the queue to be changed during DFU, which I don't think should be possible, but I guess this may be a feature, not a bug.
Here are all places where async is called: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/search?utf8=%E2%9C%93&q=DispatchQueue.main.async&type=
All Executors have access to initiator. Problem is with Packet characteristics to report the progress, but that can be added as a parameter with progressDelegat. Lomethign like: 
```swift
func sendNext(_ prnValue: UInt16, packetsFrom range: Range<Int>, of firmware: DFUFirmware,
andReportProgressTo progress: DFUProgressDelegate?, on queue: DispatchQueue, andCompletionTo complete: @escaping Callback) {
...
}
```
Last call, in LegacyDFUService, I think may stay as main thread?

DevAndArtist, 2018-11-15:
Thank you, I will soon look at it. Is there anything special about `LegacyDFUService` that lets you think it should stay at the main thread?

philips77, 2018-11-15:
This is not a callback, it just restarts dfu after some delay in some condition. That was required, as old dfu needed time to erase flash, or something like that. You may implement it in a different way, if there is better. Or use the queue given in initiator's init?

philips77, 2019-01-29:
Hello @DevAndArtist, any movement on this?

DevAndArtist, 2019-01-29:
@philips77 actually no, it took me like two month to fully fix and implement a similar guards in our library. I'm currently filled with other important work which prevents me to work on that ticket, but it's still on our TO-DO list to eventually improve the flashing with these kind of dispatching guarantees and reduce context switches.

closed 2019-03-01 by philips77