nordicsemi/IOS-DFU-Library #250: Randomic stuck on “Connecting”

caggiulio, 2018-11-15:
Sometimes the process is blocking in "enabling dfu mode in "connecting state... then nothing happens at all... seems randomical three times the trial is going to a good end

caggiulio, 2018-11-16:
when the update "failes", the logs are this: 

W: Application with buttonless update found
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
...
State changed to: Enabling DFU Mode
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Connecting to SmartTag...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to SmartTag
V: Discovering services...
D: peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
I: Services discovered
V: Legacy DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I: DFU characteristics discovered
V: Reading DFU Version number...
D: peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
I: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
A: Version number read: 0.1
...
V: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
State changed to: Starting
D: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
V: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A: DFU Control Point notifications enabled
W: Application with buttonless update found
...
State changed to: Enabling DFU Mode
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Connecting to SmartTag...
D: centralManager.connect(peripheral, options: nil)

caggiulio, 2018-11-20:
Any suggestions?

shift00, 2018-11-21:
Hello, I have the same issue. 
I changed configuration (I have marked it in the code below) but my solutions not working :(

Console:
`...
[FUS] 🔥 D: [Callback] Central Manager did disconnect peripheral
[FUS] 🔥 I: Disconnected by the remote device
[FUS] 🔥 V: Connecting to bleDevice...
[FUS] 🔥 D: centralManager.connect(peripheral, options: nil)
[FUS] 🔥 D: [Callback] Central Manager did connect peripheral
[FUS] 🔥 I: Connected to bleDevice
[FUS] 🔥 V: Discovering services...
[FUS] 🔥 D: peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
[FUS] 🔥 D: [Callback] Central 
[FUS] 🔥 I: Connected to bleDevice
[FUS] 🔥 V: Discovering services...
[FUS] 🔥 D: peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
[FUS] 🔥 I: Services discovered
[FUS] 🔥 V: Legacy DFU Service found
[FUS] 🔥 I: Services discovered
...`

My configuration:
` var dfuInitiator = DFUServiceInitiator(queue: DispatchQueue(label: "DfuUpdate")) 
        dfuInitiator.forceDfu = false
        dfuInitiator.delegate = self
        dfuInitiator.progressDelegate = self
        dfuInitiator.logger = self
        dfuInitiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true
        
        if #available(iOS 11.0, macOS 10.13, *) {
            dfuInitiator.packetReceiptNotificationParameter = 0
        }
        
        dfuInitiator = dfuInitiator.with(firmware: firmware)
        
        dfuController = dfuInitiator.start(target: dfuPeripheral)`

caggiulio, 2018-11-21:
everything seems to be resolved from the moment I implemented the DFUPeripheralSelectorDelegate functions

closed 2018-11-28 by caggiulio