nordicsemi/IOS-DFU-Library #252: 202 Device disconnected unexpectedly

rajtharan-g, 2018-11-26:
Hi, we are trying to update the firmware using DFU library. Once we start the DFU process, the DFU state changes to "connecting", "starting" and then we are getting "202 Device disconnected unexpectedly" error when the process of uploading firmware is about `to` start.

This is how we have initialised the DFU process:
```
let dfuInitiator = DFUServiceInitiator(queue: DispatchQueue(label: "DFUQueue"))
dfuInitiator.delegate = self
dfuInitiator.progressDelegate = self
dfuInitiator.logger = self
 if #available(iOS 11.0, *) {
     dfuInitiator.packetReceiptNotificationParameter = 0
 }
 dfuController = dfuInitiator.with(firmware: firmware).start(target: dfuPeripheral)

```

DFU version: 4.2.2
Device: iPhone X
iOS: 12.1

The following logs are recorded for your reference:
V Connecting to DfuTarg...
D centralManager.connect(peripheral, options: nil)
 **DFU state change: Connecting**
D [Callback] Central Manager did connect peripheral
I Connected to DfuTarg
V Discovering services...
D peripheral.discoverServices(nil)
I Services discovered
V Starting Legacy DFU...
I Connected to DfuTarg
I Services discovered
V Legacy DFU Service found
V Discovering characteristics in DFU Service...
D peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I DFU characteristics discovered
V Reading DFU Version number...
D peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
I Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0800
A Version number read: 0.8
 **DFU state change: Starting**
V Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
D peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
V Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A DFU Control Point notifications enabled
V Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
V Writing image sizes (0b, 0b, 66792b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
D peripheral.writeValue(0x0000000000000000e8040100, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
D [Callback] Central Manager did disconnect peripheral
I Disconnected by the remote device
 **DFU error: 202 Device disconnected unexpectedly**

philips77, 2018-11-26:
Do you return version 0.8 from the DFU Version characteristic in application mode? This characteristic should return 0.1 (0x0100) when in app mode with buttonless feature enabled, and 0.8 (0x0800) only in bootloader mode. The app has no clue what BL version is installed, and based on this value the DFU lib knows, that we are now in app mode and it need to jump to bootloader mode to continue. Receiving 0.8 says, that that's already the booloader and it may simply continue. From the logs I can understand, that after sending 0x0104 the device has reset, so that it was in app mode when the connection started.

philips77, 2018-11-26:
Well... then it's independent on the library. Looks like your phone just looses the connection to the device at that time. My estimation would be that after sending the fw sizes is starts to erase the memory to make a space for the new firmware and during that time it ignores the ongoing connection. Should be sending empty packets in each connection interval to keep the connection open, but looks like 100% of the cpu is busy with erasing, so the phone may assume it disconnected. This is, however, only a guess. But as you are using somewhat old DFU, from SDK 8-11, it may have issues. We haven't seen them here, so they may be related to your hardware/software on the device side. Perhaps lowering connection interval from the peripheral side may help, if it's possible?

philips77, 2018-11-29:
Well.. as you see in the previous logs, the lib is now waiting for the notification to come:
``` 
...
 A Start DFU (Op Code = 1, Upload Mode = 4) request sent
 I Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101 <---
 A Response (Op Code = 1, Status = 1) received
...
```

philips77, 2018-12-19:
No, it looks like something firmware related. Please, ask your question https://devzone.nordicsemi.com/ Our Support Team will gladly help you, they are experts in this.

philips77, 2019-01-29:
Hello, I hope you did get a response. I'll close the issue here. Feel free to reopen it or create a new one if you need more help from us.

closed 2019-01-29 by philips77