nordicsemi/IOS-DFU-Library #253: Disconnected by the remote device

haohongwei, 2018-11-27:
hi, all:

After app writing value to ble successfully , it was disconnected by the device.

this is my log:

2018-11-27 21:00:08.856164+0800 [D]SafeWallet[2040:130808] Discovered services for XXXXXX
2018-11-27 21:00:11.122263+0800 [D]SafeWallet[2040:130912] logWith Connecting to XXXXXX...
2018-11-27 21:00:11.122741+0800 [D]SafeWallet[2040:130912] logWith centralManager.connect(peripheral, options: nil)
2018-11-27 21:00:11.125145+0800 [D]SafeWallet[2040:130725] logWith [Callback] Central Manager did connect peripheral
2018-11-27 21:00:11.125574+0800 [D]SafeWallet[2040:130725] logWith Connected to XXXXXX
2018-11-27 21:00:11.125687+0800 [D]SafeWallet[2040:130725] logWith Discovering services...
2018-11-27 21:00:11.125789+0800 [D]SafeWallet[2040:130725] logWith peripheral.discoverServices(nil)
2018-11-27 21:00:11.132454+0800 [D]SafeWallet[2040:130725] logWith Services discovered
2018-11-27 21:00:11.132845+0800 [D]SafeWallet[2040:130725] logWith Starting Secure DFU...
2018-11-27 21:00:11.133674+0800 [D]SafeWallet[2040:130725] logWith Connected to XXXXXX
2018-11-27 21:00:11.133845+0800 [D]SafeWallet[2040:130725] logWith Services discovered
2018-11-27 21:00:11.133989+0800 [D]SafeWallet[2040:130725] logWith Secure DFU Service found
2018-11-27 21:00:11.135119+0800 [D]SafeWallet[2040:130725] logWith Discovering characteristics in DFU Service...
2018-11-27 21:00:11.135303+0800 [D]SafeWallet[2040:130725] logWith peripheral.discoverCharacteristics(nil, for: FE59)
2018-11-27 21:00:11.137915+0800 [D]SafeWallet[2040:130912] logWith DFU characteristics discovered
2018-11-27 21:00:11.138667+0800 [D]SafeWallet[2040:130566] 1
2018-11-27 21:00:11.140231+0800 [D]SafeWallet[2040:130912] logWith Enabling notifications for 8E400001-F315-4F60-9FB8-838830DAEA50...
2018-11-27 21:00:11.140439+0800 [D]SafeWallet[2040:130912] logWith peripheral.setNotifyValue(true, for: 8E400001-F315-4F60-9FB8-838830DAEA50)
2018-11-27 21:00:11.190001+0800 [D]SafeWallet[2040:130725] logWith Notifications enabled for 8E400001-F315-4F60-9FB8-838830DAEA50
2018-11-27 21:00:11.190192+0800 [D]SafeWallet[2040:130725] logWith Buttonless DFU notifications enabled
2018-11-27 21:00:11.190891+0800 [D]SafeWallet[2040:130725] logWith Application with buttonless update found
2018-11-27 21:00:11.191423+0800 [D]SafeWallet[2040:130566] 2
2018-11-27 21:00:11.192140+0800 [D]SafeWallet[2040:130725] logWith Writing to characteristic 8E400001-F315-4F60-9FB8-838830DAEA50...
2018-11-27 21:00:11.193559+0800 [D]SafeWallet[2040:130725] logWith peripheral.writeValue(0x01, for: 8E400001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2018-11-27 21:00:11.250217+0800 [D]SafeWallet[2040:130912] logWith Data written to 8E400001-F315-4F60-9FB8-838830DAEA50
2018-11-27 21:00:11.251524+0800 [D]SafeWallet[2040:130912] logWith Notification received from 8E400001-F315-4F60-9FB8-838830DAEA50, value (0x):200101
2018-11-27 21:00:11.252552+0800 [D]SafeWallet[2040:130912] logWith Response (Op Code = 1, Status = 1) received
2018-11-27 21:00:11.619735+0800 [D]SafeWallet[2040:130912] logWith [Callback] Central Manager did disconnect peripheral
2018-11-27 21:00:11.619989+0800 [D]SafeWallet[2040:130912] logWith Disconnected by the remote device
2018-11-27 21:00:11.620319+0800 [D]SafeWallet[2040:130912] logWith Scanning for the DFU Bootloader...

iOS 11.0.3 iOSDFULibrary (4.2.2)

philips77, 2018-11-27:
That's intended. The device was in app mode, it had buttonless service, and switching to bootloader mode is required.
The command to switch was sent correctly and the device rebooted. Now it should advertise in DFU mode with a different address.
iOS DFU library is using `DfuPeripheralSelector` instance to scan for the next target to connect, as the Bluetooth address is not available on the platform. Check with what data is your DFU advertising and why the app is no picking it up.
Possible reasons are:
1. The device does not advertise -> make sure the DFU bootloader is flashed, not only the app
2. The bootloader advertises with custom data -> create your own `DfuPeripheralSelector` and set it in the `DfuServiceInitiator`. The default one looks for a device with DFU service UUID in the adv packet.

haohongwei, 2018-11-28:
Thank you for your reply.
I will try to use my custom DfuPeripheralSelector.
But my device did not change its address after the commend response successfully.
Is it the right action?

haohongwei, 2018-11-28:
When I use the default DfuPeripheralSelector,  the device still in the powerOn state after it has been disconnected.
But after I use my own DfuPeripheralSelector, the device powered off after it has been disconnected.
I write my own DfuPeripheralSelector as below:

-(void) initDFUService:(NSURL *)firmWareFilePath{
    
    DFUFirmware *firmware = [[DFUFirmware alloc] initWithUrlToZipFile: firmWareFilePath];
    DFUServiceInitiator * dfuInitiator = [[DFUServiceInitiator alloc]initWithQueue: dispatch_get_global_queue(DISPATCH_QUEUE_PRIORITY_DEFAULT, 0)];
    [dfuInitiator withFirmware: firmware];
    dfuInitiator.delegate = self;
    dfuInitiator.progressDelegate = self;
    dfuInitiator.logger = self;
    dfuInitiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = YES;
    dfuInitiator.peripheralSelector = self;
    self.dfuController = [[dfuInitiator withFirmware:firmware] startWithTarget:self.discoveredPeripheral];
}

-(BOOL)select:(CBPeripheral *)peripheral advertisementData:(NSDictionary<NSString *,id> *)advertisementData RSSI:(NSNumber *)RSSI hint:(NSString *)name{
    NSString *localName = advertisementData[CBAdvertisementDataLocalNameKey];
    if(localName!= nil){
        return localName == name;
    }
    return true;
}

- (NSArray<CBUUID *> * _Nullable)filterByHint:(CBUUID * _Nonnull)dfuServiceUUID{
    
    return @[[CBUUID UUIDWithString:@"8E400001-F315-4F60-9FB8-838830DAEA50"]];
}

philips77, 2018-12-04:
Hi, sorry, what is the issue now?

mostafaberg, 2018-12-05:
Won't that just return true for any other device that has no name ?
```
-(BOOL)select:(CBPeripheral *)peripheral advertisementData:(NSDictionary<NSString *,id> *)advertisementData RSSI:(NSNumber *)RSSI hint:(NSString *)name{
    NSString *localName = advertisementData[CBAdvertisementDataLocalNameKey];
    if(localName!= nil){
        return localName == name;
    }
    return true;
}
```

Also, what do you mean by the device powered off ? it should restart in boot loader mode, if it "powered off" maybe it asserted ? what kind of device is that ?

mostafaberg, 2018-12-18:
Any updates on the issue? was it resolved? if we don't get any updates by the end of the week we'll close this issue and assume problem was fixed :)

closed 2019-01-29 by philips77