nordicsemi/IOS-DFU-Library #254: Multiple Upgrades of "too many" fails to complete

tompi, 2018-12-03:
Hi!

Thank you for your work, we are using this and the corresponding android library a lot, and dont see much problem.

I am kind of testing what is possible here, the usecase is our customers typically have a large number of devices needing upgrades at the same time(we sell sets of CPR-manikins), the largest set consisting of 42 devices. We are trying to speed up dfu updates by doing multiple at the same time.

When starting an update, the first thing we do is "rediscovering" the device on a new adapter, and THEN we initiate your lib with the "new" device.

This seems to work fine for 3 devices on my desk, but consistently fails for 4. I would accept that this is a limitation of BLE stack or bandwidth or similar if it wasnt so reproducible. Seems kind of fishy that all started upgrades fails once I start 4 at a time... Attached logs of 3 and 4 at the same time. Seems I get a progress update up til 100%, but finalizing is not happening, and devices are stuck in dfu mode...

Can you see any sign of what is happening, or should I not expect this to work for more than 3?

P.S. This is on my desk, devices out in the open. With devices wrapped inside a cpr-manikin the same issue seems to occur at 3 simultanous upgrades...?

[dfulog_3_at_once.txt](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/2639802/dfulog_3_at_once.txt)
[dfulog_4_at_once.txt](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/files/2639804/dfulog_4_at_once.txt)

mostafaberg, 2018-12-05:
I see that you're using Xamarin, we haven't tested that, but in theory it should work, however it's not a fully supported/tested use case, can you explain to me a bit more what those lines do exactly, I see quite a few of them ?

`2018-12-03T15:13:32.0177040+01:00 Dispose: ############## Disposing Bluetooth adapter `

tompi, 2018-12-05:
Hi, we generate the binding libraries ourselves, works fine.

Those lines are us disposing the BT adapter, it typically happens after we give up waiting for successfull dfu(or after it is successful).

You can see they happen quite a bit long after progress have stopped coming in in unsuccessful dfu attempts. If dfu is a success, it happens right after.

The logs does not seem to report anything wrong, the process just stops after 100% is reported if I try "too many"(4 at the same time). Strange thing is it happens to all procs active at the same time. Is there some shared state in the lib, that if corrupted would cause all active procs to stop?

tompi, 2018-12-05:
Thank you, will try that! Couldnt find it in the docs, is this a param on the dfu initializer?

mostafaberg, 2018-12-05:
Yes, it's the dfuinitiator, [here's](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/422caf59d89f1ded7e9f403f17a5ea863b6b40ff/Example/iOSDFULibrary/View%20Controllers/DFUViewController.swift#L199) an example of how we use it

we made it optional for iOS < 11 because we use a different API to handle that, but I recommend still enabling it regardless of the version just to test it out to see.

tompi, 2018-12-05:
OK, I just updated 6 devices at the same time :) Worked perfect, except it took forever...
(I had 1 timeout during initialization phase, but worked on second attemp, might have been my own timeout(I time out after 10 secs if no progress is reported))

Anyway, I will try to tweak this param and see if I can find a combo which allows for multiple upgrades without stops and without taking too much time. Thank you so much for your help!

Let me know if I can provide some logs that will help you debug the underlying problem(Im compiling  your lib in xcode before binding it, so I can make debug output statements as needed...)

mostafaberg, 2018-12-05:
perfect !, yeah that makes sense, 1 is a very low value so you'll get a notification after sending every packet, causing a huge delay.

now you can try with 6 and slowly towards 12 until you notice the error come back, then you know what's the optimal value for your case :)

I'm still interested in seeing why it would stop all of them when one fails, but I'm sure as you said it's a shared state issue that we need to look into.

when you have time, if you can send me a full verbose log with as much data as possible and we'll look into a fix for your use case! glad to hear it worked out so far!

closed 2018-12-05 by tompi