nordicsemi/IOS-DFU-Library #255: Compatibility with AlamoFire Download and internet connections

MMooreGitHub, 2018-12-17:
Hi, 

Are you aware of any incompatibility issues between the DFU library and using Alamofire.Download?
(https://github.com/Alamofire/Alamofire/blob/master/Documentation/Usage.md#downloading-data-to-a-file)

Once I connect to the Central Manager. For example within the ViewDidAppear and call the 
[Asyncronous] Alamofire.Download task to download a file from the internet , it constantly does not return my download data.
I verified this I am able to download without problem using a sample app which does not perform BlueTooth connections and operations such as the DFU library does.

philips77, 2018-12-18:
Did you try setting a different queue in DfuServiceInitiator's init?

mostafaberg, 2018-12-18:
We do not have any compatibility issues and I don't see how our library would affect your code.

can you show some code where he issues occur ?

philips77, 2019-01-29:
I'll close the issue due to inactivity. Please, feel free to reopen or create another one if needed.

closed 2019-01-29 by philips77