nordicsemi/IOS-DFU-Library #257: Error 6 Operation failed

rajtharan-g, 2018-12-26:
We are getting error code 6: Operation failed error on few users. The logs are recoded on iPhone 7 iOS-12.1.2

```
V Connecting to DfuTarg...
D centralManager.connect(peripheral, options: nil)
D [Callback] Central Manager did connect peripheral
I Connected to DfuTarg
V Discovering services...
D peripheral.discoverServices(nil)
I Services discovered
V Starting Legacy DFU...
I Connected to DfuTarg
I Services discovered
V Legacy DFU Service found
V Discovering characteristics in DFU Service...
D peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I DFU characteristics discovered
V Reading DFU Version number...
D peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
I Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0800
A Version number read: 0.8
**DFU state change: Starting**
V Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
D peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
V Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A DFU Control Point notifications enabled
V Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
V Writing image sizes (0b, 0b, 66784b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
D peripheral.writeValue(0x0000000000000000e0040100, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
I Data written to 00001531-1212-EFDE-1523-785FEABCD123
A Start DFU (Op Code = 1, Upload Mode = 4) request sent
I Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
A Response (Op Code = 1, Status = 1) received
A Writing Initialize DFU Parameters...
V Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D peripheral.writeValue(0x0200, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
V Writing to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
D peripheral.writeValue(0x01000100ffffffff0100feff1a9f, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
V Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D peripheral.writeValue(0x0201, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I Data written to 00001531-1212-EFDE-1523-785FEABCD123
I Data written to 00001531-1212-EFDE-1523-785FEABCD123
I Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100201
A Response (Op Code = 2, Status = 1) received
A Initialize DFU Parameters completed
V Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
**DFU state change: Uploading**
D peripheral.writeValue(0x080000, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I Data written to 00001531-1212-EFDE-1523-785FEABCD123
A Packet Receipt Notif Req (Op Code = 8, Value = 0) request sent
V Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D peripheral.writeValue(0x03, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I Data written to 00001531-1212-EFDE-1523-785FEABCD123
A Uploading firmware...
V Sending firmware to DFU Packet characteristic...
I Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100306
A Response (Op Code = 3, Status = 6) received
E Error 6: Operation failed
V Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D peripheral.writeValue(0x06, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100306
A Response (Op Code = 3, Status = 6) received
E Error 6: Operation failed
I Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100306
A Response (Op Code = 3, Status = 6) received
E Error 6: Operation failed
I Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100306
A Response (Op Code = 3, Status = 6) received
E Error 6: Operation failed
I Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100306
A Response (Op Code = 3, Status = 6) received
E Error 6: Operation failed
I Data written to 00001531-1212-EFDE-1523-785FEABCD123
A Reset (Op Code = 6) request sent
D [Callback] Central Manager did disconnect peripheral
I Disconnected by the remote device
**DFU error: 6 Operation failed**
```

Also there is a doubt on whether the packetReceiptNotificationParameter is properly used. 
```
if #available(iOS 11.0, *) {
            dfuInitiator.packetReceiptNotificationParameter = 0
        }
```
Please let us know whether the above parameter could cause a issue  or any other which may trigger the operation faced issue.

philips77, 2018-12-26:
I'll be back at work on January the 7th.

philips77, 2019-01-29:
Hello, did you manage to solve the problem? If not, could you ask a question on [DevZone](https://devzone.nordicsemi.com), as this is more DFU related than with this library. Thank you!

As to the PRN value, setting it to 0 may not work when you are using DFU bootloader from an older SDK. Error 6 may indicate that the target device is not ready to receive data yet, so you should slow the transfer down. This can be done by setting PRN value to 12, 6 or even 2 or 1. You should try what value works for you. This is no longer an issue with new SDKs, where the flash storage has been improved.

philips77, 2019-02-26:
Closing due to inactivity.

closed 2019-02-26 by philips77